espressif/esp-zigbee-sdk #347: Is it possible to maintain wifi and zigbee at the same time on ESP32C6?

MingyaoLiu, 2024-05-23:
### Question

Hi, I'm reallly new to wifi/ble/zigbee, I think wifi and BLE would work at the same time, but how about wifi and zigbee?

Any example of how it could work? I tried running wifi and zigbee and i got kernel panic. So im wondering if anyone has tried it or if there is an example somewhere.

Thx

djorr5, 2024-05-23:
2 SOC solution like the below board works for me:

[https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html](url)

Then use the gateway example. [https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway](url)

I don't think using 1 x C6 for both protocols at the same time is currently supported.

MingyaoLiu, 2024-05-23:
i see. I will try another route for my use case, probably just do wifi and nimble at the same time.
Hopefully there will be some official example in the future for zigbee and wifi.

closed 2024-05-23 by MingyaoLiu