espressif/esp-zigbee-sdk #348: esp_zb_zcl_set_attribute_val function is not working for THERMOSTAT.

mundevx, 2024-05-23:
### Question

I created thermostat server cluster with attributes system_mode, control_sequence, etc. On coordinator side, i am using Home Assistant. After successful joining networ, cluster matching and binding are OK. Now when i switch bewteen mode off or cool from Home Assistant, i got call back in espressif zb_attribute_handler using following lines  

 if (message->info.cluster == ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT) {
	if (message->attribute.id == ESP_ZB_ZCL_ATTR_THERMOSTAT_SYSTEM_MODE_ID && message->attribute.data.type == ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM) {
         //got system mode value here
       }
}

   but when i send from my espressif switch, the command doesn't reach to Home Assistant and unable to change GUI system mode or temperature or any other parameter. I am using esp_zb_zcl_set_attribute_val

    status = esp_zb_zcl_set_attribute_val(
        1,
        ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT,
        ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        ESP_ZB_ZCL_ATTR_THERMOSTAT_SYSTEM_MODE_ID,
        (uint8_t*)&ac_mode,  //mode = cool or off
        false
    );
I there any other function for thermostat cluster to update attribute?? What will be the issue?

xieqinan, 2024-05-27:
@mundevx ,

The access of the `ESP_ZB_ZCL_ATTR_THERMOSTAT_SYSTEM_MODE_ID` attribute being `READ, WRITE, and SCENE` means that changes made using `esp_zb_zcl_set_attribute_val()` will only take effect locally and will not be reported to the client.

By the way, could you please provide more details about the `system mode id` for your application? This attribute is usually set by the client side. If your client expects to get the value, it should use the read command to retrieve it. Why do you expect the value to be set from the server to the client?

mundevx, 2024-05-29:
@xieqinan , Thanks for reply. 

I build a zigbee device as Air conditioning purpose and use attribute ESP_ZB_ZCL_ATTR_THERMOSTAT_SYSTEM_MODE_ID with cool and off mode only  in thermostat cluster. Now i bind this device to Home Assistant as client cluster and my zigbee device acts as thermostat server cluster. Home Assistant shows the display with mode changing option from cool to off and vice versa and also i can also set temperature  from home assistant to my device. But when I am sending AC command on-off from my zigbee device switch using below function,

    status = esp_zb_zcl_set_attribute_val(
        1,
        ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT,
        ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        ESP_ZB_ZCL_ATTR_THERMOSTAT_SYSTEM_MODE_ID,
        (uint8_t*)&ac_mode,  //mode = cool or off
        false
    );,  then it is not working and command not reached to Home Assistant.

xieqinan, 2024-05-29:
@mundevx ,

Please note that `esp_zb_zcl_set_attribute_val()` is not a command; it only sets the attribute value locally.

mundevx, 2024-05-29:
@xieqinan , ok, but i want to know that is there any way to send data from thermostat server device  to home assistant. Is espressif zigbee sdk has any special function for thermostat cluster instead of esp_zb_zcl_set_attribute_val?

xieqinan, 2024-05-30:
@mundevx ,

According to the description in the ESP ZCL library specification, there is no command used for `ESP_ZB_ZCL_ATTR_THERMOSTAT_SYSTEM_MODE_ID (0x001c)`. You can refer to the [specification](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf). The esp-zigbee-sdk supports all commands for the thermostat cluster as defined in the specification. As far as I understand, the system mode is used by the thermostat client side to control the server. It is not meaningful for the server side to send the value to the client side just for displaying. I advise referring to the ZCL specification to better define the behavior of the device.

mundevx, 2024-05-30:
@xieqinan , Ok sure. Thank you!

chshu, 2024-06-12:
Please reopen if any follow up questions.

closed 2024-06-12 by chshu