espressif/esp-zigbee-sdk #349: What is Maximum No of layer or Hops can be achieved?

mightChamp, 2024-05-23:
### Question

I have 1000 nodes which are in a line(Street light) 50-100 meter apart. So, I want to make sure that in zigbee mesh I can connect to last one node. Data Speed is not issue, if node get data in 30-60 seconds, it will be ok. Will this system Works with ESP32-C6 as Gateway, as well as for Nodes.

mightChamp, 2024-05-29:
@xieqinan Could you help me in this case.

mightChamp, 2024-05-29:
My System is as follows:

- AC Power Supply 24 Hours running
- Each Node is 50 Meter Apart on street light (last 1000th node is at 25km from gateway).
- Node will never move it is static
- Data Frame will be Max 50 Bytes
- Each node send Data to next 5 nodes, for some trigger.
- Each node send an heartbeat data to node every 15 minute.
- Gateway can send downlink for configurations any time.

So, Based on above could you suggest best Technology to use :
Zigbee/ESP-MDF/ESP-Mesh-Lite/PainlessMesh/ESP-NOW/Openthread

xieqinan, 2024-05-30:
@mightChamp ,

Regarding this issue, due to the special and complex nature of the project scenario, we are currently conducting technical evaluation, verification, and related tests. We will get back to you as soon as possible. In the meantime, we hope you can provide more details about your project.

dipakbhutada, 2024-05-30:
> @mightChamp ,
>
> Regarding this issue, due to the special and complex nature of the project scenario, we are currently conducting technical evaluation, verification, and related tests. We will get [...]

Thank You, We await your prompt reply in order to proceed with the project

mightChamp, 2024-05-30:
> @mightChamp ,
>
> Regarding this issue, due to the special and complex nature of the project scenario, we are currently conducting technical evaluation, verification, and related tests. We will get [...]

Thank for looking this matter, We await your prompt reply in order to proceed with the project.

Hedda, 2024-05-31:
> I have 1000 nodes which are in a line(Street light) 50-100 meter apart. So, I want to make sure that in zigbee mesh I can connect to last one node. Data Speed is not issue, if node get data in 30-60 seconds, it will be ok. Will this system Works with ESP32-C6 as Gateway, as well as for Nodes.

> My System is as follows:
>
> * AC Power Supply 24 Hours running
> * Each Node is 50 Meter Apart on street light (last 1000th node is at 25km from gateway).
> * Node will never move it is static
> * Data [...]

@mightChamp Even though a long-range Zigbee network could be possible on paper, it would never practical, so suggest that you consider looking at a different wireless protocol that was purposely designed to specifically work in "long range" scenatios, (which as example [LoRa/LoRaWAN](https://en.wikipedia.org/wiki/LoRa) and [LPWAN](https://en.wikipedia.org/wiki/Low-power_wide-area_network)). The fact is that the Zigbee wireless protocol was designed for short range (as in within a single building/property even if that could be a relatively large building/property) and always work in a use either a partially connected mesh topology (still with most nodes connected to multiple nodes) or a fully connected mesh network topology, (not as a "line topology" which you describe in your scenario).

- https://en.wikipedia.org/wiki/Network_topology
 -  https://en.wikipedia.org/wiki/Mesh_networking
   - https://en.wikipedia.org/wiki/Network_topology#Fully_connected_network
   - https://en.wikipedia.org/wiki/Network_topology#Partially_connected_network

[image]

Anyway, as far as I read the Zigbee Pro specification only supports a maximum number of 255 “hops” via Zigbee Router devices. Regardless, you should not be using Zigbee for a such use case scenario as you describe above. Because, while Zigbee devices in a Zigbee network mesh can technically hop in a series via multiple Zigbee Router devices, that is not how a Zigbee mesh network is meant to work at all. You have to understand that Zigbee mesh networking tecnology was designed to fully utilize a mesh topology and optimally a fully connected mesh network.

I highly recommend you read and try to follow all the tips in this generic best-practice guide for Zogbee network optimization:

https://community.home-assistant.io/t/zigbee-networks-how-to-guide-for-avoiding-interference-optimizing-using-zigbee-router-devices-repeaters-extenders-to-get-best-possible-range-and-coverage/515752

That contains information and links which explain why Zigbee depends heavily on having a fully connected mesh network.

Here is a very relaive excerpt:

# Network optimization (optimizing for mesh networking)

You have to understand and remember that each Zigbee device by itself has very limited coverage, short range and their weak signals have poor wall penetration, so to workaround this a Zigbee network technology relies heavily on its use of [mesh networking](https://en.wikipedia.org/wiki/Mesh_networking) architecture which forms "[wireless mesh network](https://en.wikipedia.org/wiki/Wireless_mesh_network), where “Zigbee Router” devices (also known as “Zigbee repeaters” or “Zigbee extenders”) will help forward the communication so the messages/packets “[hops](https://en.wikipedia.org/wiki/Hop_(networking))” from Zigbee Router to Zigbee Router to rearch its final destination device when the Zigbee Coordinator is not within direct reach or have a week signal reception, and result is that it forms “mesh topology” (a instead of using a more classic “[star network](https://en.wikipedia.org/wiki/Star_network)” topology where every device would instead practically only talks directly with a single controller like you usually have in a common WiFi-network, which is a other type of [network topology](https://en.wikipedia.org/wiki/Network_topology) where the low-power signals from Zigbee devices do not work well at all).

[image]

That means that all devices in a Zigbee network need it to be a “mesh network” and thus really depend heavily on having a swarm of mains-powered “Zigbee Router” devices that are always-on so they can help forward messages within the network, acting both as a signal repeater and range extended by transmitting data over long distances, stengthening the total coverage and distances possible by passing data messages through the Zigbee network mesh of intermediate devices to reach more distant Zigbee devices, (which may or may not have week signals).

https://www.youtube.com/watch?v=pFpMeW6Cgdg

Therefore the key to a great and healthy Zigbee network mesh is to add/have many “Zigbee Router” devices relatively close to each other (and always powered on) in order for the Zigbee network as a whole to get good coverage and range, as well as improving the overall robustness and robustness and resilience in the network to handle the loss of some Zigbee Routers in case of failures. So often you more or less just need to add several mains-powered Zigbee devices that can act as “Zigbee Router” devices to get a stable network with good coverage, and keep adding even Zigbee router products to achieve a great Zigbee network.

https://www.youtube.com/watch?v=j-hdZOLctvo

mightChamp, 2024-06-01:
@Hedda Thanks for this Brief Information, Let I break my 1000 node line into 5 individual mesh, So each mesh have 200 devices in a line. the distance from gateway to 200th node  will be max at 10KM each at 50 meter apart.

So, will you think by this it will work?
Or make any other arrangement So, it i will be possible?

Thank You

Hedda, 2024-06-01:
> So, will you think by this it will work?

 No, again, Zigbee is not suited for your use

mightChamp, 2024-06-03:
@Hedda  Thanks for your final clarification, you saved me to choosing wrong path.

closed 2024-06-12 by chshu