espressif/esp-zigbee-sdk #350: board locks up under heavy load

djorr5, 2024-05-24:
### Question

I have implemented a coordinator that takes MQTT messages and allows control of a zigbee end device. I am now going through stress testing and noticed that the board locks up if I send "on" "off" mqtt messages in quick succession. The board locks up in such a way that the watchdog never reboots it and i don't get an assertion failure either or any failure message. The serial monitor just freezes and I need to completely power cycle the board to get it running again.  Could you point me in the direction of being able to debug this? I.e. is there another watchdog in menuconfig that I haven't enabled etc?

I have the following enabled:

[image]

I currently waiting for an esp-prog to turn up.

xieqinan, 2024-05-27:
@djorr5 ，

Could you please provide more details about your test? Replace `ieee802154_receive_done()` in `esp-idf/components/ieee802154/driver/esp_ieee802154_dev.c` with the code below. Then, provide us with the test log and sniffer packet.

```
static uint16_t count = 0;
static void ieee802154_receive_done(uint8_t *data, esp_ieee802154_frame_info_t *frame_info)
{
    // If the RX done packet is written in the stub buffer, drop it silently.
    if (s_rx_index == CONFIG_IEEE802154_RX_BUFFER_SIZE) {
        esp_rom_printf("receive buffer full, drop the current frame.\n");
    } else {
        // Otherwise, post it to the upper layer.
        if (count > 1000) {
            frame_info->process = true;
            esp_rom_printf("receive %d frame in %lld\n", count, esp_timer_get_time());
            esp_ieee802154_receive_done(data, frame_info);
            count = 0;
        }
    }
}
```

djorr5, 2024-06-01:
It was a result of a race condition. I have managed to resolve it by handling the MQTT messages better.

closed 2024-06-01 by djorr5