espressif/esp-zigbee-sdk #351: esp_zb_zdo_device_unbind_req not working

mundevx, 2024-05-30:
### Question

I am using esp_zb_zdo_device_unbind_req to UNBIND the device, and my both device types are ROUTER. Device1 Endpoint1 bind to Device2 Endpoint1 successfully bind the Endpoints.  But want to unbind the Device1 Endpoint1 with Device2 Endpoint1 using esp_zb_zdo_device_unbind_req function, it is not working. Then i tested only esp_zb_zdo_device_unbind_req function in another project, and found that using function esp_zb_zdo_device_unbind_req, instead of UNBIND, it BIND the DEVICE1(Switch) with DEVICE2(Light).

xieqinan, 2024-05-30:
@mundevx ,

Could you please attach the code tested by you?

mundevx, 2024-05-30:
Yes Sure,

void simple_binding_request(esp_zb_ieee_addr_t ieee_addr, uint16_t dst_short_addr, uint8_t dst_endpoint, 
    uint8_t src_endpoint, uint16_t cluster_id){
        /* bind the on-off light to on-off switch */
        esp_zb_zdo_bind_req_param_t bind_req;
        memcpy(&(bind_req.src_address), ieee_addr, sizeof(esp_zb_ieee_addr_t));
        bind_req.src_endp = src_endpoint;
        bind_req.cluster_id = cluster_id;
        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        esp_zb_get_long_address(bind_req.dst_address_u.addr_long);
        bind_req.dst_endp = dst_endpoint;
        bind_req.req_dst_addr = dst_short_addr;
        ESP_LOGI(TAG, "short_addr=0x%x\n", dst_short_addr);
        ESP_LOGI(TAG, "Try to BIND Light dstep:%d On/Off srcep=%d cluster=%d\n", dst_endpoint, src_endpoint, cluster_id);
        esp_zb_zdo_device_bind_req(&bind_req, bind_cb, &dst_endpoint);
}
void simple_unbinding_request(esp_zb_ieee_addr_t ieee_addr, uint16_t dst_short_addr, uint8_t dst_endpoint, 
    uint8_t src_endpoint, uint16_t cluster_id){
        /* bind the on-off light to on-off switch */
        esp_zb_zdo_bind_req_param_t bind_req;
        memcpy(&(bind_req.src_address), ieee_addr, sizeof(esp_zb_ieee_addr_t));
        bind_req.src_endp = src_endpoint;
        bind_req.cluster_id = cluster_id;
        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        esp_zb_get_long_address(bind_req.dst_address_u.addr_long);
        bind_req.dst_endp = dst_endpoint;
        bind_req.req_dst_addr = dst_short_addr;
        ESP_LOGI(TAG, "dst_short_addr=0x%x\n", dst_short_addr);
        ESP_LOGI(TAG, "Try to UNBIND Light dstep:%d On/Off srcep=%d cluster=%d\n", dst_endpoint, src_endpoint, cluster_id);
        esp_zb_zdo_device_unbind_req(&bind_req, unbind_cb, &dst_endpoint);
}

calling functions as                               

uint8_t dst_endpoint = 1;
uint8_t src_endpoint = 1;
simple_binding_request(ieee_addr, dst_short_addr, 
                            dst_endpoint, 
                            src_endpoint, 
                            ESP_ZB_ZCL_CLUSTER_ID_ON_OFF);

simple_unbinding_request(ieee_addr, dst_short_addr, 
                            dst_endpoint, 
                            src_endpoint, 
                            ESP_ZB_ZCL_CLUSTER_ID_ON_OFF);

xieqinan, 2024-05-30:
@mundevx ，

I made a simple modification to [esp_zb_switch](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c) to verify your question. It works as expected. Could you copy and test it with the [light example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c)? Please note test it after the network had been formed and this is only a test example.
```
#include "esp_check.h"
#include "esp_err.h"
#include "string.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"
#include "esp_zb_switch.h"

#if defined ZB_ED_ROLE
#error Define ZB_COORDINATOR_ROLE in idf.py menuconfig to compile light switch source code.
#endif
typedef struct light_bulb_device_params_s {
    esp_zb_ieee_addr_t ieee_addr;
    uint8_t  endpoint;
    uint16_t short_addr;
} light_bulb_device_params_t;

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

static const char *TAG = "ESP_ZB_ON_OFF_SWITCH";

static esp_zb_zdo_bind_req_param_t bind_req;
static void zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    static uint8_t count = 0;
    esp_zb_lock_acquire(portMAX_DELAY);
    if (count++ % 2) {
        ESP_EARLY_LOGI(TAG, "unbind command");
        esp_zb_zdo_device_unbind_req(&bind_req, NULL, NULL);
    } else {
        ESP_EARLY_LOGI(TAG, "bind command");
        esp_zb_zdo_device_bind_req(&bind_req, NULL, NULL);
    }
    esp_zb_zcl_on_off_cmd_t cmd_req;
    cmd_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
    cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
    cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;
    ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command");
    esp_zb_zcl_on_off_cmd_req(&cmd_req);
    esp_zb_lock_release();
}

static esp_err_t deferred_driver_init(void)
{
    ESP_RETURN_ON_FALSE(switch_driver_init(button_func_pair, PAIR_SIZE(button_func_pair), zb_buttons_handler), ESP_FAIL, TAG,
                        "Failed to initialize switch driver");
    return ESP_OK;
}

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_level_commissioning(mode_mask) == ESP_OK, , TAG, "Failed to start Zigbee bdb commissioning");
}

static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Bound successfully!");
        if (user_ctx) {
            light_bulb_device_params_t *light = (light_bulb_device_params_t *)user_ctx;
            ESP_LOGI(TAG, "The light originating from address(0x%x) on endpoint(%d)", light->short_addr, light->endpoint);
            free(light);
[... 93 lines cut ...]
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZC_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_on_off_switch_cfg_t switch_cfg = ESP_ZB_DEFAULT_ON_OFF_SWITCH_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_switch_ep = esp_zb_on_off_switch_ep_create(HA_ONOFF_SWITCH_ENDPOINT, &switch_cfg);
    esp_zb_device_register(esp_zb_on_off_switch_ep);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));

    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}
```

mundevx, 2024-05-31:
@xieqinan , yes, this code is working. I am checking my code and let you know what was the issue.

mundevx, 2024-06-01:
@xieqinan , Please check this code on latest sdk, idf version 5.1.3, - And esp-zboss-lib to v1.3.2 ,  esp-zigbee-lib to v1.3.2. This code not working correctly and esp_zb_zdo_device_unbind_req not working. I installed idf version 5.1.3 in Visual Studio and esp-idf-v5.1.2 in Espressif IDE. This code works on v5.1.2 and not on version 5.1.3. That why i have issue in my project that i am building in latest  idf version 5.1.3.

xieqinan, 2024-06-03:
@mundevx ，

I can confirm that I tested the issue using esp-idf v5.1.3, esp-zigbee-lib v1.3.2, and esp-zboss-lib v1.3.2.
[image]

Please note that `static esp_zb_zdo_bind_req_param_t bind_req()` is a global variable and will not persist after a reboot. Ensure you are using esp-idf v5.1.3 in your project, as the esp-zigbee-sdk libraries are compiled based on this version.

If you still encounter issues, please print all members of `bind_req` to console before sending the `esp_zb_zdo_device_unbind_req()` command and attach the log for me.

mundevx, 2024-06-03:
@xieqinan  

I confirmed, i am using esp-idf v5.1.3, and lib v1.3.2. 

Here is logs screenshot for UNBINDING request
______________________________________________________________________________
I (32657) ESP_ZB_ON_OFF_SWITCH: --------Unbind Request Logs--------

I (32665) ESP_ZB_ON_OFF_SWITCH: dst_short_addr=0x7648

I (32671) ESP_ZB_ON_OFF_SWITCH: dst_endp=1

I (32675) ESP_ZB_ON_OFF_SWITCH: dst_addr_mode=3

I (32681) ESP_ZB_ON_OFF_SWITCH: cluster_id=6

I (32686) ESP_ZB_ON_OFF_SWITCH: src_endp=1

I (32691) ESP_ZB_ON_OFF_SWITCH: IEEE Address: 7C:84:4A:FE:FF:CA:4C:40
______________________________________________________________________________

And logs for BINDING request
______________________________________________________________________________
I (32657) ESP_ZB_ON_OFF_SWITCH: --------Bind Request Logs--------

I (32665) ESP_ZB_ON_OFF_SWITCH: dst_short_addr=0x7648

I (32671) ESP_ZB_ON_OFF_SWITCH: dst_endp=1

I (32675) ESP_ZB_ON_OFF_SWITCH: dst_addr_mode=3

I (32681) ESP_ZB_ON_OFF_SWITCH: cluster_id=6

I (32686) ESP_ZB_ON_OFF_SWITCH: src_endp=1

I (32691) ESP_ZB_ON_OFF_SWITCH: IEEE Address: 7C:84:4A:FE:FF:CA:4C:40
______________________________________________________________________________

here are functions

void simple_binding_request(esp_zb_ieee_addr_t ieee_addr, uint16_t dst_short_addr, uint8_t dst_endpoint, 
    uint8_t src_endpoint) {
        /* bind the on-off light to on-off switch */
        esp_zb_zdo_bind_req_param_t bind_req;
        esp_zb_ieee_address_by_short(dst_short_addr, ieee_addr);
        esp_zb_get_long_address(bind_req.src_address);
        bind_req.src_endp = src_endpoint;
        bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        memcpy(bind_req.dst_address_u.addr_long, ieee_addr, sizeof(esp_zb_ieee_addr_t));
        bind_req.dst_endp = dst_endpoint;
        bind_req.req_dst_addr = dst_short_addr;
        ESP_LOGI(TAG, "--------Bind Request Logs--------\n");
        ESP_LOGI(TAG, "dst_short_addr=0x%x\n", bind_req.req_dst_addr);
        ESP_LOGI(TAG, "dst_endp=%d\n", bind_req.dst_endp);
        ESP_LOGI(TAG, "dst_addr_mode=%d\n", bind_req.dst_addr_mode);
        ESP_LOGI(TAG, "cluster_id=%d\n", ESP_ZB_ZCL_CLUSTER_ID_ON_OFF);
        ESP_LOGI(TAG, "src_endp=%d\n", bind_req.src_endp);
        print_ieee_addr(bind_req.src_address);
        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_zdo_device_bind_req(&bind_req, bind_cb, NULL);
        esp_zb_lock_release();
}

void simple_unbinding_request(esp_zb_ieee_addr_t ieee_addr, uint16_t dst_short_addr, uint8_t dst_endpoint, 
    uint8_t src_endpoint){
        /* unbind the on-off light to on-off switch */
        esp_zb_zdo_bind_req_param_t unbind_req;
        esp_zb_ieee_address_by_short(dst_short_addr, ieee_addr);
        esp_zb_get_long_address(unbind_req.src_address);
        unbind_req.src_endp = src_endpoint;
        unbind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
        unbind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        memcpy(unbind_req.dst_address_u.addr_long, ieee_addr, sizeof(esp_zb_ieee_addr_t));
        unbind_req.dst_endp = dst_endpoint;
        unbind_req.req_dst_addr = dst_short_addr;
        ESP_LOGI(TAG, "--------Unbind Request Logs--------\n");
        ESP_LOGI(TAG, "dst_short_addr=0x%x\n", unbind_req.req_dst_addr);
        ESP_LOGI(TAG, "dst_endp=%d\n", unbind_req.dst_endp);
        ESP_LOGI(TAG, "dst_addr_mode=%d\n", unbind_req.dst_addr_mode);
        ESP_LOGI(TAG, "cluster_id=%d\n", ESP_ZB_ZCL_CLUSTER_ID_ON_OFF);
        ESP_LOGI(TAG, "src_endp=%d\n", unbind_req.src_endp);
        print_ieee_addr(unbind_req.src_address);
        esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_zdo_device_unbind_req(&unbind_req, unbind_cb, NULL);
        esp_zb_lock_release();

}

simple_binding_request function sworks ok and bind with every short address that are connected zigbee on network.
but simple_unbinding_request not working. 
Callback repsonse are success for both unbind_cb and bind_cb

mundevx, 2024-06-03:
Hey @xieqinan , 
I am using destination endpoint present,, You above example code works with ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT. 

But instead of cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;  use 

    cmd_req.zcl_basic_cmd.dst_endpoint = 10;  //light endpoint
    cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = DST_SHORT_ADDR;
   cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;  , it will not work.

where DST_SHORT_ADDR is short_address of light device.

mundevx, 2024-06-10:
Can the code provided by me work in both esp-idf v5.1.3 and esp-idf v5.1.2? If not, please print all bind_req information for me based on my code (specifically, print bind_req.dst_address_u.addr_long).   YES

If the issue is only triggered in your project:  YES

The ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT address mode is used when the device has been bound. For the ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT address mode, please refer to [this example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_client/main/esp_HA_customized_switch.c#L57). Its also not working in my project

xieqinan, 2024-06-11:
@mundevx ，

Based on the responses above, the issue should is triggered by the application logic. I believe it would be more effective if you could provide a patch or a simple example for me to debug. Without this, it is difficult for me to identify where the problem is occurring.

lpy4105, 2024-07-13:
Hi @mundevx,

As @xieqinan has explained:
> The `ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT` address mode is used when the device has been bound. For the `ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT` address mode, please refer to [this [...]

To conclude, only `ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT` is related to bindings. If you send commands using `ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT`, the communication should succeed because you know the "network address" of the peer device. Please also see: https://github.com/espressif/esp-zigbee-sdk/issues/383.

If your issue hasn't been resovled, please feel free to put comments.

closed 2024-07-26 by chshu