espressif/esp-zigbee-sdk #353: esp_zb_zcl_identify_cmd_req not working

mundevx, 2024-05-31:
### Question

  
  esp_zb_zcl_identify_cmd_req is not responding, i am getting sequence number, but it is not working. Then i checked 
  esp_zb_zcl_identify_trigger_effect_cmd_req, it is also not responding.

 Destination endpoint is a Zigbee Light Device.

#define DST_SHORT_ADDR   0x29a8    //0x7648
void send_identify_command(uint16_t dst_addr, uint8_t dst_endpoint, uint16_t identify_time) {
    esp_zb_zcl_identify_cmd_t cmd_req;
    cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    cmd_req.identify_time = identify_time;
    cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = dst_addr;
    cmd_req.zcl_basic_cmd.dst_endpoint = dst_endpoint;
    cmd_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
    esp_zb_zcl_identify_cmd_req(&cmd_req);
    ESP_LOGI(TAG, "Identify command sent to address:0x%04x, endpoint:%d", dst_addr, dst_endpoint);

}

 Here is successful binding.....
I (490) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
I (500) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (500) ESP_ZB_ON_OFF_SWITCH: Start network steering
I (580) ESP_ZB_ON_OFF_SWITCH: Found light
I (580) ESP_ZB_ON_OFF_SWITCH: Try to bind On/Off
I (590) ESP_ZB_ON_OFF_SWITCH: Bound successfully!
I (590) ESP_ZB_ON_OFF_SWITCH: The light originating from address(0x29a8) on endpoint(1)
I (970) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NWK Permit Join (0x36), status: ESP_OK
I (970) ESP_ZB_ON_OFF_SWITCH: Joined network successfully (Extended PAN ID: c7:2a:a8:f3:ac:47:63:47, PAN ID: 0x2151, Channel:20)
I (1000) ESP_ZB_ON_OFF_SWITCH: Found light
I (1000) ESP_ZB_ON_OFF_SWITCH: Try to bind On/Off
I (1010) ESP_ZB_ON_OFF_SWITCH: Bound successfully!
I (1010) ESP_ZB_ON_OFF_SWITCH: The light originating from address(0x29a8) on endpoint(1)
I (15740) ESP_ZB_ON_OFF_SWITCH: identify DST_SHORT_ADDR for 5 seconds
I (15740) ESP_ZB_ON_OFF_SWITCH: Identify command sent to address:0x29a8, endpoint:1
I (18650) ESP_ZB_ON_OFF_SWITCH: identify DST_SHORT_ADDR for 2 seconds
I (18650) ESP_ZB_ON_OFF_SWITCH: Identify command sent to address:0x29a8, endpoint:1

and here is my code.....
#include "esp_check.h"
#include "esp_err.h"
#include "string.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"
#include "esp_zb_switch.h"

#if defined ZB_ED_ROLE
#error Define ZB_COORDINATOR_ROLE in idf.py menuconfig to compile light switch source code.
#endif
typedef struct light_bulb_device_params_s {
    esp_zb_ieee_addr_t ieee_addr;
    uint8_t  endpoint;
    uint16_t short_addr;
} light_bulb_device_params_t;

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

static const char *TAG = "ESP_ZB_ON_OFF_SWITCH";

static esp_zb_zdo_bind_req_param_t bind_req;

#define DST_SHORT_ADDR   0x29a8    //0x7648
void send_identify_command(uint16_t dst_addr, uint8_t dst_endpoint, uint16_t identify_time) {
    esp_zb_zcl_identify_cmd_t cmd_req;
    cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    cmd_req.identify_time = identify_time;
    cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = dst_addr;
    cmd_req.zcl_basic_cmd.dst_endpoint = dst_endpoint;
    cmd_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
    esp_zb_zcl_identify_cmd_req(&cmd_req);
    ESP_LOGI(TAG, "Identify command sent to address:0x%04x, endpoint:%d", dst_addr, dst_endpoint);

}

static void zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    static uint8_t count = 0;
    if (count++ % 2) {
        ESP_EARLY_LOGI(TAG, "identify DST_SHORT_ADDR for 2 seconds");
        send_identify_command(DST_SHORT_ADDR, 1, 2);
    } else {
    	ESP_EARLY_LOGI(TAG, "identify DST_SHORT_ADDR for 5 seconds");
        send_identify_command(DST_SHORT_ADDR, 1, 5);
    }

}

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_RETURN_ON_FALSE(esp_zb_bdb_start_top_level_commissioning(mode_mask) == ESP_OK, , TAG, "Failed to start Zigbee bdb commissioning");
}

static void bind_cb(esp_zb_zdp_status_t zdo_status, void *user_ctx)
{
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Bound successfully!");
        if (user_ctx) {
            light_bulb_device_params_t *light = (light_bulb_device_params_t *)user_ctx;
            ESP_LOGI(TAG, "The light originating from address(0x%x) on endpoint(%d)", light->short_addr, light->endpoint);
            free(light);
        }
    }
}

static void user_find_cb(esp_zb_zdp_status_t zdo_status, uint16_t addr, uint8_t endpoint, void *user_ctx)
{
    if (zdo_status == ESP_ZB_ZDP_STATUS_SUCCESS) {
        ESP_LOGI(TAG, "Found light");
        light_bulb_device_params_t *light = (light_bulb_device_params_t *)malloc(sizeof(light_bulb_device_params_t));
        light->endpoint = endpoint;
        light->short_addr = addr;
        esp_zb_ieee_address_by_short(light->short_addr, light->ieee_addr);
        esp_zb_get_long_address(bind_req.src_address);
        bind_req.src_endp = HA_ONOFF_SWITCH_ENDPOINT;
        bind_req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
        bind_req.dst_addr_mode = ESP_ZB_ZDO_BIND_DST_ADDR_MODE_64_BIT_EXTENDED;
        memcpy(bind_req.dst_address_u.addr_long, light->ieee_addr, sizeof(esp_zb_ieee_addr_t));
        bind_req.dst_endp = endpoint;
        bind_req.req_dst_addr = esp_zb_get_short_address(); /* TODO: Send bind request to self */
        ESP_LOGI(TAG, "Try to bind On/Off");
        esp_zb_zdo_device_bind_req(&bind_req, bind_cb, (void *)light);
    }
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Zigbee stack initialized");
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Start network steering");
            esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
			esp_zb_zdo_match_desc_req_param_t  cmd_req;
			cmd_req.dst_nwk_addr = DST_SHORT_ADDR;
			cmd_req.addr_of_interest = DST_SHORT_ADDR;
			esp_zb_zdo_find_on_off_light(&cmd_req, user_find_cb, NULL);
        } else {
            /* commissioning failed */
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
        }
        break;
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK) {
            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            ESP_LOGI(TAG, "Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d)",
                     extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                     extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel());

			esp_zb_zdo_match_desc_req_param_t  cmd_req;
			cmd_req.dst_nwk_addr = DST_SHORT_ADDR;
			cmd_req.addr_of_interest = DST_SHORT_ADDR;
			esp_zb_zdo_find_on_off_light(&cmd_req, user_find_cb, NULL);

        } else {
            ESP_LOGI(TAG, "Network steering was not successful (status: %s)", esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;
    default:
        ESP_LOGI(TAG, "ZDO signal: %s (0x%x), status: %s", esp_zb_zdo_signal_to_string(sig_type), sig_type,
                 esp_err_to_name(err_status));
        break;
    }
}

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZR_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_on_off_switch_cfg_t switch_cfg = ESP_ZB_DEFAULT_ON_OFF_SWITCH_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_switch_ep = esp_zb_on_off_switch_ep_create(HA_ONOFF_SWITCH_ENDPOINT, &switch_cfg);
    esp_zb_device_register(esp_zb_on_off_switch_ep);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    switch_driver_init(button_func_pair, PAIR_SIZE(button_func_pair), zb_buttons_handler);
    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}

xieqinan, 2024-05-31:
@mundevx ，

If you are using the official example in your project, I believe the `dst_endpoint` is 10 rather than 1.

By the way, I recommend installing a [sniffer](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark) to observe the communication process among the Zigbee devices. This can provide feedback on any error reasons. The esp-zigbee-sdk also provides some [documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/introduction.html) that you can refer to when encountering unexpected issues.

mundevx, 2024-05-31:
@xieqinan  i already set the destination endpoint to 1. and i am passing destination address as send_identify_command(DST_SHORT_ADDR, 1, 2);

xieqinan, 2024-06-12:
@mundevx ，

The `esp_zb_zcl_identify_cmd_req()` has been successfully tested by me. To help resolve the issue for you, I suggest you attach a reproducible patch based on the latest esp-zigbee-sdk version and a Wireshark packet. It is difficult to identify the cause based on the code provided above.

chshu, 2024-09-29:
@mundevx Do you have any update on the issue?

closed 2024-12-19 by chshu

mundevx, 2024-12-19:
Hi @chshu ,  This issue has been resolved from my side.