espressif/esp-zigbee-sdk #354: great slowness on the network due to reports that do not reach their destination

diazmanuel, 2024-05-31:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.3

### esp-zigbee-lib version.

1.3.2

### esp-zboss-lib version.

1.3.2

### Espressif SoC revision.

ESP32-H2 & ESP32-C6

### What is the expected behavior?

Hello, the issue that we have detected is a large delay in the network that can collapse it under certain conditions. This delay is due to the queuing of reports that occurs when a destination device is not reached, for example, suppose we have a device A that generates the same report to device B and C, if we read the binding table we will see that there are 2 bindings made in the following order B,C.
If we disconnect device B, the reports that device A generates will not reach it and will take a few minutes to reach device C. I understand that this is because the generation of the reports is done sequentially and that this delay should coincide with the timeout of the reports (that is, the maximum time that device A tries to send and confirm the report to device B).
In itself this is already a major issue in the performance of the network, this would indicate that if any device for some reason is disconnected from the network (lack of power, loss of signal, etc.) this will slow down a large part of the network, However, in addition to the problem of delay, there is the problem that if these report queues accumulate at some point, we can notice that some reports did not even reach their destination after these timeouts.

What we hope is that in some way we can avoid these timeouts or reduce them as much as possible, providing some function to configure these timeouts or number of attempts would be ideal. In any case, this is a serious problem within the performance of the network that should be solved as soon as possible.

### What is the actual behavior?

.

### Steps to reproduce.

set up a network that contains devices A, B and C, generate some bindings in the sequence (A->B and A->C) disconnect device B and generate the report, in this case it should be possible to appreciate the delay of the grid

### More Information.

Furthermore, this problem has been present for some time now, in the issue  (#73) I mentioned this slowness in the network but at that time I was not able to determine the cause correctly, in my last comment the 3 option i mention explains this same phenomenon as one of the possible causes that today I managed to confirm.

diazmanuel, 2024-07-26:
Hello @xieqinan , do you have any updates about this issue, were you able to identify the problem I mentioned? We performed the test you mentioned but we did not notice any differences between both cases. We are pretty sure that the delay only happens when trying to send a report to a device that is not on the network at that time as I mentioned above.

diazmanuel, 2024-09-30:
Hello @xieqinan , do you have any update about this issue, It's been a while and I still haven't received any update on this issue, were you able to identify the problem?

xieqinan, 2024-10-18:
Hi @diazmanuel 

Sorry for the delay due a urgent release of SDK.

I have identified the issue, which is caused by a conflicting logic implemented in the Zigbee stack. Currently, when the stack finds multiple entries in the binding table, it processes them sequentially, handling one entry at a time after the previous entry has been fully processed. In this case, if node B is offline, the stack will attempt retransmissions until it confirms that B is offline. Only then will it proceed to handle the C entry.

We need some time to confirm this issue. In the meantime, could you consider using groupcast to address the problem you're encountering? The [group docs](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_groups.html) may help you implement this solution.

diazmanuel, 2024-12-02:
Hello @xieqinan ,
At this point we are already too advanced in development to implement the group methodology, so we will wait for this problem to be solved correctly. In any case, I appreciate the proposal and I am glad that you have identified it as a problem. Could you tell me the status of this issue, as well as a tentative resolution date, this would help me a lot to organize the release dates of my products.

closed 2024-12-06 by espressif-bot

xieqinan, 2024-12-06:
Hi,

The esp-zigbee-sdk v1.6.1 has already resolved the binding issue. Please test again with the new version. If the issue persists, feel free to reopen this ticket.