espressif/esp-zigbee-sdk #355: Add new cluster to existing endpoint

Kris151, 2024-06-05:
### Question

If I already setup a endpoint in the begining (follow the example, used esp_zb_device_register()), is it possible to add a new cluster to that endpoint?

xieqinan, 2024-06-05:
@Kris151 ,

It is impossible to add a new cluster once `esp_zb_device_register()` is called to register the device data model in the current version.

Could you please provide the application scenarios where you need this feature support?

Kris151, 2024-06-06:
@xieqinan I am making a gateway, and I want to receive the device report. The report will send to the same cluster (like switch on off report will send to gateway on off cluster) right? I want device join -> interview cluster -> see if gateway endpoint1 have this cluster -> if not add a new cluster to the endpoint so that I can get the report.

xieqinan, 2024-06-06:
@Kris151 ,

I think you can create a gateway device that includes the client service for all clusters supported by the SDK to meet the requirements. 
By the way, we are planning to implement this in the next version.

Kris151, 2024-06-06:
@xieqinan thanks

closed 2024-06-06 by Kris151