espressif/esp-zigbee-sdk #356: Better name for esp_zb_main_loop_iteration()

Rob190, 2024-06-05:
### Is your feature request related to a problem?

It appears esp_zb_main_loop_iteration() doesn't actually perform a single iteration as the name would suggest, but instead wraps a while-true loop which calls zboss_main_loop_iteration() on each iteration of the loop. This is completely counter-intuitive and I spent many hours trying to figure out how to use esp_zb_main_loop_iteration() before someone kindly pointed out it was simply a poor choice of name:

https://esp32.com/viewtopic.php?f=22&t=40119&p=133104#p133104

### Describe the solution you'd like.

A better name for esp_zb_main_loop_iteration() e.g. esp_zb_main_loop()

### Describe alternatives you've considered.

Probably I would define my own wrapper for zboss_main_loop_iteration(). I have an aversion to misleading names appearing in code I have to support.

chshu, 2024-06-12:
Sounds reasonable, we will rename it.

closed 2024-08-28 by espressif-bot

xieqinan, 2024-08-28:
@Rob190,

The `esp_zb_main_loop_iteration()` has been deprecated in esp-zigbee-sdk v1.5.0, please use the `esp_zb_stack_main_loop()` instead.