espressif/esp-zigbee-sdk #357: Add low power multitasking example.

diazmanuel, 2024-06-05:
### Is your feature request related to a problem?

Yes, I was seeing that there are some problems in multitasking (because you have to take the zigbee resources and block them if you want to use them from another task), Also, another example that I could not find is how to manage low consumption in multiple tasks, which is why an example could be made that involves both, this being the typical use case for moderately complex devices.

### Describe the solution you'd like.

an example that demonstrates how to use zigbee api calls from another task and keep the low power by sleeping the tasks accordingly. For example, I don't know if it is necessary to call the zigbee sleep function from external tasks, call the default function of the espressif sdk or simply put the task to sleep from a queue.
There being so many different resources to sleep a task, I would like to know which one is correct, this would be evident in the example I requested.

chshu, 2024-06-12:
@diazmanuel You can start from the [light sleep](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_sleep/light_sleep/main/esp_zb_sleepy_end_device.c) example, and add more tasks for any application logic, just pay attention that [zigbee-api-lock](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#zigbee-api-lock) is required to call Zigbee APIs from other tasks.

The Zigbee sleep mode is controlled by the Zigbee main task (main loop), you don't need to do anything in your application tasks.

diazmanuel, 2024-06-12:
hello @chshu ,
thanks for the suggestion

closed 2024-06-12 by diazmanuel