espressif/esp-zigbee-sdk #358: Connecting an ESP32-C6 with "Zigbee_Light_Bulb" example to the Philips Hue Bridge

m4nu-el, 2024-06-05:
### Question

Lately I was experimenting with a Waveshare ESP32-C6-DEV-KIT-N8 and zigbee communication. I got the communication working with another ESP32-C6 (loaded with `Zigbee_Light_Switch.ino` example). However I'd like to connect the ESP32-C6 with my Hue Bridge to later control a light connected to ESP32-C6. When I start searching for a new light with the hue bridge I get an `ZDO signal with ZDO Leave (0x3)`. As soon as the the Hue Bridge stops searching, this signal also disappeares (as shown in the logs).

`[   278][I][sketch_jun5d.ino:107] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[   288][I][sketch_jun5d.ino:75] esp_zb_app_signal_handler(): Zigbee stack initialized
[   297][I][sketch_jun5d.ino:81] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[   307][I][sketch_jun5d.ino:83] esp_zb_app_signal_handler(): Start network formation
[  2540][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[  6311][I][sketch_jun5d.ino:107] esp_zb_app_signal_handler(): ZDO signal: ZDO Leave (0x3), status: ESP_OK
[  6321][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 10093][I][sketch_jun5d.ino:107] esp_zb_app_signal_handler(): ZDO signal: ZDO Leave (0x3), status: ESP_OK
[ 10103][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 13875][I][sketch_jun5d.ino:107] esp_zb_app_signal_handler(): ZDO signal: ZDO Leave (0x3), status: ESP_OK
[ 13885][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 17719][I][sketch_jun5d.ino:107] esp_zb_app_signal_handler(): ZDO signal: ZDO Leave (0x3), status: ESP_OK
[ 17729][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 21498][I][sketch_jun5d.ino:107] esp_zb_app_signal_handler(): ZDO signal: ZDO Leave (0x3), status: ESP_OK
[ 21508][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 24744][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 27981][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 31218][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[ 34455][I][sketch_jun5d.ino:103] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)`

I already read about this problem in this issue: #168 . However I haven't understood how to optain that key. I found some blogs online where people tried to optain that key by sniffing the connection with wireshark like here: [https://www.hal9k.dk/sniffing-philips-hue-zigbee-traffic-with-wireshark/](url). I also found this resource [https://moh-computer.de/zigbee-zll-master-key-fuer-philips-hue/](url). With the key from the last resource I modified the `esp_zb_task()`.

```c++
static void esp_zb_task(void *pvParameters) {
  esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
  esp_zb_init(&zb_nwk_cfg);
  esp_zb_enable_joining_to_distributed(true); // not declared in scope
  uint8_t secret_zll_trust_center_key[] = { 0x9F, 0x55, 0x95, 0xF1, 0x02, 0x57, 0xC8, 0xA4, 0x69, 0xCB, 0xF4, 0x2B, 0xC9, 0x3F, 0xEE, 0x31 };
  esp_zb_secur_TC_standard_distributed_key_set(secret_zll_trust_center_key);
  esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
  esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
  esp_zb_device_register(esp_zb_on_off_light_ep);
  esp_zb_core_action_handler_register(zb_action_handler);
  esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
  //Erase NVRAM before creating connection to new Coordinator
  esp_zb_nvram_erase_at_start(true);  //Comment out this line to erase NVRAM data if you are conneting to new Coordinator
  ESP_ERROR_CHECK(esp_zb_start(false));
  esp_zb_main_loop_iteration();
}
``` 

However as commented in the code, the function `esp_zb_enable_joining_to_distributed(true)` is not declared in scope. I checked the docs here: components/esp-zigbee-lib/include/esp_zigbee_core.h and seems like the function is only declared when the constant `ZB_DISTRIBUTED_SECURITY_ON` is defined. But still with this constant defined in the `Zigbee_Light_Bulb.ino` the compiler throws the same error. 

To summarize I'd like to know what I'm doing wrong when trying to connect the ESP to the Hue Bridge.

xieqinan, 2024-06-06:
@m4nu-el ,

Which version of the library are you using in your project? I tested the `esp_zb_enable_joining_to_distributed(true);` API on my side with the esp-zigbee-sdk v1.3.2, and it worked as expected. The issue may be triggered by the use of the Arduino IDE 2.3.2 and the esp32 3.0.0 Board Package. Could you please confirm the version of the libraries you are using first?

MakMaciej, 2024-06-18:
have you figured something out? trying to and running the code but its till just displays "ZDO signal: ZDO Leave (0x3)"

m4nu-el, 2024-06-18:
Nope, unfortunately nothing new. @MakMaciej have you used the Arduino IDE or the esp-idf? Maybe anyone who contributed to the arduino implementation got an idea?

MakMaciej, 2024-06-20:
@m4nu-el I used the esp-idf.

Haerteleric, 2024-12-20:
you got the wrong key mate. check out this post :

https://blog.smartere.dk/2019/03/sniffing-philips-hue-zigbee-traffic-with-wireshark/

m4nu-el, 2024-12-25:
Thanks man for your answer, I already thought this issue is already dead. I checked out the post again and tried all the 3 different Key-Transport keys for the `uint8_t secret_zll_trust_center_key[]` but unfortunately no better result.

xieqinan, 2024-12-26:
Hi @m4nu-el,

I believe the issue is caused because `esp_zb_enable_joining_to_distributed(true);` cannot be called in Arduino IDE 2.3.2. Since the `Philips Hue Bridge` acts as a distributed router, this function is essential. Could you also submit an issue to [Arduino](https://github.com/espressif/arduino-esp32/issues) to confirm this? If the API is available, I think the code should work fine. Or combine the esp-zigbee-sdk and esp-idf to run the code.

MakMaciej, 2024-12-26:
> you got the wrong key mate. check out this post :
> 
> https://blog.smartere.dk/2019/03/sniffing-philips-hue-zigbee-traffic-with-wireshark/

so i need to get the Key-Transport Key? or what do you mean? use the first or third key?

m4nu-el, 2024-12-29:
With the use of ESP-IDF Extension for Visual Studio Code I was finally able to at least get successful network connection when searching for a new light in the Phillips Hue App.
```
I (456) main_task: Returned from app_main()
I (466) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (466) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (466) ESP_ZB_ON_OFF_LIGHT: Device started up in  factory-reset mode
I (476) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (2866) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (6836) ESP_ZB_ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: XX:XX:...:XX , PAN ID: 0xXXXXX, Channel:25, Short Address: 0xXXXX)
```

Furthermore after a quick reboot I get the message `ESP_ZB_ON_OFF_LIGHT: Device rebooted`. Am I still connected? I checked the code and turns out that it's produced by the `esp_zb_app_signal_handler`. The weird thing that the philips hue app stopped the searching animation as soon as my ESP established a connection but after a while it told me that no light was found. The list of my philips hue lights still doesn't contain my ESP.

I used the Example `HA_on_off_light` which can be found in the ESP-IDF Examples. ESP-IDF Version 5.2.1. The following code is the only modification I made to the code
```
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_enable_joining_to_distributed(true);

    // “default global trust center link key” which is 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39
    uint8_t secret_zll_trust_center_key[] = { 0x81, 0x42, 0x86, 0x86, 0x5D, 0xC1, 0xC8, 0xB2, 0xC8, 0xCB, 0xC5, 0x2E, 0x5D, 0x65, 0xD1, 0xB8 };
    //{ 0x81, 0x42, 0x86, 0x86, 0x5D, 0xC1, 0xC8, 0xB2, 0xC8, 0xCB, 0xC5, 0x2E, 0x5D, 0x65, 0xD1, 0xB8 };
    //{0x5A, 0x69, 0x67, 0x42, 0x65, 0x65, 0x41, 0x6C, 0x6C, 0x69, 0x61, 0x6E, 0x63, 0x65, 0x30, 0x39};
    
    //{ 0x9F, 0x55, 0x95, 0xF1, 0x02, 0x57, 0xC8, 0xA4, 0x69, 0xCB, 0xF4, 0x2B, 0xC9, 0x3F, 0xEE, 0x31 };
    esp_zb_secur_TC_standard_distributed_key_set(secret_zll_trust_center_key);

    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    esp_zb_device_register(esp_zb_on_off_light_ep);
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

Turns out that you need the light link commissioning key to join the network. Tomorrow I will submit an issue to [https://github.com/espressif/arduino-esp32/issues](url) regarding the implementation of that method.

xieqinan, 2024-12-30:
@Haerteleric,

> Am I still connected? 

Yes, you are.

> The weird thing that the philips hue app stopped the searching animation as soon as my ESP established a connection but after a while it told me that no light was found. 

Based on our testing, the issue may be caused by an incorrect `app_device_version`. The `esp_zb_ep_list_add_ep()` function provides a way to set the `app_device_version`. You can refer to [this example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_temperature_sensor/main/esp_zb_temperature_sensor.c#L159). Please try setting the `app_device_version` to 1.

m4nu-el, 2024-12-30:
I implemented the following after a bit of digging into the documentation.
```
static esp_zb_ep_list_t *custom_on_off_light_ep_create(uint8_t endpoint_id, esp_zb_on_off_light_cfg_t *light_cfg){
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create(); // Creating empty endpoint list

    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = endpoint_id,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID,
        .app_device_version = 1
    };

    esp_zb_attribute_list_t *attr_list = esp_zb_on_off_cluster_create(&(light_cfg->on_off_cfg));

    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_on_off_cluster(cluster_list, attr_list, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

    esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config);
    return ep_list;
}

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_enable_joining_to_distributed(true);

    // “default global trust center link key” which is 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39
    uint8_t secret_zll_trust_center_key[] = { 0x81, 0x42, 0x86, 0x86, 0x5D, 0xC1, 0xC8, 0xB2, 0xC8, 0xCB, 0xC5, 0x2E, 0x5D, 0x65, 0xD1, 0xB8 };
    //{ 0x81, 0x42, 0x86, 0x86, 0x5D, 0xC1, 0xC8, 0xB2, 0xC8, 0xCB, 0xC5, 0x2E, 0x5D, 0x65, 0xD1, 0xB8 };
    //{0x5A, 0x69, 0x67, 0x42, 0x65, 0x65, 0x41, 0x6C, 0x6C, 0x69, 0x61, 0x6E, 0x63, 0x65, 0x30, 0x39};
    
    //{ 0x9F, 0x55, 0x95, 0xF1, 0x02, 0x57, 0xC8, 0xA4, 0x69, 0xCB, 0xF4, 0x2B, 0xC9, 0x3F, 0xEE, 0x31 };
    esp_zb_secur_TC_standard_distributed_key_set(secret_zll_trust_center_key);

    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    
    esp_zb_ep_list_t *esp_zb_on_off_light_ep = custom_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    //esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    esp_zb_device_register(esp_zb_on_off_light_ep);
    
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
}
```
I also had to update the `idf_component.yml` to get the newest version of the zigbee lib because in the HA_on_off_light example on ESP-IDF v5.2.1 it uses an old version of that library( I don't remember exactly but i think something like 1.0.6). It now looks like this
```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zboss-lib: "~1.6.0"
  espressif/esp-zigbee-lib: "~1.6.0"
  espressif/led_strip: "~2.0.0"
  ## Required IDF version
  idf:
    version: ">=5.0.0"
```
When I tried it on the newest IDF version v5.3.2 it didn't even compile because it offers a newer version of the HA_on_off_light example where the library "zcl_utility.h" is not found in the `esp_zb_light.h`. But that's another issue. I just wanted to say it in the case someone wants to reproduce it with the new ESP-IDF.

m4nu-el, 2024-12-30:
I implemented the following after a bit of digging into the documentation.
```
static esp_zb_ep_list_t *custom_on_off_light_ep_create(uint8_t endpoint_id, esp_zb_on_off_light_cfg_t *light_cfg){
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create(); // Creating empty endpoint list

    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = endpoint_id,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID,
        .app_device_version = 1
    };

    esp_zb_attribute_list_t *attr_list = esp_zb_on_off_cluster_create(&(light_cfg->on_off_cfg));

    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_on_off_cluster(cluster_list, attr_list, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

    esp_zb_ep_list_add_ep(ep_list, cluster_list, endpoint_config);
    return ep_list;
}

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_enable_joining_to_distributed(true);

    // “default global trust center link key” which is 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39
    uint8_t secret_zll_trust_center_key[] = { 0x81, 0x42, 0x86, 0x86, 0x5D, 0xC1, 0xC8, 0xB2, 0xC8, 0xCB, 0xC5, 0x2E, 0x5D, 0x65, 0xD1, 0xB8 };
    //{ 0x81, 0x42, 0x86, 0x86, 0x5D, 0xC1, 0xC8, 0xB2, 0xC8, 0xCB, 0xC5, 0x2E, 0x5D, 0x65, 0xD1, 0xB8 };
    //{0x5A, 0x69, 0x67, 0x42, 0x65, 0x65, 0x41, 0x6C, 0x6C, 0x69, 0x61, 0x6E, 0x63, 0x65, 0x30, 0x39};
    
    //{ 0x9F, 0x55, 0x95, 0xF1, 0x02, 0x57, 0xC8, 0xA4, 0x69, 0xCB, 0xF4, 0x2B, 0xC9, 0x3F, 0xEE, 0x31 };
    esp_zb_secur_TC_standard_distributed_key_set(secret_zll_trust_center_key);

    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    
    esp_zb_ep_list_t *esp_zb_on_off_light_ep = custom_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    //esp_zb_ep_list_t *esp_zb_on_off_light_ep = esp_zb_on_off_light_ep_create(HA_ESP_LIGHT_ENDPOINT, &light_cfg);
    esp_zb_device_register(esp_zb_on_off_light_ep);
    
    esp_zb_core_action_handler_register(zb_action_handler);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
}
```
I also had to update the `idf_component.yml` to get the newest version of the zigbee lib because in the HA_on_off_light example on ESP-IDF v5.2.1 it uses an old version of that library( I don't remember exactly but i think something like 1.0.6). It now looks like this
```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zboss-lib: "~1.6.0"
  espressif/esp-zigbee-lib: "~1.6.0"
  espressif/led_strip: "~2.0.0"
  ## Required IDF version
  idf:
    version: ">=5.0.0"
```
When I tried it on the newest IDF version v5.3.2 it didn't even compile because it offers a newer version of the HA_on_off_light example where the library "zcl_utility.h" is not found in the `esp_zb_light.h`. But that's another issue. I just wanted to say it in the case someone wants to reproduce it with the new ESP-IDF.

But let's talk about the results
> Please try setting the `app_device_version` to 1.

Unfortunately this did not make a change to result. It again found the network, the Hue App was "freezing" but after aborting the scanning process no light is added to my bridge. Here you can see the logs
```
I (438) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (438) ESP_ZB_ON_OFF_LIGHT: Device started up in  factory-reset mode
I (448) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (2698) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (5948) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (9198) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (12978) ESP_ZB_ON_OFF_LIGHT: Joined network successfully (Extended PAN ID: XX:XX:...:XX, PAN ID: 0xXXXXX, Channel:25, Short Address: 0xXXXX)
I (63568) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (70558) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
I (77558) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
```

wejn, 2024-12-31:
So I have a similar problem... I can link up a esp32c6 to the philips-hue-bridge managed network by using `esp_zigbee_HA_sample/HA_color_dimmable_light` modded by enabling the `esp_zb_enable_joining_to_distributed()` with the `81:...:B8` key.

But then the "ESP32 lightbulb" isn't accepted as a lightbulb, whereas `Gledopto GL-C-008P` is discovered & configured instantly.

I've been poring over the differences using Wireshark, and there are two cases:

When I leave the light as `.esp_zb_role = ESP_ZB_DEVICE_TYPE_ROUTER` (which is the default), the process seems to break around `Route Reply`:

Gledopto:
[image]

ESP32C6:
[image]

When I switch the `.esp_zb_role = ESP_ZB_DEVICE_TYPE_ED,`, the whole process gets stuck elsewhere, this time around `Node descriptor response`, where the only meaningful difference I've found is the `Stack Compliance Revision` (`21` for gledopto, `22` for esp32):

Gledopto:
[image]

ESP32:
[image]

Now, I'm as far away from a Zigbee expert as you can throw a person... so I have no idea if what I'm saying has some merit. Because the ZB spec has ~600 pages (and another ~1200 for the ZCL), and it's not exactly a light read.

But tldr is -- I couldn't get the Philips Hue Bridge to accept an ESP32C6 (with the `esp_zigbee_HA_sample/HA_color_dimmable_light` example, and bunch of other people's lights, namely [ESP32H2-ZigbeeCCT](https://github.com/username-AL/ESP32H2-ZigbeeCCT)) as a lightbulb, and I've been several hours on it. And I've tried several versions of the SDK, I've tried even the "touchlink example" linked from #208, etc.

I'm happy to share the full wireshark dumps, but not here. Because network keys etc. So if you want them, hit me up via email.

Also happy to assist in further debugging, if you have specific task for me to try.

---

Btw, I'm slowly leaning towards abandoning ESP32 for this "lightbulb" task and going to nrf52 instead. Because this stuff is so hard to debug, and the esp-zigbee-sdk documentation is so minimalistic, that it feels too painful to make any meaningful progress (without becoming an expert in the field).

Also, one more aside -- having everything as a binary blob doesn't help either. What help is `ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 3` when I can't look up the source, so I have no idea what's going on. Is that success? Failure? Something in between? :-( )

wejn, 2025-01-01:
Holy karate in the air, I found the culprit. :-/

There's the [comment from xieqinan on #208 that says how Hue needs `.app_device_version = 1` for the endpoint](https://github.com/espressif/esp-zigbee-sdk/issues/208#issuecomment-2031038598).

Well, funny thing, that was exactly my hangup.

The default `HA_color_dimmable_light` is broken (can't be linked to Hue) because it uses `.app_device_version = 0`, and editing the example and reuploading seemed to do nothing, because the Hue bridge somehow caches info about devices (on a per-MAC basis) it encounters. Which explains the strange packet captures where the `gledopto` seemed to be interrogated to oblivion on the first try (and immediately admitted), whereas my `esp32c6` seemed to be ignored -- never even asked about its endpoints.

So I was running in circles not being able to link the esp32... until I flashed a different MAC to the device (using [the info in #155](https://github.com/espressif/esp-zigbee-sdk/issues/155#issuecomment-1825447177)), at which point the devboard was accepted to the Hue app just fine.

So the diff I needed for the `HA_color_dimmable_light` was roughly:

``` diff
diff --git a/main/esp_zb_light.c b/main/esp_zb_light.c
index 2a8d50c..d16c866 100644
--- a/main/esp_zb_light.c
+++ b/main/esp_zb_light.c
@@ -186,7 +186,15 @@ static void esp_zb_task(void *pvParameters)
     // esp_zb_nvram_erase_at_start(true); // FIXME: if you want the config killed at start...
 
     esp_zb_color_dimmable_light_cfg_t light_cfg = ESP_ZB_DEFAULT_COLOR_DIMMABLE_LIGHT_CONFIG();
-    esp_zb_ep_list_t *esp_zb_color_dimmable_light_ep = esp_zb_color_dimmable_light_ep_create(HA_COLOR_DIMMABLE_LIGHT_ENDPOINT, &light_cfg);
+    esp_zb_ep_list_t *esp_zb_color_dimmable_light_ep = NULL;
+    esp_zb_color_dimmable_light_ep = esp_zb_ep_list_create();
+    esp_zb_endpoint_config_t endpoint_config = {
+        .endpoint = HA_COLOR_DIMMABLE_LIGHT_ENDPOINT,
+        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
+        .app_device_id = 0x0102,
+        .app_device_version = 1, // maybe important for Hue? Oh HELL yes.
+    };
+    esp_zb_ep_list_add_ep(esp_zb_color_dimmable_light_ep, esp_zb_color_dimmable_light_clusters_create(&light_cfg), endpoint_config); 
     zcl_basic_manufacturer_info_t info = {
         .manufacturer_name = ESP_MANUFACTURER_NAME,
         .model_identifier = ESP_MODEL_IDENTIFIER,
```

And with that plus `python3 esp_zb_mfg_tool.py -m CAFEBEEF50C0FFE0 -c 0x07FFF800 -mn Espressif -mc 0x131B` and flash to the `zb_fct` region it works like a charm.

So, no touchlink and/or commissioning endpoint needed. The example is _almost_ functional. :-D

Haerteleric, 2025-01-06:
you have to implement the scene table yourself. you can adapt my boilterplate for that:

```c
static void storeEndpointStateToScene(esp_zb_zcl_store_scene_message_t * message)
{
    printf("Store Scene %i for Group %i\n", message->scene_id, message->group_id);
    flsEndpoint_t * ep = NULL;
    endpointManager_getEndpointById(message->info.dst_endpoint, &ep);

    bool colorEnabledEp = ((ep->capabilities.reg & ENDPOINT_CAPABILITIES_COLORTEMP) == ENDPOINT_CAPABILITIES_COLORTEMP)
        || ((ep->capabilities.reg & ENDPOINT_CAPABILITIES_RGB) == ENDPOINT_CAPABILITIES_RGB);

    onOffExtensionField_t onOffExtensionField = {0};
    levelControlExtensionField_t levelControlExtensionField = {0};
    colorControlExtensionField_t colorControlExtensionField = {0};

    esp_zb_zcl_scenes_extension_field_t onOffClusterData = {
        .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF,
        .length = sizeof(onOffExtensionField.raw),
        .extension_field_attribute_value_list = onOffExtensionField.raw,
        .next = NULL
    };

    esp_zb_zcl_scenes_extension_field_t levelControlClusterData = {
        .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_LEVEL_CONTROL,
        .length = sizeof(levelControlExtensionField.raw),
        .extension_field_attribute_value_list = levelControlExtensionField.raw,
        .next = &onOffClusterData
    };

    esp_zb_zcl_scenes_extension_field_t colorControlClusterData = {
        .cluster_id = ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
        .length = sizeof(colorControlExtensionField.raw),
        .extension_field_attribute_value_list = colorControlExtensionField.raw,
        .next = &levelControlClusterData
    };

    if(esp_zb_lock_acquire(portMAX_DELAY))
    {
        if(colorEnabledEp)    
        {
            uint8_t mode = *(uint8_t *)(esp_zb_zcl_get_attribute(
                    message->info.dst_endpoint, 
                    ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
                    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                    ESP_ZB_ZCL_ATTR_COLOR_CONTROL_COLOR_MODE_ID
                )->data_p);
            
            printf("Mode: %i\n", mode);

            switch (mode)
            {
            case 0x00:
                colorControlExtensionField.fields.hue = (*(uint8_t *)(esp_zb_zcl_get_attribute(
                    message->info.dst_endpoint, 
                    ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
                    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                    ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_HUE_ID
                )->data_p)) << 8; //Hue is 8 bit, so shift it to the left by 8 bits to get the Enhanced Hue

                colorControlExtensionField.fields.saturation = *(uint8_t *)(esp_zb_zcl_get_attribute(
                    message->info.dst_endpoint, 
                    ESP_ZB_ZCL_CLUSTER_ID_COLOR_CONTROL,
                    ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, 
                    ESP_ZB_ZCL_ATTR_COLOR_CONTROL_CURRENT_SATURATION_ID
                )->data_p);
                break;

            case 0x01:
                colorControlExtensionField.fields.x = *(uint16_t *)(esp_zb_zcl_get_attribute(
[... 190 lines cut ...]
        {
            esp_zb_zcl_store_scene_message_t * storeSceneMessage = (esp_zb_zcl_store_scene_message_t *)message;
            printf("Store Scene %i for Group %i\n", storeSceneMessage->scene_id, storeSceneMessage->group_id);
            
            // NVM WA
            // storeSceneData(storeSceneMessage->info.dst_endpoint, storeSceneMessage->scene_id, storeSceneMessage->group_id);

            // Stack Mechanism
            storeEndpointStateToScene(storeSceneMessage);
        }
        break;
    case ESP_ZB_CORE_SCENES_RECALL_SCENE_CB_ID:
        {
            esp_zb_zcl_recall_scene_message_t * recallSceneMessage = (esp_zb_zcl_recall_scene_message_t *)message;
            printf("Recall Scene %i for Group %i\n", recallSceneMessage->scene_id, recallSceneMessage->group_id);

            // NVM WA
            // restoreScene(recallSceneMessage->info.dst_endpoint, recallSceneMessage->scene_id, recallSceneMessage->group_id);

            // Stack Mechanism
            restoreEndpointToSceneState(recallSceneMessage);
        }
        break;
[...] 
```

wejn, 2025-01-06:
@Haerteleric that's amazing, thank you. Did you come up with it by trial & error, or was there any official docs you followed?

Also, would you be interested in collaborating on making a minimal yet fully fledged esp32 lightbulb example[^1]? I already started at https://github.com/wejn/esp32-huello-world/ but it's definitely missing scenes, and possibly more than that...

[^1]: We could take this conversation off-bugtracker, perhaps to email...

Haerteleric, 2025-01-07:
there are not really official docs (kind of a theme with this ESP-Component...) for that. I figured it out by reading header files, old issues and the Zigbee Spec.

sure hit me up at erh(at)dresden-elektronik.de

jaccobezemer, 2025-02-11:
Hi, I successfully compiled the Zigbee_Dimmable_Light.ino and got it working with Home Assistant ZHA.
Now I want to join it to my Philips HUE Bridge and have to change the secret_zll_trust_center_key[].
Silly question but in what file is the trust center key defined.