espressif/esp-zigbee-sdk #359: Route record

HOSTEER, 2024-06-06:
### Is your feature request related to a problem?

Not having access to the available devices, and their related path cost and relay count

### Describe the solution you'd like.

A get function to look out the route record data, table 3-59 Route Record Table Entry Format from rev 22 of Zigbee specification

### Describe alternatives you've considered.

use a Zigbee sniffer to guess what should be the data in the route record by looking the route request and reply frames

### Additional context.

When benchmarking the latency of the data transission across mesh network it's dificult to know the number of router a packet is passing through just by looking the Zigbee sniffer

xieqinan, 2024-06-12:
@HOSTEER ，

The [esp_zb_nwk_get_next_route()](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/nwk/esp_zigbee_nwk.h#L275) seems to meet your requirement.

HOSTEER, 2024-06-12:
The [esp_zb_nwk_get_next_route()](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/nwk/esp_zigbee_nwk.h#L275) is missing some variables of the route record. There is the destination adress for sure but the relay count and the path is missing.

closed 2024-07-19 by espressif-bot

xieqinan, 2024-07-19:
@HOSTEER,
The feature has been supported in esp-zigbee-sdk v1.4.1, refer to https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/nwk/esp_zigbee_nwk.h#L304

HOSTEER, 2024-07-29:
@xieqinan Thanks a lot !