espressif/esp-zigbee-sdk #360: add a method that returns the structure of the zigbee network and the quality of each of its links

diazmanuel, 2024-06-06:
### Is your feature request related to a problem?

I would like to be able to see the status of the network and the quality of each of the links to know if I should add repeaters or not.

### Describe the solution you'd like.

I would like you to add a method to be able to read the structure of the zigbee network, that is, a method that is called from the cordiador which returns a list of devices with the following data:

- Device Mac Adress
- Parent Mac Adress
- Device Type
- Link Quality with the parent (LQI) 

With this data I could construct how the network and the link of each of them is formed.

Suppose we have the following devices
Coordinator, Router A, Router B, Router C, Endpoint A, Endpoint B, Endpoint C, Endpoint D, Endpoint E, Endpoint F, Endpoint G, Endpoint H
With this method I hope to be able to obtain a structure like the following

[image]
In this example, Endpoint F would be outside the network

I was looking at the different methods found in the functions of the current SDK and some of the functions of the Zboss library; and I found some that might work
by the espressif SDK the esp_zb_nwk_get_next_neighbor() function.This function provides the necessary information but can only be used locally. A solution would be to make this function available in the format of an external request which takes the destination address as a parameter, thus allowing you to read the neighbor table of any device of the network (similar to the method used to read the bindings table externally esp_zb_zdo_binding_table_req()). 

On the other hand, I saw that ZBoss achieves what I'm looking for through the iterative implementation of zb_zdo_mgmt_lqi_req(), making this function available could be another solution, but the best way would be to create a new method which prevents the user from having to iterate and survey each of the devices manually

chshu, 2024-06-12:
It sounds a reasonable requirement, which is similar the Thread Network Diagnostics feature. We will evaluate and add the feature support.

diazmanuel, 2024-07-26:
hello @chshu, We are currently testing the new functions available in version 1.4.1, said function is esp_zb_nwk_get_next_route_record(), we understand that this function would be the one that would try to solve our necessity. Reading the documentation for this I saw that nothing is mentioned about the quality of the signal (lqi), anyway we try to implement this function but it always returns that it has no data, I would like to know if this function would be the one we are requesting or if they will implement another one later, I take this opportunity to ask if you have any update on this issue that you can share with us

lpy4105, 2024-07-29:
Hi @diazmanuel,

`esp_zb_nwk_get_next_route_record()` is used to get the "Routing Table" on the local device, which only contains the next hop information of routing. The appropriate method should be, as you said in the issue description, using `zdo_mgmt_lqi_req` to get "Neighbour Tables".

We do have plan to add some high level APIs to get network topology, but this would take some time. Before it is done, we can provide an API to do `zdo_mgmt_lqi_req` first, so that you can implement such functionality yourselves for time being.

diazmanuel, 2024-08-01:
hello @lpy4105 ,I would appreciate it if you could provide us an API to implement the zdo_mgmt_lqi_req method in the next update, anyway please leave this issue open until you add the high level APIs method

BaWe155, 2024-08-30:
This will be a good feature, what is the status?

diazmanuel, 2024-09-30:
Hello @lpy4105, do you have any update about this issue?.
 Are you currently implementing this method or has it been left for later?
 I would be interested in knowing a little about the timing because it is the last resource we need to be able to close some developments.

lpy4105, 2024-09-30:
Hi @diazmanuel an API for `zdo_mgmt_lqi_req` will be available in the next release.

diazmanuel, 2024-09-30:
hello @lpy4105, 
Thanks for the implementation of this method, anyway this function is not the development that @chshu mentions correct? I understand that this function is just the espressif implementation of the zboss function. As for the complete implementation of the high level API, is progress being made or do you plan to leave it only with the zdo_mgmt_lqi_req API?

xieqinan, 2024-10-21:
@diazmanuel ,

The esp-zigbee-sdk v1.5.1 already supports the `zdo_mgmt_lqi_req` feature, which you can refer to [here](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zdo/esp_zigbee_zdo_command.h#L576). In theory, the network topology structure can be obtained using `zdo_mgmt_lqi_req()` and `esp_zb_nwk_get_next_route_record()` in your application. The additional support for this feature has not been supported temporarily.

diazmanuel, 2024-12-02:
hello @xieqinan ,
Thanks for making the API available, we were indeed able to recreate the network with these functions, in any case it was somewhat cumbersome, I understand that your team is working on the implementation of a high-level method that resolves all the network state at the same time, instead of consult all devices 1 by 1. Could you tell me the status of this issue and the tentative date of this new method? thank you so much

xieqinan, 2025-03-19:
Hello @diazmanuel ,

>  I understand that your team is working on the implementation of a high-level method that resolves all the network state at the same time, instead of consult all devices 1 by 1.

Regarding this issue, after evaluation, we have decided to abandon the development of the high-level method, as it is not a necessary feature for the SDK. The `esp_zdo_mgmt_lqi_req()`, `esp_zb_nwk_get_next_route_record()`, and `esp_zb_nwk_get_next_neighbor()` APIs already provide users with the capability to obtain the complete network topology.  

However, I believe there are some misconceptions. The network state must be queried from each device individually rather than being retrieved all at once. Therefore, the high-level method will not be supported in the SDK, as it functions the same way as using the above APIs in your application to obtain the network topology.