espressif/esp-zigbee-sdk #361: Using wifi and zigbee at the same time

Andelarrr, 2024-06-09:
### Question

Hello dear Developers,

We are working on a device that needs to connect to the zigbee network and at the same time it needs to connect on wifi, in both cases it is a slave/client/endpoint. The device needs to keep its context on the zigbee network, while active on wifi. Being unable to communicate on zigbee during this time is ok, as long as it can continue after finishing the wifi transfer.

Is this possible using one ESP32-C6 or maybe another ESP32 module?

We have seen solutions using 2 ESP32 devices, this makes sense, but it increases costs, so we will use it only if it is not possible with one ESP32.

We have seen as well the "coexistence" feature, we tried esp_coex_wifi_i154_enable(), but we had some issues. Is this the way to go? Is the coexistence working between wifi and zigbee?

Alternatively if would be fine if we could do something like this every time we need to use the wifi:
zigbee_pause()
wifi_connect()
// Communicate...
wifi_disconnect()
zigbee_resume()
Provided that the zigbee context would remain and no network reconnect would be requried.

Please advise on the best way forward, I would really appreciate some further documentation on the topic as well, if available.

Thanks!
Andelar

xieqinan, 2024-06-11:
@Andelarrr ,

The ESP32-C6 supports WiFi and Zigbee coexistence. You can try using the [gateway example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway).

Andelarrr, 2024-06-11:
Hi @xieqinan, thanks, if I understand correctly this gateway example is not about the coexistence mechanism, this is the solution with 2 ESP32 devices, one for zigbee and one for wifi.
Let me know if I'm mistaken or there is another example for the coexistence or any other approach.

chshu, 2024-06-12:
Since there is only one RF path in ESP32-C6, which means Wi-Fi and Zigbee can't receive simultaneously, it has a significant impact on performance. Hence the two SoCs solution is recommended.