espressif/esp-zigbee-sdk #362: Active scan do not work at all

rickou, 2024-06-09:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### esp-zigbee-lib version.

3.2.0

### esp-zboss-lib version.

3.2.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

I would like to do an active scan to get all PAN around me.
i only add the command to the HA_on_off_light example.
`
void active_scan_cb(esp_zb_zdp_status_t zdo_status, uint8_t count, esp_zb_network_descriptor_t *nwk_descriptor)
{
	ESP_LOGI(TAG, "Active scan");
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    switch (sig_type) {
    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        ESP_LOGI(TAG, "Zigbee stack initialized");
//        esp_zb_zdo_active_scan_request(ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK,10,active_scan_cb);
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Start network steering");
            esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
        } else {
            /* commissioning failed */
            ESP_LOGW(TAG, "Failed to initialize Zigbee stack (status: %s)", esp_err_to_name(err_status));
        }
        break;
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status == ESP_OK) {
            esp_zb_ieee_addr_t extended_pan_id;
            esp_zb_get_extended_pan_id(extended_pan_id);
            ESP_LOGI(TAG, "Joined network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d)",
                     extended_pan_id[7], extended_pan_id[6], extended_pan_id[5], extended_pan_id[4],
                     extended_pan_id[3], extended_pan_id[2], extended_pan_id[1], extended_pan_id[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel());
        } else {
            ESP_LOGI(TAG, "Network steering was not successful (status: %s)", esp_err_to_name(err_status));
    esp_zb_zdo_active_scan_request(ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK,200,active_scan_cb);
//            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 1000);
        }
        break;
    default:
        ESP_LOGI(TAG, "ZDO signal: %s (0x%x), status: %s", esp_zb_zdo_signal_to_string(sig_type), sig_type,
                 esp_err_to_name(err_status));
        break;
    }
}
`

### What is the actual behavior?

I (607) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (617) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 5 comm status 0
I (637) ESP_ZB_ON_OFF_LIGHT: Start network steering
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
I (657) main_task: Returned from app_main()
ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 15 - drop it
ZBOSS: zdo/zdo_commissioning.c:210   Can't find PAN to join to! param 25
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 3
I (2907) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
W (2917) ESP_ZIGBEE_API_ZDO: active scanned network failed!
ZBOSS: nwk/nwk_main.c:2336   Drop packet with improper protocol version: 0x5

i get active scan failure..
i tryed to call this function at different locations, different scan time, without success..
For sure i have 3 Zigbee networks around me.

i read the docs, humm... absolutly nothing writen on it...
no source code, so impossible to debug what's happen..
no examples
nothing on internet about active scan with ESP Zigbee lib..

i need help... 

### Steps to reproduce.

just open Espressif IDE, create a new project from HA_on_off_light
and modify it to add the function 'esp_zb_zdo_active_scan_request'

`I (607) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (617) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 5 comm status 0
I (637) ESP_ZB_ON_OFF_LIGHT: Start network steering
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
I (657) main_task: Returned from app_main()
ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 15 - drop it
ZBOSS: zdo/zdo_commissioning.c:210   Can't find PAN to join to! param 25
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 3
I (2907) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
W (2917) ESP_ZIGBEE_API_ZDO: active scanned network failed!
ZBOSS: nwk/nwk_main.c:2336   Drop packet with improper protocol version: 0x5
`

xieqinan, 2024-06-11:
@rickou 

The issue may be triggered by the `esp_zb_zdo_active_scan_request(ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK, 200, active_scan_cb);`. If the `scan_duration` parameter is set to 200, according to the specification, the wait time for the scan to finish is `(2^200 + 1) * beacon_time` * all_channel_number milliseconds, which is excessively long. Could you please try using `esp_zb_zdo_active_scan_request(ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK, 3, active_scan_cb);` instead?

rickou, 2024-06-15:
hi, 
I tried what you recommanded.
I got it to work and find only 1 PANID

Also tried to set scan_duration to 5 and 4 without success.. 
back to 3, and now i get again "ESP_ZIGBEE_API_ZDO: active scanned network failed!"

it is very random.. and again, no clues about reason of failure..

rickou, 2024-06-15:
it seems the scan fails when no panid found.
but how to know the failure at application side ?
i have not found any callbacks or function result ?

it seems also the function only found networks with "permit join" enabled..
how to list all networks around ?

xieqinan, 2024-06-17:
@rickou 

> it seems the scan fails when no panid found.
> but how to know the failure at application side ?

It seems to be a bug. The callback for scanning failure will be provided in the next release.

> it seems also the function only found networks with "permit join" enabled..
> how to list all networks around ?

The `esp_zb_zdo_active_scan_request()` should scan all networks around. It is not limited by the stack. Could you please double-check it?

rickou, 2024-06-17:
> > it seems also the function only found networks with "permit join" enabled..
> > how to list all networks around ?
> 
> The `esp_zb_zdo_active_scan_request()` should scan all networks around. It is not limited by the stack. Could you please double-check it?

i made a simple test, exactly same environment, ESP same place and don't move, so radio conditions are exactly the same
Around me, i have at least 3 panIds
1 network could be joined anytime (always open to join)
2 networks can't be joined (need to activate join)

in these conditions, result is:
```
scan
I (230131) VRGATEWAY: Start active scan (timeout=3)
esp32c6> I (232381) VRGATEWAY: Active scan result 0 count: 1
I (232381) VRGATEWAY:   PANID 47:42:30:30:30:31:31:37 (0xdeba) join permited
```

if i extend scan timeout (same result)
```
scan 5
I (565911) VRGATEWAY: Start active scan (timeout=5)
esp32c6> I (574071) VRGATEWAY: Active scan result 0 count: 1
I (574071) VRGATEWAY:   PANID 47:42:30:30:30:31:31:37 (0xdeba) join permited

```

if i open 1 of the 2 other networks
```
esp32c6> scan
I (655521) VRGATEWAY: Start active scan (timeout=3)
esp32c6> I (663691) VRGATEWAY: Active scan result 0 count: 2
I (663691) VRGATEWAY:   PANID 47:42:30:30:30:31:31:37 (0xdeba) join permited
I (663691) VRGATEWAY:   PANID 18:11:23:17:00:00:00:00 (0x2317) join permited
```

here is the code
```
void active_scan_cb(esp_zb_zdp_status_t zdo_status, uint8_t count, esp_zb_network_descriptor_t *nwk_descriptor)
{
	ESP_LOGI(TAG, "Active scan result %d count: %d",zdo_status, count);
	for (int i=0;i<count;i++)
	{
		ESP_LOGI(TAG, "  PANID %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x (0x%04x) join %s",
						nwk_descriptor[i].extended_pan_id[7],nwk_descriptor[i].extended_pan_id[6],
						nwk_descriptor[i].extended_pan_id[5],nwk_descriptor[i].extended_pan_id[4],
						nwk_descriptor[i].extended_pan_id[3],nwk_descriptor[i].extended_pan_id[2],
						nwk_descriptor[i].extended_pan_id[1],nwk_descriptor[i].extended_pan_id[0],
						nwk_descriptor[i].short_pan_id,(nwk_descriptor[i].permit_joining)?"permited":"disabled");
		
	}
}

static int do_activescan(int argc, char **argv)
{
	int nerrors = arg_parse(argc, argv, (void **) &scan_args);
    if (nerrors != 0) {
        arg_print_errors(stderr, scan_args.end, argv[0]);
        return 1;
    }
    int timeout=3;
    if (scan_args.timeout->count)
    {
		timeout=*scan_args.timeout->ival;
	}
    ESP_LOGI(TAG, "Start active scan (timeout=%d)",
    	timeout);
	if (esp_zb_lock_acquire(100))
	{
	    esp_zb_zdo_active_scan_request(ESP_ZB_TRANSCEIVER_ALL_CHANNELS_MASK,timeout,active_scan_cb);
		esp_zb_lock_release();
	}
    return 0;
}

```

i never seen a network with join disabled

rickou, 2024-06-18:
hi again, 
i just made a test at same location but with another usb dongle (EZSP) with bellows on python.

```
bellows -b 57600 -d COM5 scan -t 100
Scanning channels 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26
[EmberZigbeeNetwork(channel=11, panId=0x2317, extendedPanId=18:11:23:17:00:00:00:00, allowingJoin=<Bool.false: 0>, stackProfile=2, nwkUpdateId=0), 35, -73]
[EmberZigbeeNetwork(channel=11, panId=0x2317, extendedPanId=18:11:23:17:00:00:00:00, allowingJoin=<Bool.false: 0>, stackProfile=2, nwkUpdateId=0), 194, -68]
[EmberZigbeeNetwork(channel=11, panId=0x2317, extendedPanId=18:11:23:17:00:00:00:00, allowingJoin=<Bool.false: 0>, stackProfile=2, nwkUpdateId=0), 251, -63]
[EmberZigbeeNetwork(channel=14, panId=0xDEBA, extendedPanId=47:42:30:30:30:31:31:37, allowingJoin=<Bool.true: 1>, stackProfile=2, nwkUpdateId=0), 255, -67]
```

i can found 2 networks (i don't know why the 1st one appears 3 times, sometimes only 2 times)
as you can see, RSSI levels are quite similar, and 1 network do not allow join, the 2nd one permit it..

I confirm, i suspect something is wrong in the ESP ZB lib

Eric.

xieqinan, 2024-06-18:
@rickou ，

Alright, thanks for your test, I will double-check the implementation of `esp_zb_zdo_active_scan_request()` .

xppt, 2024-07-02:
@xieqinan, hi!
Is it also possible to add `void *user_ctx` transparent argument for both `esp_zb_zdo_active_scan_request` and its callback?

closed 2024-07-04 by espressif-bot

xieqinan, 2024-07-04:
@rickou,

The `esp_zb_zdo_active_scan_request()` already has the capability to scan closed networks in esp-zigbee-sdk v1.4.0. Could you update to this version and test again? If you have any other questions, please reopen this issue.

rickou, 2024-07-06:
Yes it works now !
Is there any changelog somewhere ?

xieqinan, 2024-07-24:
@rickou ，

The change log can refer to https://github.com/espressif/esp-zigbee-sdk/blob/main/RELEASE_NOTES.md#features-1