espressif/esp-zigbee-sdk #363: ESP32C6 does not join, as it cannot find PAN to join for HA_on_off_light example

matthiasbuettner159, 2024-06-10:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### esp-zigbee-lib version.

1.3.2

### esp-zboss-lib version.

1.3.2

### Espressif SoC revision.

ESP32C6

### What is the expected behavior?

I expect it to connect to my zigbee2mqtt coorinator without issue.

### What is the actual behavior?

After starting the Zigbee Stack it start the steering process, which fails every time with the following debug logs:

```
I (1469) ESP_ZB_ON_OFF_LIGHT: Start network steering
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:349   dev type 2, joined 0, ext_pan_id 0, authenticated 0, tclk_valid 1
ZBOSS: zdo/zdo_commissioning.c:210   Can't find PAN to join to! param 25
ZBOSS: commissioning/bdb/zdo_commissioning_bdb.c:549   COMMISSIONING_STOP: app signal 10 comm status 3
I (3739) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
```

### Steps to reproduce.

1. Load HA_on_off_example
2. Update dependency-versions to match most recent ones
3. Flash to device
4. Monitor

### More Information.

This is what I can see in Wireshark when sniffing the traffic.
[image]
It seems like there is a response with the Extended PAN ID but I am not sure if it is received on the ESP. Leaving the ESP running these lines appear over and over again: Beacon Request -> Zigbee Response with EPID -> repeat.

As I am very new to using Zigbee on an ESP I am not sure how to proceed from here and how to get the device to pair with my coordinator?

I am using zigbee2mqtt with a network attached Zigstar UZG-1 as coordinator device. zigbee2mqtt is setup to use channel 25 as well as a network encryption key. All other purchased devices (e.g. from Aqara) are joining without issue in the current setup. Therefore I assume the issue lies within some missing or incorrect configuration on the ESP part. 

Thanks a lot for your input :)

xieqinan, 2024-06-11:
@matthiasbuettner159 ,

The issue is likely not a bug but rather a device pairing problem. Could you please attach the Wireshark packet and network key for me? I need to understand the exact network pairing process.

xieqinan, 2024-06-11:
By the way, esp-idf v5.1.3 is recommended for use with esp-zigbee-sdk v1.3.2.

matthiasbuettner159, 2024-06-11:
Hey,

thanks already for your responses. I also tried with esp-idf v.5.1.3 and the same issue occurs. I tried using versions 5.1.3, 5.1.4, 5.2.2 and 5.3. All with the same problem.

Here is the wireshark capture: [Pairing-Error.zip](https://github.com/user-attachments/files/15783761/Pairing-Error.zip)
The network key is: 15:25:3E:41:A9:8F:16:76:20:83:57:AB:08:08:CE:EA

Additionally here are the logs I receive in VSCode: [VSCode-Terminal-logs.txt](https://github.com/user-attachments/files/15783792/VSCode-Terminal-logs.txt)

There is one other sensor in my test zigbee network which might have sent some data during the capture - just to have mentioned it.

xieqinan, 2024-06-11:
@matthiasbuettner159 ,

Based on the Wireshark packet provided by you, the issue appears to be triggered by the "Association Permit: False" of Beacon. This indicates that the network of the coordinator "MyHome43" is closed. As a result the Zigbee device can not find an available network to join.

[image]

matthiasbuettner159, 2024-06-11:
@xieqinan ,
thank you very much. As mentioned I am using zigbee2mqtt as broker and there I have enabled the setting to always allow new devices to join. And I am puzzled why it is not working in this scenario, as I can join other devices without problems.
Do you have a suggestion what I could try? Otherwise I will reach out to the zigbee2mqtt community :)

xieqinan, 2024-06-11:
@matthiasbuettner159 

I have found the network created by the `57:e6:39:8c:d9:70:ca:97` device is open, it may related with the another limitation for device joining is the low Link Quality Indicator (LQI) value set in the SDK. If the link between the coordinator and the end device is weak, the end device may reject choosing it as its parent. By default, the smallest LQI value for network joining is 32. You can use the `esp_zb_secur_network_min_join_lqi_set()` function to set a lower LQI threshold.

matthiasbuettner159, 2024-06-11:
@xieqinan ,
I also just checked again and saw that the MyHome43 is not my network but the one you mentioned is. Here I also see the Association Permit as True.
I will try to lower the threshold, but I would not understand why the LQI is that low, as I am trying to connect basically right next to my coordinator. Is there a way to see the LQI during the joining process?

closed 2024-06-11 by matthiasbuettner159

BaWe155, 2024-08-06:
@xieqinan @matthiasbuettner159 

Is the problem with the low lqi solved? 
I use:
  espressif/esp-zboss-lib: "=1.1.2"
  espressif/esp-zigbee-lib: "=1.1.2"
  espressif/led_strip: "=2.0.0"
  ## Required IDF version
  idf:
    version: ">=5.1.3"

ss89, 2024-10-13:
For those who also struggled with this:
I had to go super close (less than 1 meter) to the coordinator to successfully pair it.
I also set in the code before the "zigbee stack initialization" message: `esp_zb_secur_network_min_join_lqi_set(0);`, however I do not know and did not test if that made a difference.

xieqinan, 2024-10-21:
@BaWe155 ,

The version you used is old, could you please test again with the esp-zigbee-sdk v1.5.1?

xieqinan, 2024-10-21:
@ss89 ,

Could you please provide more details on how to reproduce your issue? I tried running the `HA_color_dimmable_light` and `HA_color_dimmable_switch` examples from the `esp-zigbee-sdk` master branch on an ESP32-H2 without any changes, and they were able to form a network even at distances greater than 5 meters.

bechhansen, 2025-02-02:
Hi @ss89 

I'm playing around with the examples and trying to connect an ESP32C6 running HA_temperature_sensor to my Zigbee coordinator and I have the same issues as you do.

My Zigbee coordinator is a Sonoff ZB-Dongle-E.

I'm only able to create a connection from a distance of around 1 meter, which is absolutely useless for an application like this.
I have other devices connected to the coordinator from a much larger distance.

/Peer

ss89, 2025-02-02:
Not really a cause, but having more zigbee routers seems to help.
So far i don't have too many zigbee devices and they seem more flaky then regular wifi devices (which already isn't working too reliable in my home recently).
I did a few weeks ago test that the esp could actually get into the network via the ikea lightbulb acting as a router.
Maybe also flash the most recent coordinator firmware (there was a recent release for 7.4.4.0 or so).