espressif/esp-zigbee-sdk #365: Defining custom zigbee cluster and attribute for RFID reader

amarteaux, 2024-06-10:
### Question

I want to create a custom attribute to display rfid card_id into a zigbee attribute with esp32-H2.

Currently i am blocked on the definition of the custom attribute.
I only want a string attribute were i could store badge id and notify home assistant .

Still no luck with that :cry: . Would be really thankful if someone could tell me what i am going wrong :pray: .

here my cluster creation:
```c
 static void esp_zb_task(void *pvParameters) {
  /* initialize Zigbee stack */
  esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
  esp_zb_init(&zb_nwk_cfg);

  // Cluster BASIC
  esp_zb_basic_cluster_cfg_t basic_cluster_cfg = {
    .zcl_version = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE,
    .power_source = 0x03
  };

  uint32_t ApplicationVersion = 0x0001;
  uint32_t StackVersion = 0x0002;
  uint32_t HWVersion = 0x0002;
  DEFINE_PSTRING(ManufacturerName, "sadManufacturer");
  DEFINE_PSTRING(ModelIdentifier, "AlarmPad");
  DEFINE_PSTRING(DateCode, "20240609");

  esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_basic_cluster_create(&basic_cluster_cfg);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_APPLICATION_VERSION_ID, &ApplicationVersion);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_STACK_VERSION_ID, &StackVersion);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_HW_VERSION_ID, &HWVersion);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, (void *)&ManufacturerName);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, (void *)&ModelIdentifier);
  esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_DATE_CODE_ID, (void *)&DateCode);

  // ------------------------------ Cluster IDENTIFY ------------------------------
  esp_zb_identify_cluster_cfg_t identify_cluster_cfg = {
      .identify_time = 0,
  };
  esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_identify_cluster_create(&identify_cluster_cfg);

  // ------------------------------ Cluster RFID ------------------------------
  esp_zb_attribute_list_t *esp_zb_rfid_cluster = esp_zb_zcl_attr_list_create(LSL_CLUSTER_ID); // use id not used for another cluster
  const uint16_t attr_id = 0;
  const uint8_t attr_type = ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING;
  const uint8_t attr_access = ESP_ZB_ZCL_ATTR_ACCESS_REPORTING | ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE;
  ESP_ERROR_CHECK(esp_zb_custom_cluster_add_custom_attr(esp_zb_rfid_cluster, attr_id, attr_type, attr_access, "4TOTO"));
  // ------------------------------ Create cluster list ------------------------------
  esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_rfid_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));

  // ------------------------------ Create endpoint list ------------------------------
  esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
  esp_zb_endpoint_config_t endpoint_config = {
      .endpoint = LSL_ENDPOINT,
      .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
      .app_device_id = ESP_ZB_HA_CUSTOM_ATTR_DEVICE_ID,
  };

   ESP_ERROR_CHECK(esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, endpoint_config));

  // ------------------------------ Register Device ------------------------------
  esp_zb_device_register(esp_zb_ep_list);
  esp_zb_core_action_handler_register(zb_action_handler);
  esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

  ESP_ERROR_CHECK(esp_zb_start(false));
  esp_zb_main_loop_iteration();
}
```

xieqinan, 2024-06-11:
@amarteaux，

There are two potential issues that may be causing the failure in your project:

- The `LSL_CLUSTER_ID` should be greater than 0x8000, please double-check it.
- The `esp_zb_custom_cluster_add_custom_attr(esp_zb_rfid_cluster, attr_id, attr_type, attr_access, "4TOTO")` might be the root of the issue. If you intend to set the string size to 4, I recommend using `esp_zb_custom_cluster_add_custom_attr(esp_zb_rfid_cluster, attr_id, attr_type, attr_access, "\x04""TOTO")`.

chshu, 2024-07-26:
@amarteaux Hope the comment https://github.com/espressif/esp-zigbee-sdk/issues/365#issuecomment-2159899653 helped, feel free to reopen if follow up questions.

closed 2024-07-26 by chshu

mundevx, 2024-12-18:
I have same issue.

I am using this:
    esp_zb_attribute_list_t *esp_zb_custom_cluster = esp_zb_zcl_attr_list_create(0xFC00);
    esp_zb_custom_cluster_add_custom_attr(esp_zb_custom_cluster, 0, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &dst_addr);
    esp_zb_cluster_list_add_custom_cluster(cluster_list, esp_zb_custom_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

and home assistant not showing attribute (0).

[image]

mundevx, 2024-12-19:
hi @xieqinan,

Any update.

mundevx, 2024-12-21:
The LSL_CLUSTER_ID should be greater than 0x8000, please double-check it.
The esp_zb_custom_cluster_add_custom_attr(esp_zb_rfid_cluster, attr_id, attr_type, attr_access, "4TOTO") might be the root of the issue. If you intend to set the string size to 4, I recommend using esp_zb_custom_cluster_add_custom_attr(esp_zb_rfid_cluster, attr_id, attr_type, attr_access, "\x04""TOTO").    

These doesn't work for me.