espressif/esp-zigbee-sdk #366: Assert failed when changing the IAS Zone cluster zone status (0x0002) attribute value

Vprime007, 2024-06-11:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### esp-zigbee-lib version.

1.3.2

### esp-zboss-lib version.

1.3.2

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

When i want to change the IAS Zone cluster zone status (0x0002) attribute, my program crash on an assert in the zcl/zcl_ias_zone_commands.c file. The expected behavior is that the attribut value change without crashing the program.

### What is the actual behavior?

When i try to set the IAS Zone Status attribute value (0x0002) with the following function:    

esp_zb_zcl_status_t ret = esp_zb_zcl_set_attribute_val(1,
                                                           ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE,
                                                           ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
                                                           ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID,
                                                           (void*)&status,
                                                           false);

the program crash with the following message:

[image]

I im interrested to know what cause the assert to failed and what is related to this assert. Is there specific steps required to change the zone status attribute value?

### Steps to reproduce.

1. Create an attribtue list for the IAS Zone cluster :
[image]

2. Add Zone status attribute to the list:
[image]

3. Add the IAS Zone cluster to the cluster list:
[image]

4. Add clusters list to the endpoint list and register the device:
[image]

5. Start the Zigbee stack :
[image]

6. Try to change the IAS Zone - zone status attribute value :
[image]

xieqinan, 2024-06-12:
@Vprime007,

The issue is triggered by a missing attribute. You can call `esp_zb_ias_zone_cluster_create()` to create the IAS zone cluster, which includes the mandatory attributes of IAS zone.

Vprime007, 2024-06-13:
Thank you for the advice. My issue is fixed.

closed 2024-06-13 by Vprime007