espressif/esp-zigbee-sdk #368: esp_zb_af_simple_desc_1_1_t definition not correct

rickou, 2024-06-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### esp-zigbee-lib version.

3.2.0

### esp-zboss-lib version.

3.2.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

the definition of esp_zb_af_simple_desc_1_1_t is not correct.
typedef struct esp_zb_af_simple_desc_1_1_s {
    uint8_t    endpoint;                        /*!< Endpoint */
    uint16_t   app_profile_id;                  /*!< Application profile identifier */
    uint16_t   app_device_id;                   /*!< Application device identifier */
    uint32_t   app_device_version: 4;           /*!< Application device version */
    uint32_t   reserved: 4;                     /*!< Reserved */
    uint8_t    app_input_cluster_count;         /*!< Application input cluster count */
    uint8_t    app_output_cluster_count;        /*!< Application output cluster count */
    uint16_t   app_cluster_list[2];             /*!< Application input and output cluster list */
} ESP_ZB_PACKED_STRUCT
esp_zb_af_simple_desc_1_1_t;

### What is the actual behavior?

app_cluster_list is only 2 cells..

### Steps to reproduce.

it should be a list or something similar.
best should be to have 2 separate lists for in and out clusters lists

xieqinan, 2024-06-17:
@rickou ,

I don't think it is a bug. The `app_cluster_list` carries `app_input_cluster_count + app_output_cluster_count` cells in the stack. You can find the `app_input_cluster_count` at the beginning of the array, and the remaining cells are for `app_output_cluster_count`.

rickou, 2024-06-17:
I understand well the concept and you're right, it not really a bug but a mis-coded definition

only the definition is not good.
`uint16_t app_cluster_list[2]; /!< Application input and output cluster list */`

should be
`uint16_t* app_cluster_list; /!< Application input and output cluster list */`

The cluster order is still not clear (in then out, or out then in..)
so, for me **the best** should be to have a definition like this:
```
uint16_t* app_cluster_in_list; /!< Application input cluster list */
uint16_t* app_cluster_out_list; /!< Application output cluster list */
```
it doesn't change anything on tthe behavior, but it is more clear.

Eric

xieqinan, 2024-07-22:
@rickou ,

Thanks for your suggestion. I think it’s a great idea. However, it involves a breaking change, so we will consider implementing it in the next major version.

Alfff, 2025-08-04:
@xieqinan 

I’ve faced the same issue.

It doesn't make sense for something that is essentially a pointer to an array of size `app_input_cluster_count + app_output_cluster_count` to be declared as a fixed array of two elements.

What makes it even more confusing is the fact that there are other structures where the data is organized in exactly the same way — for example, `esp_zb_zdo_match_desc_req_param_t` explicitly uses a pointer named `cluster_list`, whose size is `num_in_clusters + num_out_clusters`.

I believe only if you’ve previously worked with structures like that one can you assume that `app_cluster_list` is actually a pointer to a dynamically sized array. Otherwise, it's very unclear how you're supposed to use it.

Alfff, 2025-08-05:
I believe the proposed definitions are clearer :)