espressif/esp-zigbee-sdk #369: Support for the poll controll cluster

Vprime007, 2024-06-17:
### Is your feature request related to a problem?

The device i am working on require the Zigbee Poll Controll cluster (0x0020) but it seem that it is not supported yet by the zigbee stack (v1.3.2). 

### Additional context.

Do you have a timeline for the support of the standard Zigbee clusters? Also, is there an alternative way to implement the unsupported clusters ?

xieqinan, 2024-07-24:
@Vprime007 ，

We plan to support the poll control cluster in the future, so please be patient as we work on it.

theorlangur, 2025-01-18:
+1 for poll control cluster
would be nice to have the support

xieqinan, 2025-01-20:
@theorlangur ,

> can I implement the support for it via a custom cluster in the meantime?

No, the latest sdk (v1.6.2) does not allow to use the custom cluster to define the standard cluster.

xieqinan, 2025-07-28:
The Poll Control cluster has already been supported in esp-zigbee-sdk v1.6.6, see [here](https://github.com/espressif/esp-zigbee-sdk/commit/f0cbffef8d6ab140b08a38fa67592649562c8d9d#diff-569f6b58cae5dabea24155a1a3cbc4bc12ce7ea2b793100799b4ceab4af560a7). Please update to this version and test again.

Let's close this discussion. If you encounter any other related issues, feel free to reopen it.

closed 2025-07-28 by xieqinan