espressif/esp-zigbee-sdk #371: Error when running esp_zb_mfg_tool.py

mjdswan, 2024-06-22:
### Question

On attempting to run the esp_zb_mfg_tool.py using the given example command lines, I get "unsupported operand type(s)" errors.

I am using esp-zigbee-sdk/tools 1.3.0, ESP-IDF 5.2 on Windows 10.

Please suggest what might be the issue.

### Additional context.

PS C:\Users\mswan\.espressif\Espressif\esp-zigbee-sdk\tools\mfg_tool> python esp_zb_mfg_tool.py -i NULL -m NULL -c 0x8000 -mn Espressif -mc 0x131B
unsupported operand type(s) for +: 'NoneType' and 'str'
PS C:\Users\mswan\.espressif\Espressif\esp-zigbee-sdk\tools\mfg_tool> python esp_zb_mfg_tool.py --csv ./single_device.csv
unsupported operand type(s) for +: 'NoneType' and 'str'
PS C:\Users\mswan\.espressif\Espressif\esp-zigbee-sdk\tools\mfg_tool>

xieqinan, 2024-06-24:
@mjdswan ，

Please refer to this issue [https://github.com/espressif/esp-zigbee-sdk/issues/237#issuecomment-1925671397](https://github.com/espressif/esp-zigbee-sdk/issues/237#issuecomment-1925671397), which is similar to the one you encountered.

mjdswan, 2024-06-24:
Hi @xieqinan, yes I saw there is a similar question there (last post) but it is not answered.  Please explain?

xieqinan, 2024-06-24:
@mjdswan ,

The issue is caused by an invalid `+` operation between a `NoneType data` and a `string`. Could you help us locate the specific line where this occurs?

mjdswan, 2024-06-24:
Hi @xieqinan, I managed to trace the Python file through to the error at line 415 in **esp_zb_mfg_tool.py**.
It seems that the function
**mfg_gen.get_fileid_val(file_identifier, key_value_data, fileid_value)**
returns a value of None and this is why the + operation fails.

I'm not really sure why this happens, hope you can please assist.

mjdswan, 2024-06-27:
Hi @xieqinan, do you have any suggestion?  Sorry but I'm quite new to Python and I'm not really sure what the above function is supposed to return.

Some background on this: we need to use the tool to create the ZB Mfr name and ID because we need to integrate our ESP32C6 based host device with a SONOFF ZB hub.  I can see from sniffing that this hub requests our mfr name but we respond with "unsupported attribute".  This response unfortunately causes the SONOFF hub to reboot, which makes testing impossible because we are stuck in a loop.

I am assuming that ESP ZigBee SDK responds like this because the mfr name is empty.
The response in this case would appear to be incorrect, and I would think it's a better approach to respond with a default value of "Espressif" to avoid these kinds of problems - I can open a separate issue on that if you think it's needed.
By the way, if there is some other way to create the mfr name and ID directly (without the tool), please let me know.

Would appreciate your update soonest, so we can continue our development and testing thanks.

mjdswan, 2024-07-03:
Hi @xieqinan,

So I finally managed to make `esp_zb_mfg_tool.py` work by editing line 447 from
`csv_values_file = open(file_name, 'w')`
to
`csv_values_file = open(file_name, 'w', newline='\n')`

It seems that the utility generates a temporary CSV file, and by default creates blank rows due to the good old line termination issue, which results in the above error when it tries to read the rows back.
I still get this error: `ERROR:root:C:\Users\mswan\.espressif\Espressif\esp-zigbee-sdk\tools\mfg_tool\single_device.csv`
for some reason, but the bin file is created and the script ends normally.

IMO, this tool could use some improvement to make it work the way it should on all environments.

closed 2024-07-03 by mjdswan

xieqinan, 2024-07-15:
@mjdswan ,

Thanks for your test and solution. We will improve the tool to make it work in more environments.

xieqinan, 2025-04-15:
@JCperezesp 

> Currently i'm still getting this unsupported operand type(s) for +: 'NoneType' and 'str' error while running the tool on Windows 11. Editing the line as @mjdswan showed solved the problem.

What command triggered this error? I suspect the issue might be caused by differences in Python versions across operating systems. The complete Python code is available in `esp_zb_mfg_tool.py`—could you help debug and pinpoint where exactly the error occurs?

JCperezesp, 2025-04-25:
The error was triggered in line 446 from `esp_zb_mfg_tool.py`. The line was

```
csv_values_file = open(file_name, 'w')
```         
I changed it to be as following:
```
csv_values_file = open(file_name, 'w', newline='\n')
```

The command that triggered the error was `python esp_zb_mfg_tool.py`. My goal was to build a binary partition for an esp32-c6 module in order to change its install code.

Yes, the issue was caused by an incompatibility between Windows and Linux, in relation to EOL characters, Windows uses two characters to indicate the end of a line (CRLF) whilst GNU/Linux only uses one (LF).

xieqinan, 2025-04-27:
> The command that triggered the error was python esp_zb_mfg_tool.py. My goal was to build a binary partition for an esp32-c6 module in order to change its install code.

Thanks for your test. If your goal cannot be achieved by running `esp_zb_mfg_tool.py`, please submit another issue to describe your problem (since this issue is closed and not directly related). We will do our best to resolve your problem.