espressif/esp-zigbee-sdk #373: Bidirectional Communication

kgkask, 2024-06-24:
### Question

### How to make the Endpoint write attribute to the coordinator to add a bidirectional communication ability? 

### Additional context.

I am using the code in #244, I wanted to edit it to make both ESP32H2 send string messages. Form the documentation I found that  I have to set an extra cluster for each sides to add this feature, I added the clusters in both sides and I expected to work just fine I have updated the codes to handle the inquiries for each side, however when the end point send the write commands the coordinator does react at all NOT SURE WHAT'S HAPPENING?!

I have tried multiple examples and similar opened issues but I couldn't figure out the source of the issue.

Coordinator node :
`#ifndef ZIGBEE_MODE_ZCZR
#error "Zigbee coordinator mode is not selected in Tools->Zigbee mode"
#endif

#include "esp_zigbee_core.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"

/* Switch configuration */
#define GPIO_INPUT_IO_TOGGLE_SWITCH GPIO_NUM_9
#define PAIR_SIZE(TYPE_STR_PAIR)    (sizeof(TYPE_STR_PAIR) / sizeof(TYPE_STR_PAIR[0]))

#define CUSTOM_CLUSTER_ID 0xFC00
#define CUSTOM_CLUSTER_ID1 0xFC01

#define CUSTOM_STRING_MAX_SIZE 127
#define LED_PIN RGB_BUILTIN

typedef enum {
  SWITCH_ON_CONTROL,
  SWITCH_OFF_CONTROL,
  SWITCH_ONOFF_TOGGLE_CONTROL,
  SWITCH_LEVEL_UP_CONTROL,
  SWITCH_LEVEL_DOWN_CONTROL,
  SWITCH_LEVEL_CYCLE_CONTROL,
  SWITCH_COLOR_CONTROL,
} switch_func_t;

typedef enum {
  SWITCH_IDLE,
  SWITCH_PRESS_ARMED,
  SWITCH_PRESS_DETECTED,
  SWITCH_PRESSED,
  SWITCH_RELEASE_DETECTED,
} switch_state_t;

typedef struct {
  uint8_t pin;
  switch_func_t func;
} switch_func_pair_t;

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

typedef struct light_bulb_device_params_s {
  esp_zb_ieee_addr_t ieee_addr;
  uint8_t endpoint;
  uint16_t short_addr;
} light_bulb_device_params_t;

/* Default Coordinator config */
#define ESP_ZB_ZC_CONFIG()                                                                                        \
  {                                                                                                               \
    .esp_zb_role = ESP_ZB_DEVICE_TYPE_COORDINATOR,                                                                \
      .install_code_policy = INSTALLCODE_POLICY_ENABLE,                                                           \
      .nwk_cfg = {                                                                                                \
      .zczr_cfg =                                                                                                 \
[... 819 lines cut ...]
  if (xQueueReceive(gpio_evt_queue, &button_func_pair, portMAX_DELAY)) {
    pin = button_func_pair.pin;
    switch_gpios_intr_enabled(false);
    evt_flag = true;
  }
  while (evt_flag) {
    bool value = digitalRead(pin);
    switch (switch_state) {
      case SWITCH_IDLE:           switch_state = (value == LOW) ? SWITCH_PRESS_DETECTED : SWITCH_IDLE; break;
      case SWITCH_PRESS_DETECTED: switch_state = (value == LOW) ? SWITCH_PRESS_DETECTED : SWITCH_RELEASE_DETECTED; break;
      case SWITCH_RELEASE_DETECTED:
        switch_state = SWITCH_IDLE;
        /* callback to button_handler */
        (*esp_zb_buttons_handler)(&button_func_pair);
        break;
      default: break;
    }
    if (switch_state == SWITCH_IDLE) {
      switch_gpios_intr_enabled(true);
      evt_flag = false;
      break;
    }
    vTaskDelay(10 / portTICK_PERIOD_MS);
  }
}`

xieqinan, 2024-06-24:
@kgkask ,

Could you please refer to this comment https://github.com/espressif/esp-zigbee-sdk/issues/202#issuecomment-2005690377 for your question? The example demonstrates a bidirectional communication case.

Additionally, it would be helpful to attach a patch or a zip file of your project to facilitate our review.

kgkask, 2024-06-24:
@xieqinan 
Thanks for your quick response, I actually I have used the same code but in Arduino IDE. It allows sending the string from one side not both so based on my understanding I have to add a client cluster with the server cluster to achieve bidirectional communication (and vice versa).
  

 sorry for the code. here is .rar file:

[bidirectional _communication.zip](https://github.com/user-attachments/files/15953226/bidirectional._communication.zip)

kgkask, 2024-06-26:
@xieqinan 

I have looked closely to the code, I can receive the response to the write command but when I configure the End device to send write command "`esp_zb_zcl_write_attr_cmd_req(&write_req);`" like the coordinator it fails

xieqinan, 2024-07-04:
@kgkask ，

Do you have any other questions that need to be discussed here? If the `bidirectional communication` issue has been resolved, please consider closing it.

closed 2024-07-04 by kgkask