espressif/esp-zigbee-sdk #374: HA_temperature_sensor - fails to join network (ConBee II) - with fix

OOHehir, 2024-06-24:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-beta2

### esp-zigbee-lib version.

1.3.2

### esp-zboss-lib version.

1.3.2

### Espressif SoC revision.

ESP32-H6

### What is the expected behavior?

HA_temperature_sensor sample would be discoverable & fully join to an ConBee II coordinator

### What is the actual behavior?

Failed to join ConBee II coordinator

### Steps to reproduce.

1. Build HA_temperature_sensor sample
2. Try to join ConBee II network
3. Notice it fails to join correctly

### More Information.

I believe this failure occurs because [this line](https://github.com/espressif/esp-zigbee-sdk/blob/6d288fbc32894ac8a85a9071c3b9d2b83c900946/examples/esp_zigbee_HA_sample/HA_temperature_sensor/main/esp_zb_temperature_sensor.h#L32) should be:

```
#define MODEL_IDENTIFIER                "\x07"CONFIG_IDF_TARGET
```
As CONFIG_IDF_TARGET when using an esp32c6 is 7 characters. 

Changing this allows the ESP32 to join the ConBee II network.

Further to this issue however, even if the ESP32 successfully joins the network I note it is not fully recognized. I had this issue with the HA_on_off_switch sample also, but following the discussion [here](https://github.com/espressif/esp-zigbee-sdk/issues/328#issuecomment-2146056071) I was able to get the ESP32 to be recognized & controllable from the Phoscon App. 

I realize there is an issue in the Zigbee ecosystem of interoperability but has anyone been able to get this sample fully recognized by a Zigbee coordinator (other than another ESP32)?

Many thanks

[image]

OOHehir, 2024-06-24:
Further to this I tried mimicking a [temperature sensor](https://dresden-elektronik.github.io/deconz-rest-doc/devices/computime-salus/ss909zb_temp_sensor/) that is known to the ConBee II but without success. I was using:

```
// @ref: https://dresden-elektronik.github.io/deconz-rest-doc/devices/computime-salus/ss909zb_temp_sensor/
char modelid[] = {9, 'C', 'O', 'M', 'P', 'U', 'T', 'I', 'M', 'E'};
char manufname[] = {7, 'S', 'S', '9', '0', '9', 'Z', 'B'};
```

I see in the [attributes](https://dresden-elektronik.github.io/deconz-rest-doc/devices/computime-salus/ss909zb_temp_sensor/) a number that are not implemented by the sample code & I wonder if this is the issue?

xieqinan, 2024-07-04:
@OOHehir ，

Based on the above description, it appears that the failure is related to the string content of `model_identifier`. If the following code is used, can the ESP32H2 be recognized by the ConBee II?
```
char modelid[] = {9, 'H', 'L', 'Q', 'D', 'Q', '0', '1', 'L', 'M'};
char manufname[] = {9, 'E', 'S', 'P', 'R', 'E', 'S', 'S', 'I', 'F''};
```

OOHehir, 2024-07-05:
@xieqinan Thanks for the response. I've updated to commit 81ade242 of the SDK. I erased the ESP32C6 & attempted to ass the ESP32C6 using the code you supplied. Unfortunately no improvement, the device joins the network:

```
I (166398) ESP_ZB_TEMP_SENSOR: Joined network successfully (Extended PAN ID: 00:21:2e:ff:ff:09:d2:20, PAN ID: 0x4b51, Channel:15, Short Address: 0xc787)
```
But seems to not to be fully recognized (Not visible in the Phoscon App). I've a Sonoff temperature sensor & notice that they have a 'Power Configuration' cluster. Perhaps that is required to 'fully join' the network?

[image]

xieqinan, 2024-07-15:
@OOHehir,

I believe this is an ecological compatibility issue. We need to determine how the ConBee II recognizes the device. Is it based on the clusters residing on the device or the `manufacturer code` and `model identifier`? Could you try using an ESP32C6 device to simulate the `Sonoff temperature sensor & notice` device, including the carried clusters, manufacturer code, and model identifier, and then enable the device to join the ConBee II network? If the esp-zigbee-sdk lacks some APIs to accomplish this, please feel free to let us know.

MoBaFanTorsten, 2025-02-19:
I had exactly the same problem, using Conbee II and an ESP32C6 microcontroller. Reading as well elsewhere, I came to the conclusion that deCONZ recognize a Zigbee device, but does not know how to treat it. This is indicated by deCONZ app in showing the device with the hexadecimal presentation of the MacID (I believe), instead of a proper name. This results into phoscon not showing and finding the sensor device.
The issue can be solved by creating a customized DDF, when performing a right click on the device box (in the graph given above the box showing "0xC787", one can select the option "Edit DDF". This opens the DDF editor. Here, one shall create a dedicated DDF for the device envisaged, which may also have any manufacturer ID and model ID one wishes, as long as in the basic cluster those attributes are set identically with the once set in the DDF editor. I understand those two form the key for detection. By the way, details on how to use the editor and how to configure the device can be found at Dresden electronics, see 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Important to know is that not only the "items" section must be used to configure this for deCONZ unknown new device, but also the "bindings" is to be used in order to configure the automatic reading of the temperature. The above given url addresses at the buttom as well how this is done.
The only thing which is not - yet - totally clear for me is the exact meaning of "Min interval", "Max interval" & "Reportable change". I assume here the min and max times between the reading of the device is to be configured (in seconds ?), as well as the smallest change of value. Nonetheless, the fields must be set in order for deCONZ to function.
If the DDF is then saved, I understand that after a restart of the deCONZ service the DDF is read and then deCONZ / phoscon understand what to do with the device.
Hope that helps :-)

OOHehir, 2025-02-21:
@MoBaFanTorsten @xieqinan Thanks for the help, as you say this is more like an interaction of the Conbee & ESP32-C6 devices rather than an 'issue'. I'll close this issue.

closed 2025-02-21 by OOHehir