espressif/esp-zigbee-sdk #375: Poor connections to Home Assistant Yellow

blaubaer, 2024-06-27:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-1205-ge7070e777a

### esp-zigbee-lib version.

1.3.2

### esp-zboss-lib version.

1.3.2

### Espressif SoC revision.

ESP32C6 and ESP32H2

### What is the expected behavior?

Be able to join a Zigbee network reliable

### What is the actual behavior?

Lot of luck to join a Zigbee network. See usually only the following outout:
```
BESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1804
load:0x4086c110,len:0xe74
load:0x4086e610,len:0x2e7c
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4-dev-1205-ge7070e777a 2nd stage bootloader
I (24) boot: compile time Jun 27 2024 16:31:16
I (24) boot: chip revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 000e1000
I (81) boot:  3 zb_storage       Unknown data     01 81 000f1000 00004000
I (88) boot:  4 zb_fct           Unknown data     01 81 000f5000 00000400
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=420a0020 size=32c64h (207972) map
I (152) esp_image: segment 1: paddr=00042c8c vaddr=40800000 size=0d38ch ( 54156) load
I (165) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=99294h (627348) map
I (295) esp_image: segment 3: paddr=000e92bc vaddr=4080d38c size=0281ch ( 10268) load
I (298) esp_image: segment 4: paddr=000ebae0 vaddr=4080fbb0 size=02434h (  9268) load
I (307) boot: Loaded app from partition at offset 0x10000
I (307) boot: Disabling RNG early entropy source...
I (324) cpu_start: Unicore app
I (333) cpu_start: Pro cpu start user code
I (333) cpu_start: cpu freq: 160000000 Hz
I (334) app_init: Application information:
I (336) app_init: Project name:     on_off_light_bulb
I (342) app_init: App version:      6d288fb-dirty
I (347) app_init: Compile time:     Jun 27 2024 16:30:56
I (353) app_init: ELF file SHA256:  XXXXXXXXXX...
I (358) app_init: ESP-IDF:          v5.4-dev-1205-ge7070e777a
[... 79 lines cut ...]
   > * [Seeed Studio XIAO ESP32C6](https://wiki.seeedstudio.com/xiao_esp32c6_getting_started/) with internal and external antenna (See pin `14`)
2. Have a [Home Assistant Yellow](https://www.home-assistant.io/yellow/) in the regular configuration with Zigbee enabled. This also means NO multiprotocol support - Zigbee only.
3. Checkout `esp-zigbee-sdk` in revision [`#4383011`](https://github.com/espressif/esp-zigbee-sdk/tree/438301125bdfa70150622d905094f79315456774).
4. Go to [`examples/esp_zigbee_HA_sample/HA_on_off_light`](https://github.com/espressif/esp-zigbee-sdk/tree/438301125bdfa70150622d905094f79315456774/examples/esp_zigbee_HA_sample/HA_on_off_light).
5. Run:
   ```bash
   idf.py set-target esp32c6
   ```
6. Enable Zigbee debugging of ESP32 with:
   ```bash
   idf.py menuconfig
   ```
6. Start searching for new Zigbee devices in Home Assistant (Default: http://homeassistant.local:8123/config/zha/add).
8. Run:
   ```bash
   idf.py -p <PORT> erase-flash build flash monitor
   ```

### More Information.

1. This was tested with several devices.
2. If the Home Assistant Yellow is in range of 50cm to 1m it always works.
3. If another Zigbee device (tested with [Philips LOM007 Hue Smart Plug](https://zigbee.blakadder.com/Philips_LOM007.html)) is in mode _Router_ is in range of 50cm to 1m it always works. Also if the distance between Home Assistant Yellow and the ESP32 is more than 10m.
4. If another Zigbee device (tested with [Philips LOM007 Hue Smart Plug](https://zigbee.blakadder.com/Philips_LOM007.html)) is in mode _Router_ is 5m away it does also not work.
5. Another Zigbee End Device (tested with [Philips RWL021 Hue Dimmer Switch](https://zigbee.blakadder.com/Philips_RWL021.html)) which is in the same location as the ESP32 this device always works. Also if the ESP32C6 fails to work.

lpy4105, 2024-07-04:
Could you please provide the `pcap` file if you have a sniffer?

BTW, did all the three ESP boards fail to work?

xieqinan, 2024-07-15:
@blaubaer ,
> This was tested with several devices.
> If the Home Assistant Yellow is in range of 50cm to 1m it always works.
> If another Zigbee device (tested with [Philips LOM007 Hue Smart Plug](https://zigbee.blakadder.com/Philips_LOM007.html)) >is in mode Router is in range of 50cm to 1m it always works. Also if the distance between Home Assistant Yellow and the ESP32 is more than 10m.
> If another Zigbee device (tested with [Philips LOM007 Hue Smart Plug](https://zigbee.blakadder.com/Philips_LOM007.html)) is in mode Router is 5m away it does also not work.
> Another Zigbee End Device (tested with [Philips RWL021 Hue Dimmer Switch](https://zigbee.blakadder.com/Philips_RWL021.html)) which is in the same location as the ESP32 this device always works. Also > if the ESP32C6 fails to work. Many other devices teste and all are working as expected.

According to the above description, the issue seems to be related to the energy of joining the network. The ESP device will determines whether to join the network based on the network's LQI. But the threshold value for joining LQI can be set using the `esp_zb_secur_network_min_join_lqi_set()` API. The default joining LQI threshold is 32. Could you set it to a smaller value and test it again?

blaubaer, 2024-07-15:
Hi @xieqinan

In steps of ~5 I lowered it down to `10`. There the connection works reliable, also in the same ranges as the Philips devices.

I would recommend to either change the default value or at least improve the documentation (for example the [FAQ section](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/faq.html)).

xieqinan, 2024-07-15:
@blaubaer ,

> I would recommend to either change the default value or at least improve the documentation (for example the [FAQ section](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/faq.html)).

No problem, I will consider adding the minimum joining LQI issue to the FAQ. Thanks for your testing and suggestion.