espressif/esp-zigbee-sdk #376: hardtime pairing esp32c6 and esp32h2  to hubitat (IDFGH-13158)

mukeshbauskar, 2024-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to pair esp32 dev kit to my hubitat using zigbee. I am trying to flash the devkit with [this example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_light).

I tried devkit with c6 and h2 controllers and also hubitat c7 and c8 as well.

Please let me know what logs do you need and how to gather it to help me out.

blaubaer, 2024-06-28:
Sounds similar to #375

mukeshbauskar, 2024-06-28:
@blaubaer I do not think they are similar. In my case, the hub and the devkit are literally 5cm apart. 

And here are logs from the monitor.
[zigbee_logs.txt](https://github.com/user-attachments/files/16027446/zigbee_logs.txt)

blaubaer, 2024-06-28:
Ok. You're right. If they are that close to each other, it sounds differently.

mukeshbauskar, 2024-06-28:
Here logs with zigbee debugging turned on.
[zigbee_debug_logs.txt](https://github.com/user-attachments/files/16029792/zigbee_debug_logs.txt)

This might be caused because of error
ZBOSS: mac/mac.c:1039   Oops - error beacon payload is too small buf_sz 15 - drop it

mukeshbauskar, 2024-07-07:
@igrr I have not heard from anyone about this.

xieqinan, 2024-07-08:
@mukeshbauskar ,

Regarding this issue, it appears to be triggered by a mismatched TC link key between `hubitat` and the `on_off_light` example of the esp-zigbee-sdk. You can refer to the log below:

```
ZBOSS: secur/apsme_secur.c:653   received link key identical to provisional link key - drop packet and fail tclk exchange
```

The default TC link key for the esp-zigbee-sdk example is `5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39` ("ZigbeeAlliance09"). However, you can call the [esp_zb_secur_TC_standard_preconfigure_key_set()](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_secur.h#L125) function to configure the TC link key to match the `hubitat`.

mukeshbauskar, 2024-07-08:
@xieqinan Do you know how to find TC link key for hubitat. I looked at zigbee settings on hubitat it only has PAN id.

mukeshbauskar, 2024-07-16:
I finally got it working. The issue was with Hubitat. It has problem working with zigbee 2.0 devices. There is  a workaround mentioned [here](https://community.hubitat.com/t/2-3-9-157-zigbee-plugs-are-found-but-do-not-function/139919/34). And it has been working smoothly. Thank you.

closed 2024-07-16 by mukeshbauskar