espressif/esp-zigbee-sdk #377: Several difficulties to associate zigbee device from other manufacturers

nalves23, 2024-06-28:
### Question

Hi.

I am developing a zigbee communication system made up of coordinator-type devices (Zigbee gateway), routers and end devices, all of which use an ESP32-H2 and I am using the espressif zigbee-SDK and the ZBoss library, both in version 1.3.2.
The ESP-IDF I am using is the stable version v5.2.2.
When associating the devices that I am developing and using the ESP32-H2, the association process is carried out quickly and without problems.
When I try to associate other ZigBee devices that use the manufacturer Texas Instruments, for example and its ZigBee stack, the association process almost always fails, with greater difficulty when the ZigBee network is created above channel 18.
Attached below is a Wireshark sniffing generated from the ZigBee network, in this case, operating on channel 19: 

[image]

I Send the wireshark log file attached:
[Zigbee TI association issues.zip](https://github.com/user-attachments/files/16026200/Zigbee.TI.association.issues.zip)

If anyone has had the same problem or if there is any way to mitigate it in the network coordinator, I would be very grateful. This process of associating devices from other manufacturers is very important for my project

Kind regards.
nalves23

lpy4105, 2024-07-12:
Hi @nalves23 
From the pcap file, I saw that esp didn't response for the association request in time. But finally when it responded, the TI failed to accept that?

Could you please also attach the log with trace enabled on the ESP side? The log of TI side is also helpful to narrow down the problem.

BTW, what is success rate of the association of your case?

xieqinan, 2025-08-13:
Hi @nalves23 ,

Since we have not received any response from you for a long time, we will close this issue. If the problem still persists, please try updating the esp-zigbee-sdk to the latest version. If it still fails, you can reopen it and refer to https://github.com/espressif/esp-zigbee-sdk/issues/377#issuecomment-2224353486 to provide us with more detailed information. Thank you.