espressif/esp-zigbee-sdk #379: "include/zcl/esp_zigbee_zcl_thermostat.h"

petermeter69, 2024-06-29:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.4+

### esp-zigbee-lib version.

1.0.1

### esp-zboss-lib version.

1.0.1

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

I am coming from the Arduino IDE, but I was advised to report this here.
As opposed to the other 3 examples, I was not able to compile the 4th default example (zigbee_thermostat).

### What is the actual behavior?

This is the error:
`In file included 
from C:\Users\.../include/espressif__esp-zigbee-lib/include/zcl/esp_zigbee_zcl_common.h:38,
from C:\Users\.../include/esp_zigbee_attribute.h:14,
from C:\Users\.../include/esp_zigbee_core.h:16,
from C:\Users\...\Zigbee_Thermostat.ino:33:
        C:\Users\...\Zigbee_Thermostat.ino: In function 'void esp_zb_task(void*)':
C:\Users\.../include/zcl/esp_zigbee_zcl_thermostat.h:180:63: error: narrowing conversion of '65535' from 'int' to 'int16_t' {aka 'short int'} [-Wnarrowing]
  180 | #define ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE 0xffff
      |                                                               ^~~~~~
C:\Users\.../include/ha/esp_zigbee_ha_standard.h:329:38: note: in expansion of macro 'ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE'
  329 |                 .local_temperature = ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE,                         \
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\...\Zigbee_Thermostat.ino:479:44: note: in expansion of macro 'ESP_ZB_DEFAULT_THERMOSTAT_CONFIG'
  479 |   esp_zb_thermostat_cfg_t thermostat_cfg = ESP_ZB_DEFAULT_THERMOSTAT_CONFIG();
      |                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1`

### Steps to reproduce.

Pressing compile button in IDE.

### More Information.

By changing this line in "include/zcl/esp_zigbee_zcl_thermostat.h", it compiles:
From:
`#define ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE 0xffff`
To:
`#define ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE 0x7fff`

lpy4105, 2024-07-10:
Hi @petermeter69 

I failed to reproduce the issue on Linux (tested by gcc/clang/arduino-esp). It seems to be an issue related to the compiler behaviour. 

On the other hand, the ZCL specification defines the default value of `local temperature` attribute to be `FF`. It's not correct to change the value to `0x7fff`.

Does https://github.com/espressif/arduino-esp32/issues/9951#issuecomment-2197432987 works for you?

damian-kurek-wizzdev, 2024-12-30:
Just change 0xFFFF when it is assigned to int16_t to -0x7fff
e.g. #define ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE -0x7fff

xieqinan, 2024-12-30:
@damian-kurek-wizzdev @petermeter69 ,

Could you please change `#define ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE` from `0xffff` to `ESP_ZB_ZCL_VALUE_FF` for a try?

damian-kurek-wizzdev, 2024-12-30:
@xieqinan It compiles with ESP_ZB_ZCL_VALUE_FF

closed 2025-02-08 by chshu

damian-kurek-wizzdev, 2025-02-10:
@chshu Will this be fixed in the next release? Do you plan to do it soon?