espressif/esp-zigbee-sdk #380: Why is device registration needed for sending commands?

Nanowires, 2024-06-29:
### Question

I've debugged my code for three days, while trying to set the level of my tradfri IKEA lamp via an ESP32c6, which served as coordinator. Starting from the Off/on switch example, I tried to use the esp_zb_zcl_level_move_to_level_cmd_req() method, which lead to an assert failure, with not much more than a stack trace for debugging.
As mentioned above, before opening a question on this repository, I tried to get the root of the problem by myself, and finally I've found that I have to register the correct device in my code:
Instead of registering the result of esp_zb_on_off_switch_ep_create() I needed the esp_zb_color_dimmable_switch_ep_create() result.

Now my questions:

+ Why do I have to register the device in the first place? Is it demanded by the standard, that the sending device (the coordinator in this case) must have a matching cluster?
+ If the above is the case, wouldn't it make sense to register the cluster somehow while calling the esp_zb_zcl_level_move_to_level_cmd_req(), before crashing? (but maybe this "somehow" is the tricky part and may be difficult)
+ Then last thing: Why not at least mention, why the assert failed? Would it be much more effort to print a fatal log like "you don't have registered the matching cluster/device for this command" before crashing?

xieqinan, 2024-07-17:
@Nanowires ，

> Why do I have to register the device in the first place? Is it demanded by the standard, that the sending device (the coordinator in this case) must have a matching cluster?

The reason is that the stack does not currently support the dynamic device data model. You need to call `esp_zb_device_register()` to register the data model with the stack. The `esp_zb_on_off_switch_ep_create()` function only handles the creation process. As a result, `esp_zb_device_register()` is necessary for any device in the current version. 

The esp-zigbee-sdk v1.4.0 provides a gateway endpoint mechanism for sending commands without matching clusters, which may be helpful for you. You can call `esp_zb_ep_list_add_gateway_ep(ep_list, NULL, endpoint_config)` and `esp_zb_device_register(ep_list)` to implement it.

> If the above is the case, wouldn't it make sense to register the cluster somehow while calling the esp_zb_zcl_level_move_to_level_cmd_req(), before crashing? 

The `esp_zb_zcl_level_move_to_level_cmd_req()` can be sent successfully after `esp_zb_device_register()` is called. However, it is recommended to register the clusters once before calling `esp_zb_start()`.

> Why not at least mention, why the assert failed? Would it be much more effort to print a fatal log like "you don't have registered the matching cluster/device for this command" before crashing?

This is the next point we need to optimize. We will provide more explicit debug information in subsequent versions. For now, you can enable the "Zigbee Debug Mode" in the menuconfig; it may be helpful for your debugging.

Nanowires, 2024-07-24:
@xieqinan: Thanks for the reply and the clarification. I'm looking forward for further versions of the SDK 😀

closed 2024-07-24 by Nanowires