espressif/esp-zigbee-sdk #382: Support for PH and EC into ZCL

DanielJimenezHernandez, 2024-07-09:
### Question

is there any plan in  the near future to implement some of the ZCL revision 8? or is there a way to implement it myself? 

### Additional context.

i have requirement in a project for soil measurements and need those clusters implemented

DanielJimenezHernandez, 2024-07-09:
the measurements tat are a must are PH and EC (electrical conductivity)

lpy4105, 2024-07-10:
Thanks for you suggestion. We will plan to add these two clusters.

closed 2024-08-28 by espressif-bot

xieqinan, 2024-08-28:
@DanielJimenezHernandez ,

These two clusters have been supported in esp-zigbee-sdk v1.5.0, please refer to https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_ec_measurement.h and https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_ph_measurement.h