espressif/esp-zigbee-sdk #384: Zigbee stack assertion failed when I send  alevel command for devices with lightbulb and windows covering characteristics.

nalves23, 2024-07-15:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.2

### esp-zigbee-lib version.

1.3.2

### esp-zboss-lib version.

1.3.2

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

When I try to send level commands to devices with lightbulb brightness and windows covering characteristics, with any values (0 to 255) the code crash and give a panic Zigbee stack assertion failed. It was supposed to send a zigbee oficial level command and the lightbulb move the brightness to 50%.

### What is the actual behavior?

When I try to send level commands to devices with lighbulb brightness and windows covering characteristics, with any values (0 to 255) the code crash and give a panic Zigbee stack assertion failed.

Below, the panic output that I see on the monitor window:

[image]

The instruction that I use to change the level is:

[image]

where payload is the uint8 value that send the level of we want to brightness light.

We try to use one of the level functions (esp_zb_zcl_level_move_to_level_cmd_req and esp_zb_zcl_level_move_to_level_with_onoff_cmd_req) but with any success. The panic happens in both function calls!

Any update to this issue?

Kind regards.
Nalves

### Steps to reproduce.

Described above!

xieqinan, 2024-07-17:
@nalves23 

Could you show me the code for registering the level control cluster data model? The esp-zigbee-sdk v1.3.2 does not support sending level control commands with the level control client. However, this feature is supported in v1.4.0. You can upgrade to the latest version and test it again.

nalves23, 2024-07-17:
Hi @xieqinan 

Thank you for your feedback. Above, I send my code to registering the level control cluster data model on the esp_zb_task(void *pvParameters) function:

(...)
/* on-off cluster create with standard cluster config*/
esp_zb_on_off_cluster_cfg_t on_off_cfg;
on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);

esp_zb_level_cluster_cfg_t level_cfg;
level_cfg.current_level = ESP_ZB_ZCL_LEVEL_CONTROL_CURRENT_LEVEL_DEFAULT_VALUE;
esp_zb_attribute_list_t *esp_zb_level_cluster = esp_zb_level_cluster_create(&level_cfg);

esp_zb_attribute_list_t *esp_zb_efapel_cluster = esp_zb_zcl_attr_list_create( 0xF000);  

/* create cluster lists for this endpoint */
esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();

esp_zb_cluster_list_add_on_off_cluster( esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE );
esp_zb_cluster_list_add_level_cluster( esp_zb_cluster_list, esp_zb_level_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE );
esp_zb_cluster_list_add_custom_cluster( esp_zb_cluster_list, esp_zb_efapel_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE );

esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
/* add created endpoint (cluster_list) to endpoint list */
esp_zb_endpoint_config_t endpoint_config = {
  .endpoint = EFAPEL_SWITCH_ENDPOINT,
  .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
  //.app_device_id = ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID,
  .app_device_id = ESP_ZB_HA_HOME_GATEWAY_DEVICE_ID,
  .app_device_version = 0
};

esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, endpoint_config );
esp_zb_device_register(esp_zb_ep_list);

esp_zb_core_action_handler_register(zb_action_handler);

esp_zb_set_primary_network_channel_set( ESP_ZB_PRIMARY_CHANNEL_MASK );

ESP_ERROR_CHECK( esp_zb_start(true) );

esp_zb_main_loop_iteration(); 

(...)

Could you please ca check if something is missing in my code?

Even so, i will upgrade the zigbee and zboss library for the new version and I send my feedback after. 

kind regards
nalves

xieqinan, 2024-07-22:
@nalves23 ,

The issue is triggered by the missing level control cluster client, it MUST be assigned when send level control command in esp-zigbee-sdk v1.3.2. You can add the below code to fix it. The esp-zigbee-sdk v1.4.0 allows the user to send commands without client cluster.
```
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);

    esp_zb_level_cluster_cfg_t level_cfg;
    level_cfg.current_level = ESP_ZB_ZCL_LEVEL_CONTROL_CURRENT_LEVEL_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_level_server_cluster = esp_zb_level_cluster_create(&level_cfg);
    // level control client for sending command
    esp_zb_attribute_list_t *esp_zb_level_client_cluster = esp_zb_level_cluster_create(NULL);

    esp_zb_attribute_list_t *esp_zb_efapel_cluster = esp_zb_zcl_attr_list_create(0xF000);

    /* create cluster lists for this endpoint */
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();

    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_level_cluster(esp_zb_cluster_list, esp_zb_level_server_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_level_cluster(esp_zb_cluster_list, esp_zb_level_client_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
    esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_efapel_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    /* add created endpoint(cluster_list) to endpoint list */
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = ESP_ZB_AF_HA_PROFILE_ID,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        //.app_device_id = ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID,
        .app_device_id = ESP_ZB_HA_HOME_GATEWAY_DEVICE_ID,
        .app_device_version = 0,
    };

    ESP_ERROR_CHECK(esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, endpoint_config));
    ESP_ERROR_CHECK(esp_zb_device_register(esp_zb_ep_list));
```

chshu, 2024-09-29:
Closing as the question is answered.

closed 2024-09-29 by chshu