espressif/esp-zigbee-sdk #385: How to add ESP_ZB_ZCL_CLUSTER_ID_BINARY_VALUE & ESP_ZB_ZCL_CLUSTER_ID_MULTI_INPUT clusters in cluster list

mundevx, 2024-07-18:
### Question

Hello,
I am trying to create a Zigbee binary value for my project with the following clusters:
1. ESP_ZB_ZCL_CLUSTER_ID_BINARY_VALUE
2. ESP_ZB_ZCL_CLUSTER_ID_MULTI_INPUT

{
    esp_zb_attribute_list_t *esp_zb_custom_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BINARY_VALUE);
    esp_zb_cluster_list_add_binary_value_cluster(esp_zb_cluster_list, esp_zb_custom_cluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

    esp_zb_attribute_list_t *esp_zb_cluster_multi_input_value = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_MULTI_INPUT);
    esp_zb_cluster_list_add_multistate_input_cluster(esp_zb_cluster_list, esp_zb_cluster_multi_input_value, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);
}

The problem comes when I try to create the clusters in list which doesn't have their corresponding functions.
Function "esp_zb_cluster_list_add_binary_value_cluster" & "esp_zb_cluster_list_add_multistate_input_cluster" doesn't exists.

It also cannot be added as custom cluster using function "esp_zb_cluster_list_add_custom_cluster".  And I am not able to find another function to add these two clusters.  

How i can add these two clusters ESP_ZB_ZCL_CLUSTER_ID_BINARY_VALUE and ESP_ZB_ZCL_CLUSTER_ID_MULTI_INPUT  in esp_zb_cluster_list cluster list.

xieqinan, 2024-07-19:
@mundevx,

The `ESP_ZB_ZCL_CLUSTER_ID_BINARY_VALUE` and `ESP_ZB_ZCL_CLUSTER_ID_MULTI_INPUT` have not been supported in current SDK temporarily, we will consider it in the future.

mundevx, 2024-07-20:
@xieqinan , Ok.

closed 2025-08-13 by xieqinan