espressif/esp-zigbee-sdk #387: Why APS Counter appears in APS Payload?

Alfff, 2024-07-24:
### Question

I'm new to Zigbee, so sorry if my questions are too obvious.

My task is to obtain some raw data from a third party, which I don't know what it is, and once I have it, forward it via Zigbee to another device as an APS payload.

In order to check what functions I should use, I took the examples HA_on_off_switch and HA_on_off_light and modified it.

First of all, I tried to modify function `esp_zb_buttons_handler` of HA_on_off_switch, so I send the toggle command using APS API. The data to send should be only a byte with value 2, so I tried:

```c
static void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    uint8_t value[] = {2};
    
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
      esp_zb_apsde_data_req_t req;
      req.dst_addr_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
      req.dst_short_addr = shortAddress;
      req.dst_endpoint = 10;
      req.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
      req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
      req.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
      req.asdu_length = sizeof(value);
      req.asdu = value;
      req.radius = 1;
      req.tx_options = (ESP_ZB_APSDE_TX_OPT_ACK_TX | ESP_ZB_APSDE_TX_OPT_FRAG_PERMITTED);
      req.use_alias = false;
      esp_zb_aps_data_request(&req);
}
```

Board with HA_on_off_light example did not turn on the LED.

Then, I reverted changes in HA_on_off_switch and modified HA_on_off_light to see what APS data the example HA_on_off_switch sends: I called `esp_zb_aps_data_indication_handler_register(aps);` in `esp_zb_task` and defined `aps` as follows:

```c
static bool aps (esp_zb_apsde_data_ind_t ind) {
  printf("APS:\n");
  for (uint8_t i = 0; i < ind.asdu_length; i++) {
    printf("0x%02x ", ind.asdu[i]);
  }
  printf("\n");
  return false;
}
```

Then, I proceed to send toggle commands using the board with HA_on_off_switch example and I saw the following in the terminal:

`0x01 0x00 0x02`
`0x01 0x01 0x02`
`0x01 0x02 0x02`
`0x01 0x03 0x02`
...

Then, I came back to HA_on_off_switch and modified `esp_zb_buttons_handler` as follows:

```c
static void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    static uint8_t value[] = {0x01, 0x00, 0x02};
    
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
      esp_zb_apsde_data_req_t req;
      req.dst_addr_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
      req.dst_short_addr = shortAddress;
      req.dst_endpoint = 10;
      req.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
      req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
      req.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
      req.asdu_length = sizeof(value);
      req.asdu = value;
      req.radius = 1;
      req.tx_options = (ESP_ZB_APSDE_TX_OPT_ACK_TX | ESP_ZB_APSDE_TX_OPT_FRAG_PERMITTED);
      req.use_alias = false;
      esp_zb_aps_data_request(&req);
      value[1]++;
}
```

Now, everything works ok and the board with HA_on_off_light toggles the LED.

But now I have the following questions:

* I assume that the byte which is incremented (`value[1]`) is the APS Counter. Am I right? If yes, why does it appear in the APS payload? If I use the APS API, is this data not managed by the stack?
* What is the value `0x01` which appears first in every frame? I have searched in Zigbee Specification and I did not find what that byte means.
* If I have, for example, the data `0x0A, 0x0B, 0x0C` to be sent to the other Zigbee device and I want to use the APS API, should I fill an array of the form `0x01, aps_counter, 0x0A, 0x0B, 0x0C`?

Thank you in advance

xieqinan, 2024-07-24:
@Alfff 

```
static void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    uint8_t value[] = {2};
    
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
      esp_zb_apsde_data_req_t req;
      req.dst_addr_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
      req.dst_short_addr = shortAddress;
      req.dst_endpoint = 10;
      req.profile_id = ESP_ZB_AF_HA_PROFILE_ID;
      req.cluster_id = ESP_ZB_ZCL_CLUSTER_ID_ON_OFF;
      req.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
      req.asdu_length = sizeof(value);
      req.asdu = value;
      req.radius = 1;
      req.tx_options = (ESP_ZB_APSDE_TX_OPT_ACK_TX | ESP_ZB_APSDE_TX_OPT_FRAG_PERMITTED);
      req.use_alias = false;
      esp_zb_aps_data_request(&req);
}
```

It is a wired issue. I copied the code you provided into the [on_off_switch example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch), and it worked as expected without the extra bytes. The output log on the on_off_light is as follows.

```
APS:
0x04 0x70 0xfc 0x04 0x01 0x01 0x06 0x00 0x00 
APS:
0x02 
APS:
0x02 
APS:
0x02
```

Alfff, 2024-07-24:
@xieqinan if I send 0x02 in my modified on_off_switch example, I receive 0x02 in the on_off_light example (OK) but the LED does not toggle (NOK). If I just download the original on_off_switch example and run the modified on_off_light example, the LED toggles but the received APS is 0x01, 0xXX, 0x02, where 0xXX is a number which is incremented every time I push the button.

If the toggle command is only 0x02, why don't I receive only 0x02 in the APS payload of the receiver? Maybe I'm missing something...

xieqinan, 2024-07-25:
@Alfff，

Alright, I understand your point now. As we know, the Zigbee protocol is divided into IEEE802154, MAC, NWK, APS, and ZCL layers. The `esp_zb_aps_data_request()` operates under the APS layer, and the ZCL data is included in the payload of the APS data. For example, `0x01 0xXX 0x02` represents the ZCL data.

Looking at the ZCL specification, the general ZCL frame format is as follows:
- `0x01` is the frame control, indicating no manufacturer code.
- `0xXX` is the transaction sequence number.
- `0x02` is the command identifier.

So, the `0xXX` represents the ZCL sequence.

[image]

Alfff, 2024-07-25:
Thank you so much for the explanation! @xieqinan

closed 2024-07-25 by Alfff