espressif/esp-zigbee-sdk #390: How to change network key of all end devices of a network?

Alfff, 2024-07-26:
### Question

I am developing a system which requires my device, acting as a coordinator, to periodically rotate the network key. How can I achieve this with the ESP32H2?

Specifically, I am looking for the equivalents of the following two functions:

- [emberBroadcastNextNetworkKey](https://docs.silabs.com/d/zigbee-stack-api/7.2.2/trust-center#ember-broadcast-next-network-key)
- [emberBroadcastNetworkKeySwitch](https://docs.silabs.com/d/zigbee-stack-api/7.2.2/trust-center#ember-broadcast-network-key-switch)

I have tried using `esp_zb_secur_network_key_set`, but it seems to only change the key locally and does not broadcast the change order to the end devices, because the end devices stop responding after the change.

I would appreciate any guidance or examples on how to effectively rotate the network key on the ESP32H2.

Thank you in advance.

xieqinan, 2024-07-29:
@Alfff ,

The `esp_zb_secur_network_key_set()` only supports setting the local network key. The APSME-SWITCH-KEY request is supported in the latest v1.4.1 version. We will include this feature in the next release. Thank you for your patience.

Alfff, 2024-07-29:
@xieqinan Will this feature be included in the NCP protocol?

xieqinan, 2024-07-30:
@Alfff ,

The `esp-zigbee-ncp` is implemented based on the `esp-zigbee-sdk`, allowing developers to use the APIs provided by the `esp-zigbee-sdk` for development. As a result, if the `esp-zigbee-sdk` supports a particular feature, the NCP protocol will also support it.

Alfff, 2024-08-02:
Thanks @xieqinan

Alfff, 2024-08-28:
@xieqinan,

Is there anything new? I don't see any changes in this regard in the release notes for version 1.5.0

xieqinan, 2024-08-28:
@Alfff ,

Please use the following code instead:
```
#include "zboss_api.h"

extern void zb_secur_send_nwk_key_switch(uint8_t param);

typedef struct zb_apsme_switch_key_req_s {
    esp_zb_ieee_addr_t dest_address; /*!< Destination address  */
    uint8_t key_seq_number;          /*!< Sequence Number */
} zb_apsme_switch_key_req_t;

static uint8_t s_dest_addr[] = {0x01, 0x02, 0x03, 0x04, 0x01, 0x02, 0x03, 0x04}; // Unicast to destination address of network

static esp_err_t esp_zb_secur_apsme_switch_key_req(bool broadcasted)
{
    esp_err_t ret = ESP_OK;
    uint8_t param = zb_buf_get_out();
    if (broadcasted) {
        uint16_t *ptr = (uint16_t *)ZB_BUF_GET_PARAM(param, uint16_t);
        *ptr = broadcasted;
    } else {
        zb_apsme_switch_key_req_t *ptr = (zb_apsme_switch_key_req_t *)ZB_BUF_GET_PARAM(param, zb_apsme_switch_key_req_t);
        memcpy(ptr->dest_address, s_dest_addr, sizeof(esp_zb_ieee_addr_t));
        ptr->key_seq_number = 0;
    }
    zb_secur_send_nwk_key_switch(param);
    return ret;
}
```

Alfff, 2024-08-28:
Thank you so much @xieqinan 🥇

closed 2024-08-28 by Alfff