espressif/esp-zigbee-sdk #392: ESP_ZB_ZCL_STATUS_INSUFF_SPACE when reading the first time a large string attribute type

diazmanuel, 2024-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.4

### esp-zigbee-lib version.

1.4.1

### esp-zboss-lib version.

1.4.1

### Espressif SoC revision.

ESP32-H6 and ESP32-H2

### What is the expected behavior?

I am trying to read a string type attribute that contains a large data size (170 bytes) with the function esp_zb_zcl_read_attr_cmd_req. When I make the first attempt of this reading, it returns me with error 0x89 ESP_ZB_ZCL_STATUS_INSUFF_SPACE, after this error I try again said reading and it works correctly. The problem only occurs in the first case, I understand that there is some buffer that after this error modifies its length, I would like to know how to configure said default value to avoid this error every time the device turns on

### What is the actual behavior?

.

### Steps to reproduce.

.

lpy4105, 2024-07-26:
Hi @diazmanuel,

Generally speaking, this is an expected behavior.

The error happens when the response of your `read_attr_command` couldn't fit into a single NPDU and the responding device didn't know the APS fragmentaion capability of the peer (requesting device), usually when the responding device just turns on or the two devices don't send messages to each other for a long time. The second attempt works because the responding device will query the APS fragmentaion capability of the peer device via `node_desc_req`.

Is implementing some retry mechanisms for reading such attributes acceptable in your case?

diazmanuel, 2024-07-26:
Hello @lpy4105 , thanks for the quick response, ideally we want this error not to occur, since that means having to add a retry logic (which for the moment we will implement to be able to move forward), anyway you mention that through the node_desc_req request the device obtains the fragmentation capacity of the survey device, is it possible to configure this parameter by hand?

diazmanuel, 2024-09-30:
Hello @lpy4105 do you have any update about this issue? I understand that the error response is correct from what you mention, but I don't think that correct logic like the one I implement should consider that the first response is always an error, is there a way to solve it or do you plan to solve it in the future?

lpy4105, 2024-09-30:
Hi @diazmanuel You could try to manually do `esp_zb_zdo_node_desc_req` to ensure that the capability of APS fragmentation is cached on the requested device.

closed 2024-12-02 by diazmanuel