espressif/esp-zigbee-sdk #393: How to update OTA firmware with Home Assistant (zigbee)

mundevx, 2024-07-27:
### Question

Hi, 
I need some help in OTA update of zigbee device through Home Assistant. How i can do this? I read many documents, i tried and failed to execute OTA update from Home Assistant.

mundevx, 2024-08-05:
i am using espressif zigbee light example with OTA client.

lhespress, 2024-08-05:
@mundevx Did you try https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_ota example?

mundevx, 2024-08-05:
yes, i tried this example. This for OTA server and client and its working fine. But in my case, OTA server is Home assistant, who will update the firmware through GUI.  I combine espressif zigbee Light example with OTA client example. Now, how i can update firmware through Home Assistant.

lhespress, 2024-08-06:
@mundevx Could you try again use OTA  client example with Home assistant? BTW, could you share how to select firmware through GUI? I want to check it locally.

mundevx, 2024-08-06:
@lhespress, I also need answer for this.

lhespress, 2024-08-07:
@mundevx What coordinator do you have? Do you confirm whether support OTA cluster or not?

mundevx, 2024-08-07:
@lhespress ,  i am using Home assistant as coordinator? OTA cluster is supported in Home Assistant.

lhespress, 2024-08-12:
@mundevx I mean which `Zigbee coordinator hardware` work on Home assistant, eg: deCONZ, EmberZNet or other hardware.

mundevx, 2024-08-12:
@lhespress , i am working on EmberZNet hardware.

lhespress, 2024-08-21:
@mundevx 
- You need run [ota_client](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_ota/ota_client) firmware on H2 Device.
- Your Home Assistant ZHA subsystem must be configured for [over-the-air updates (OTA)](https://www.home-assistant.io/integrations/zha#ota-updates-of-zigbee-device-firmware).
- Copy the updated firmware file into the OTA directory using a file transfer program.
- Run the `ZHA Toolkitr` and push the firmware to the device.

chshu, 2024-09-29:
Closing as the questions is answered.

closed 2024-09-29 by chshu

P-R-O-C-H-Y, 2025-02-07:
@chshu Can this be reviewed and checked again?
I am struggling to make the Home Assistant OTA to work to update the ESPs

lhespress, 2025-02-10:
@P-R-O-C-H-Y PTAL [ZHA OTA Instructions.pdf](https://github.com/user-attachments/files/18729434/ZHA.OTA.Instructions.pdf)

lhespress, 2025-02-12:
@P-R-O-C-H-Y I'm not sure because of i didn't have`SonOFF Zigbee`dongle. BTW, could you share which step fail?

P-R-O-C-H-Y, 2025-02-12:
The step which is failing is just the Update not showing up in the ZHA for the device.

P-R-O-C-H-Y, 2025-02-12:
@lhespress Is there any way to force ZHA to check if there is an update available?

P-R-O-C-H-Y, 2025-02-12:
@lhespress All good now :) Thank you again for the PDF Instructions. I was using old code in the ota handlers.

mundevx, 2025-02-22:
> [@lhespress](https://github.com/lhespress) All good now :) Thank you again for the PDF Instructions. I was using old code in the ota handlers.

Hello @lhespress , can you share the script. In case of me, pdf instructions are not working. 

There are two parameters in esp OTA code, which is running vesrion.
#define OTA_UPGRADE_RUNNING_FILE_VERSION    0x01010100                              
#define OTA_UPGRADE_DOWNLOADED_FILE_VERSION 0x01010101

Then i updated code to generate bin file. 
#define OTA_UPGRADE_RUNNING_FILE_VERSION    0x01010101                             
#define OTA_UPGRADE_DOWNLOADED_FILE_VERSION 0x01010102

The script which is not working is:
zha:
  zigpy_config:
    ota:
      providers:
      - type: advanced
        warning: I understand I can destroy me devices by enabling OTA updates
          from files. Some OTA updates can be mistakenly applied to the wrong device,
          breaking it. I am consciously using this at my own risk.
        path: /config/zigpy_ota

what should be the actual Home Assistant script?

lhespress, 2025-02-24:
@mundevx PTAL [generate-and-apply-ota-upgrade-image](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32h2/user-guide/zcl_ota_upgrade.html#generate-and-apply-ota-upgrade-image) which have shared the script for generate the bin file.

mundevx, 2025-02-25:
[image] In this image, providers: advanced is not working in home assistant.

P-R-O-C-H-Y, 2025-02-25:
Hi @mundevx, tha HA configuration is in the [ZHA OTA Instructions.pdf](https://github.com/user-attachments/files/18729434/ZHA.OTA.Instructions.pdf). If your HA don't recognize the advanced, try to update to the latest version of HA.

mundevx, 2025-02-25:
@P-R-O-C-H-Y ,

As per pdf, now there is no error in Home Assistant. I also generated the image file from "python c:/esp/esp-zigbee-sdk/tools/image_builder_tool/image_builder_tool.py --create build/ota_data_initial.bin --manuf-id 0x1001 --image-type 0x1011 --version 0x01010110 --tag-id 0x0000 --tag-file build/ota_data_initial.bin". and copied to the "/config/zigpy_ota" folder. But still not getting new firmware upgrade visibility in device. Please help me

P-R-O-C-H-Y, 2025-02-25:
@mundevx If you have correct version in the ota_file and lower version in the running image on the device, you should see the Update. If not, can you remove/add the device again to the HA Zigbee network or request the update from the device.

mundevx, 2025-02-27:
Hi @P-R-O-C-H-Y ,

I run [ota_client](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_ota/ota_client) firmware on H2 Device with Home assistant. But i stuck at this message:
Queried OTA image from address: 0x0000, endpoint: 5
I (63779) ESP_OTA_CLIENT: Image version: 0x0, manufacturer code: 0x0, image size: 0
I (63789) ESP_OTA_CLIENT: Approving OTA image upgrade 

I am getting Image Version, manuf_code, and image_size to zero.  On Home assistant side, its showing 'unknow' in firmware.
Can you please tell me what is happening.

P-R-O-C-H-Y, 2025-02-27:
I was seeing this also. Did you try to restart ZHA, remove/add the device again?

mundevx, 2025-02-27:
> I was seeing this also before. Did you try to restart ZHA, remove/add the device again?

Yes, I did it many times, but no success

urbalke, 2025-10-26:
im  stuck here aswell :) any luck resolveing it? Prochy? Anyone?