espressif/esp-zigbee-sdk #394: How to recover the network from a failure of one of its elements?

Alfff, 2024-07-29:
### Question

I'm trying to figure out what happens in case an element of the network fails.

If I use the examples `HA_on_off_light` and `HA_on_off_switch`, everything works well if either the light or the switch reset.

However, I have some problems if I modify `HA_on_off_switch` to not to use binding, modifying `esp_zb_buttons_handler` like this:

```c
static void esp_zb_buttons_handler(switch_func_pair_t *button_func_pair)
{
    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
        /* implemented light switch toggle functionality */
        esp_zb_zcl_on_off_cmd_t cmd_req;
        cmd_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
        cmd_req.zcl_basic_cmd.dst_endpoint = 10; //Added by me
        cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = shortaddr;  //Added by me. This is a global variable filled in ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE
        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT; //Changed by me from ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT
        cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;
        ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command");
        esp_zb_zcl_on_off_cmd_req(&cmd_req);
    }
}
```

With this code, it works perfectly when the end device joins the network. It keeps working after the end device resets and re-join the network, **but it stops working when the coordinator resets**: I can see `ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command` in the terminal, but the light does not switch. I need the end device to reset so that it works again.

Why does it work with binding and not without? What should I do to get the network working again after a coordinator failure if I don't want to use binding?

xieqinan, 2024-07-30:
@Alfff ,

Could you please print the `shortaddr` when you send a command, such as `ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command to short: 0x%4hx", shortaddr);`? I think this issue is triggered by the reset of the global variable `shortaddr` when the coordinator resets.

Alfff, 2024-08-02:
You were right... I completely forgot about the value of the variable after the reset. I saved it in NVS and now it works.
Thanks!

closed 2024-08-02 by Alfff