espressif/esp-zigbee-sdk #395: esp_zb_address_short_by_ieee and esp_zb_ieee_address_by_short does not send out packets in router mode.

remenyo, 2024-07-29:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### esp-zigbee-lib version.

1.4.1

### esp-zboss-lib version.

1.4.1

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

The device should send out a ZDO nwk address request on esp_zb_address_short_by_ieee() and waits for a response, or times out.__

### What is the actual behavior?

The router device does not even send out the request and immediately returns 0xffff.

### Steps to reproduce.

- Set up a coordinator
- Flash the code below to a 802.15.4 capable esp32 (c6 or h2, in my case it was a c6)
- Connect an end device and the just flashed router to the coordinator.

```c
#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"

#include <freertos/FreeRTOS.h>
#include <freertos/FreeRTOSConfig.h>
#include <freertos/task.h>
#include <freertos/queue.h>

#include "esp_attr.h"
#include "esp_eth.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_mac.h"
#include "esp_netif.h"
#include "esp_spi_flash.h"
#include "esp_spiffs.h"
#include "esp_timer.h"
#include "esp_vfs_fat.h"
#include "esp_system.h"

#include "string.h"
#include "nvs_flash.h"

#include "esp_zigbee_core.h"

static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    ESP_ERROR_CHECK(esp_zb_bdb_start_top_level_commissioning(mode_mask));
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p = signal_struct->p_app_signal;
    esp_zb_app_signal_type_t sig_type = /* static_cast<esp_zb_app_signal_type_t>( */ *p_sg_p /* ) */;
    esp_err_t err_status = signal_struct->esp_err_status;

    ESP_LOGI("Demo", "> ZDO signal: \"%s\" (0x%x), status: %s", esp_zb_zdo_signal_to_string(sig_type), sig_type, esp_err_to_name(err_status));

    switch (sig_type)
    {

    case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_INITIALIZATION);
        break;
    case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
    case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK)
        {
            if (esp_zb_bdb_is_factory_new())

            {
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            }
        }
        else
        {
            esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
        }
        break;
    case ESP_ZB_BDB_SIGNAL_STEERING:
        if (err_status != ESP_OK)
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb, ESP_ZB_BDB_MODE_NETWORK_STEERING, 5000);
        break;
    default:
        break;
    }
}

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg;
    memset(&zb_nwk_cfg, 0, sizeof(esp_zb_cfg_t));

    zb_nwk_cfg.install_code_policy = false;

    // router config
    zb_nwk_cfg.esp_zb_role = ESP_ZB_DEVICE_TYPE_COORDINATOR;
    zb_nwk_cfg.nwk_cfg.zczr_cfg.max_children = 10;

    esp_zb_set_trace_level_mask(ESP_ZB_TRACE_LEVEL_VERBOSE,
                                ESP_ZB_TRACE_SUBSYSTEM_MAC | ESP_ZB_TRACE_SUBSYSTEM_APP);

    esp_zb_init(&zb_nwk_cfg);

    esp_zb_set_tx_power(20);
    esp_zb_set_rx_on_when_idle(true);                    // it's a mains powered device.
    esp_zb_set_primary_network_channel_set((1U << 25U));
    esp_zb_set_secondary_network_channel_set(ZB_TRANSCEIVER_ALL_CHANNELS_MASK);

    ESP_ERROR_CHECK(esp_zb_start(true));
    esp_zb_main_loop_iteration();
}

void controller_task(void *pvParameters)
{
    vTaskDelay(pdMS_TO_TICKS(10000));

    uint64_t mac = 0xa4c13823********;

    esp_zb_lock_acquire(portMAX_DELAY);

    ESP_LOGI("Demo", "Getting short address of 0xa4c13823********...");
    uint16_t short_address = esp_zb_address_short_by_ieee(/* reinterpret_cast<uint8_t *>( */ &mac /* ) */);
    ESP_LOGI("Demo", "Short address is: 0x%04x", short_address);

    ESP_LOGI("Demo", "Getting ieee address of 0x23f4...");
    esp_err_t e = esp_zb_ieee_address_by_short(0x23f4, /* reinterpret_cast<uint8_t *>( */ &mac /* ) */);
    ESP_LOGI("Demo", "esp_zb_ieee_address_by_short: %s 0x%016llx", esp_err_to_name(e), mac);

    esp_zb_lock_release();

    vTaskDelete(NULL);
}

void app_main(void)
{
    esp_zb_platform_config_t config;
    memset(&config, 0, sizeof(esp_zb_platform_config_t));
    config.radio_config.radio_mode = ZB_RADIO_MODE_NATIVE;
    config.host_config.host_connection_mode = ZB_HOST_CONNECTION_MODE_NONE;

    ESP_ERROR_CHECK(esp_zb_platform_config(&config));
    ESP_ERROR_CHECK(nvs_flash_init());

    xTaskCreate(esp_zb_task, "Zigbee_main", 8192 * 2, NULL, 5, NULL);
    xTaskCreate(controller_task, "device_finder", 8192 * 2, NULL, 5, NULL);
}
```

### More Information.

I used wireshark to capture the packets, but there is no network request in it. The device with the short address does exist on the network, because if I create a custom APSDE-DATA.request with esp_zb_aps_data_request containing an end device's hard coded short address, the end device receives the request properly.

I tried finding ieee by short, this also does not send out any packets, instead fills the ieee address field with 0xff.

xieqinan, 2024-07-30:
@remenyo ,

The `esp_zb_address_short_by_ieee()` and `esp_zb_ieee_address_by_short()` only perform conversions between the long address and short address in a known address map. They are not requests. Please consider using the `esp_zb_zdo_ieee_addr_req()` to implement your feature.

remenyo, 2024-07-30:
Thank you for the clarification. I close the issue, as the problem is in my code and not the library.

May I ask you however, how to request network address by ieee address? I want to send requests with esp_zb_aps_data_request, that seemingly only accepts short address, and I did not find the function that allows me to query that by known ieee address.
I would like basically a zigbee equivalent of ARP requests.

Thank you for your help.

closed 2024-07-30 by remenyo