espressif/esp-zigbee-sdk #397: Zigbee transmission interrupt via ESP32C6 to SONOFF Zigbee 3.0 USB Dongle Plus V2 coordinator

MaXeN9818, 2024-07-31:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf-release_v5.1-dc859c1e67

### esp-zigbee-lib version.

1.4.0

### esp-zboss-lib version.

1.4.0

### Espressif SoC revision.

ESP32C6 (esp32c6-20220919)

### What is the expected behavior?

It connects to my Coordinator (sonoff zigbee 3.0 usb dongle plus v2 with Home Assistant integration) and sends temperature data correctly on a regular basis

### What is the actual behavior?

The device connects easily, however, after a few seconds, it stops transmitting data and the ESP32C6 starts to overheat. When I try to reboot the device, the data transmission function does not work properly, forcing me to erase the device from memory and establish a new connection.

### Steps to reproduce.

the code is the example _Arduino-ESP32 Zigbee Temperature Sensor Example_

I compiled everything with Arduino IDE V2.3.2 & esp32 V3.0.3 (I already requested assistance on the esp32 repo)
SONOFF Zigbee 3.0 USB Dongle Plus V2 uses EZNet V.6.10.3 out of the box

1. Compile the _Zigbee Temperature Sensor_ example in an ESP32C6
2. Start the search on the SONOFF Zigbee 3.0 USB Dongle Plus V2 integration
3. Integrate the board
4. Wait a few seconds you will notice that the board stops sending data

### More Information.

I have already requested assistance on the arduino-esp32 repo and was sent here by @P-R-O-C-H-Y 

I emphasize that with the old version (v5.1-b6a66b7d8c) everything works without problems

xieqinan, 2024-08-01:
@MaXeN9818,

> I emphasize that with the old version (v5.1-b6a66b7d8c) everything works without problems

Is it an ESP-IDF version issue?

Could you please perform the above test using the Arduino-ESP32 Zigbee Temperature Sensor and Thermostat Examples with the `idf-release_v5.1-dc859c1e67` version? Additionally, observe whether the overheating phenomenon still exists.

MaXeN9818, 2024-08-01:
> Could you please perform the above test using the Arduino-ESP32 Zigbee Temperature Sensor and Thermostat Examples with the `idf-release_v5.1-dc859c1e67` version? Additionally, observe whether the overheating phenomenon still exists.

I can't do this test "at least not immediately" because this example is not compatible with the old version, I have my own "similar in purpose" script that I developed for both versions I will try with that and let you know

MaXeN9818, 2024-08-01:
**Small update**. I tried to remove the Sonoff Gateway and build my own Zigbee network with two ESP32C6, one ED and one ZCZR, and in this case it seems to work without problems (even if I have not done a long-term test).
Then I tried to flash the firmware of the Sonoff Gateway (keeping the same factory version) and the device worked for 2 hours and 30 before losing the connection and forcing me to manually intervene. I read on some forums that updating the Firmware makes the network more stable.

Tests that I want to try:

- [ ] Create a network between two ESP32C6 and keep it active for 24 hours.
- [ ] Try to update the Sonoff Gateway.

MaXeN9818, 2024-08-08:
> Could you please perform the above test using the Arduino-ESP32 Zigbee Temperature Sensor and Thermostat Examples with the `idf-release_v5.1-dc859c1e67` version? Additionally, observe whether the overheating phenomenon still exists.

Sorry if this took a while. I couldn't get the example to work in the old version but I tested it using my own script. The temperatures are quite similar _(at home the temperature is 26°C/29°C):_
**- 40°C version idf-release_v5.1-dc859c1e67**
**- 38°C version v5.1-b6a66b7d8c**
But honestly the difference is too small and the only difference in the code is the addition of the mutexes `esp_zb_lock_acquire`.

MaXeN9818, 2024-08-08:
> * [x]  Create a network between two ESP32C6 and keep it active for 24 hours.

I created the network between two ESPs and it actually didn't give any problems for more than 24 hours
> * [x]  Try to update the Sonoff Gateway.

I just flashed the USB dongle, it skipped once and lost the connection after that it didn't give me any more problems

**I close this issue by saying that in my opinion the problem was something with the USB dongle firmware**

closed 2024-08-08 by MaXeN9818