espressif/esp-zigbee-sdk #398: INSTALLCODE_POLICY_ENABLE

skoczekrs, 2024-08-02:
### Question

I haven't found any example where INSTALLCODE_POLICY_ENABLE is set to true. I am unable to modify the examples to enable this option. Maybe someone managed to do it?

xieqinan, 2024-08-05:
@skoczekrs,

The patch below may be helpful for you; it is formatted based on esp-zigbee-sdk v1.4.1.

[0001-test-install-code.zip](https://github.com/user-attachments/files/16493653/0001-test-install-code.zip)

xieqinan, 2024-08-29:
@skoczekrs ,

Does the https://github.com/espressif/esp-zigbee-sdk/issues/398#issuecomment-2268591073 help for you? Please feel free to discuss the problems you encounter in this example. If the issue has been resolved, please consider closing it.

closed 2024-09-29 by chshu

josthlee, 2024-10-21:
I have the same issue. I just tried https://github.com/espressif/esp-zigbee-sdk/issues/398#issuecomment-2268591073 and it works. But how can I generate the CRC by myself? Can I do it by esp_crc16_le in esp_crc.h? What is the value of CRC_POLYNOMIAL parameter? Or how can I do? Thank you

josthlee, 2024-10-21:
> @josthlee，
> 
> > I have the same issue. I just tried [#398 (comment)](https://github.com/espressif/esp-zigbee-sdk/issues/398#issuecomment-2268591073) and it works. But how can I generate the CRC by myself? Can I do it by esp_crc16_le in esp_crc.h? What is the value of CRC_POLYNOMIAL parameter? Or how can I do? Thank you
> 
> The install code of the esp-zigbee-sdk uses `x^16 + x^12 + x^5 + x^2 + 1` as the CRC_POLYNOMIAL. You should be able to generate the CRC on your own based on this.

Thank you for promptly reply.
I just research CRC alg a bit,
x^16 + x^12 + x^5 + x^2 + 1 => 1 0001 0000 0010 0101
according to the CRC spec I drop off x^16 bit and  the polynomial is 0x1025
the Kermit CRC version need a reversed (reflected) bit, and the reversed polynomial is 0xA408
I also tried the initial CRC value either 0xFFFF or 0x0000.
 Of course I also tried the esp_crc16_le too.
I found that there are some variant versions of CRC but all the results are incorrect. 
Can you please hint or let me know what's wrong with me? Of course, paste the function that works is absolutely straightforward. Thank you so much.

The following is my code:

#define CRC_POLYNOMIAL 0x1025 
#define CRC_INIT_VALUE 0xFFFF

uint16_t calculate_crc16_ccitt(const uint8_t *data, size_t length) {
    uint16_t crc = CRC_INIT_VALUE;  // Initial value for CRC
    for (size_t i = 0; i < length; i++) {
        crc ^= (data[i] << 8);  // XOR byte into the most significant byte of CRC

        for (uint8_t bit = 0; bit < 8; bit++) {
            if (crc & 0x8000) {
                crc = (crc << 1) ^ CRC_POLYNOMIAL;  // Shift left and XOR with polynomial
            } else {
                crc <<= 1;  // Just shift left
            }
        }
    }
    return crc;  // Return the calculated CRC
}

josthlee, 2024-10-23:
> Thank you for promptly reply. I just research CRC alg a bit, x^16 + x^12 + x^5 + x^2 + 1 => 1 0001 0000 0010 0101 according to the CRC spec I drop off x^16 bit and the polynomial is 0x1025 the [...]

I am afraid that you misunderstood something.
I am still struggling with this CRC issue and it is NOT WORKING.

xieqinan, 2024-10-29:
@josthlee ,

The reversed CRC-16/X25 algorithm can be used to calculate the CRC as shown in the code below:

```c
uint16_t crc16(uint8_t *data, size_t length)
{
    uint16_t crc = 0xFFFF;
    uint16_t polynomial = 0x8408;
 
    for (size_t i = 0; i < length; i++) {
        crc ^= data[i];
        for (uint8_t j = 0; j < 8; j++) {
            if (crc & 0x0001) {
                crc = (crc >> 1) ^ polynomial;
            } else {
                crc >>= 1;
            }
        }
    }
 
    crc ^= 0xFFFFU;
    return (crc >> 8) | (crc << 8);
}
```

For the input `966b9f3ef98ae605`, the resulting CRC is `9708`.

josthlee, 2024-10-30:
> @josthlee ,
>
> The reversed CRC-16/X25 algorithm can be used to calculate the CRC as shown in the code below:
>
> ```c
> uint16_t crc16(uint8_t *data, size_t length)
> {
>     uint16_t crc = 0xFFFF; [...]

Thank you so much.
I have already tested it and it works like a charm!