espressif/esp-zigbee-sdk #400: How to know when a sleepy end device has disappeared from the network?

Alfff, 2024-08-05:
### Question

If I flash both HA_on_off_switch and HA_on_off_light examples, I can see these logs from the coordinator in the terminal if I turn off the end device after pairing and press the button in the coordinator:

```
I (32472) ESP_ZB_ON_OFF_SWITCH: Send 'on_off toggle' command
I (32492) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (35502) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (38512) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (41522) ESP_ZB_ON_OFF_SWITCH: ZDO signal: NLME Status Indication (0x32), status: ESP_OK
I (44512) ESP_ZB_ON_OFF_SWITCH: ZDO signal: ZDO Device Unavailable (0x3c), status: ESP_OK
```

I can see in Wireshark that each NLME Status Indication corresponds to an attempt to send the message without obtaining a response from the end device.

However, if I flash the light_sleep example, I can see that the coordinator only sends messages after it receives `Data Request` from the end device. Then, it is not possible to obtain a NLME Status Indication if the end device has disappeared.

I think that a workaround would be monitoring the reception of `Data Request` commands, but I don't know how to do this using the SDK.

Could you tell me if this is possible and how to do it?

Thank you in advance.

xieqinan, 2024-08-12:
@Alfff ,

> However, if I flash the light_sleep example, I can see that the coordinator only sends messages after it receives Data Request from the end device. Then, it is not possible to obtain a NLME Status Indication if the end device has disappeared.

I believe that when you send a command to a missing sleeping end device, the NLME Status Indication signal will also be triggered, but the interval is 15 seconds. However, relying on this method to determine whether a sleeping device has disappeared is not ideal.

Instead, you can call `esp_zb_nwk_get_next_neighbor()` to iterate through the neighbor table to detect the status of the sleeping device. If the end device have not send the `data_request` to its parent, the router parent will remove it.

Alfff, 2024-08-14:
Thanks @xieqinan

closed 2024-08-14 by Alfff