espressif/esp-zigbee-sdk #401: esp_zigbee_type.h seems to be missing Zigbee HA standard illuminance sensor clusters.

elgerg, 2024-08-08:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

n/a

### esp-zigbee-lib version.

Current

### esp-zboss-lib version.

Unknown, looking at the source in this repo

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

I'd expect a illuminance section in esp_zigbee_type.h similar to:

```
/**
 * @brief Zigbee HA standard temperature sensor clusters.
 *
 */
typedef struct esp_zb_temperature_sensor_cfg_s {
    esp_zb_basic_cluster_cfg_t basic_cfg;                /*!<  Basic cluster configuration, @ref esp_zb_basic_cluster_cfg_s */
    esp_zb_identify_cluster_cfg_t identify_cfg;          /*!<  Identify cluster configuration, @ref esp_zb_identify_cluster_cfg_s */
    esp_zb_temperature_meas_cluster_cfg_t temp_meas_cfg; /*!<  Temperature measurement cluster configuration, @ref esp_zb_temperature_meas_cluster_cfg_s */
} esp_zb_temperature_sensor_cfg_t;
```

### What is the actual behavior?

Nothing, it's not there

### Steps to reproduce.

open esp_zigbee_type.h and search for illuminance, there is only one section for attributes but not one for the cluster.

xieqinan, 2024-08-09:
Hello @elgerg,

I couldn't find the definition for an illuminance sensor in the Zigbee Home Automation specification. Did you mean a light sensor? If so, adding the structure is straightforward. You can follow the code below.

```
typedef struct esp_zb_light_sensor_cfg_s {
    esp_zb_basic_cluster_cfg_t basic_cfg;                /*!<  Basic cluster configuration, @ref esp_zb_basic_cluster_cfg_s */
    esp_zb_identify_cluster_cfg_t identify_cfg;          /*!<  Identify cluster configuration, @ref esp_zb_identify_cluster_cfg_s */
    esp_zb_illuminance_meas_cluster_cfg_t illuminance_meas_cfg;
} esp_zb_light_sensor_cfg_t;
```

elgerg, 2024-08-09:
Ah, my bad. I assume light sensor was illuminance based on this being in the header:

```
/**
 * @brief Zigbee standard mandatory attribute for illuminance measurement cluster
 *
 */
typedef struct esp_zb_illuminance_meas_cluster_cfg_s {
    uint16_t measured_value;                     /*!<  The attribute indicates the illuminance from 0x0000 to 0xffff */
    uint16_t min_value;                          /*!<  The attribute indicates minimum value of the measured value */
    uint16_t max_value;                          /*!<  The attribute indicates maximum value of the measured value */
} esp_zb_illuminance_meas_cluster_cfg_t;
```

So, to add your code do I need to edit my local version of esp_zigbee_type.h?

Sorry for the dumb questions, I'm trying to get this working under Arduino until the official wrapper is available.

Thanks in advance!

xieqinan, 2024-08-12:
@elgerg ,

> So, to add your code do I need to edit my local version of esp_zigbee_type.h?

I suggest adding the code to your local file. I'm concerned that adding it to `esp_zigbee_type.h` in the IDF component might cause a compilation issue.

P-R-O-C-H-Y, 2024-08-14:
@xieqinan The light sensor is defined in Zigbee Home Automation specification as 0x0106. Can this be added please?

xieqinan, 2024-08-15:
@P-R-O-C-H-Y ，

> the light sensor is defined in Zigbee Home Automation specification as 0x0106. Can this be added please?

The Light Sensor Number will also be added in the next release. In the meantime, please add it to your local file.

closed 2024-08-28 by espressif-bot

xieqinan, 2024-08-28:
@elgerg ,

This issue has been resolved in esp-zigbee-sdk v1.5.0. Please update to this version and test again.