espressif/esp-zigbee-sdk #405: ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!

rcaceiro, 2024-08-17:
### Answers checklist.

- [X] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [X] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### esp-zigbee-lib version.

1.4.1

### esp-zboss-lib version.

1.4.1

### Espressif SoC revision.

ESP32-H2

### What is the expected behavior?

I am trying to create a dehumidifier, as the zigbee specification expects.

Because you doesn't have a kind of builder (esp_zb_basic_cluster_create or esp_zb_on_off_cluster_create) I am using a custom attr list (esp_zb_custom_cluster_add_custom_attr) to add the atrributes to the enpoint.

When I start to call the function esp_zb_custom_cluster_add_custom_attr I expect the method add the attribute to the list.

### What is the actual behavior?

The behavior that I received is an error doing it, ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!

### Steps to reproduce.

``` 
uint8_t zero = 0x00;
uint8_t other = 0x1E;
uint8_t another = 0x14;
esp_zb_attribute_list_t *esp_zb_dehumidifier_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_DEHUMID_CONTROL);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0000, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &zero);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0001, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &zero);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0010, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &other);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0013, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &another);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0014, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &other);
```

xieqinan, 2024-08-19:
@rcaceiro ,

The `esp_zb_cluster_add_attr()` might be suitable for your case. However, please note that the `ESP_ZB_ZCL_CLUSTER_ID_DEHUMID_CONTROL` cluster is not supported in version v1.4.1, meaning there are no dehumid control handlers for its commands. You **must** implement this yourself by registering the `esp_zb_raw_command_handler_register()`. You can refer to the following link for more details: https://github.com/espressif/esp-zigbee-sdk/issues/94#issuecomment-1706280284.

aerosh, 2024-08-19:
@xieqinan , I'm facing a similar problem and I tried using `esp_zb_clusterer_add_attr()`
I am creating a custom cluster in this way:
```
#define TVOC_CUSTOM_CLUSTER              0xFFF2
uint16_t undefined_value;
undefined_value = 0;
esp_zb_attribute_list_t *custom_tvoc_attributes_list = esp_zb_zcl_attr_list_create(TVOC_CUSTOM_CLUSTER);
esp_zb_custom_cluster_add_custom_attr(custom_tvoc_attributes_list, 0, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_MANUF_SPEC | ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &undefined_value);
esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, custom_tvoc_attributes_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
// ------------------------------ Create endpoint list ------------------------------
// -------------------------------- Register Device ---------------------------------
```
I tried using `esp_zb_cluster_add_attr(custom_tvoc_attributes_list, TVOC_CUSTOM_CLUSTER, 0, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_MANUF_SPEC | ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &undefined_value);` instead of `esp_zb_custom_cluster_add_custom_attr(custom_tvoc_attributes_list, 0, ESP_ZB_ZCL_ATTR_TYPE_U16, ESP_ZB_ZCL_ATTR_MANUF_SPEC | ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &undefined_value);`

Then I try to set the value in this way:
```
uint16_t tvoc = 100;
reportAttribute(SENSOR_ENDPOINT, TVOC_CUSTOM_CLUSTER, 0, &tvoc, 2);
```

Here is the body of the `reportAttribute` function:
```
void reportAttribute(uint8_t endpoint, uint16_t clusterID, uint16_t attributeID, void *value, uint8_t value_length)
{
    esp_zb_zcl_report_attr_cmd_t cmd = {
        .zcl_basic_cmd = {
            .dst_addr_u.addr_short = 0x0000,
            .dst_endpoint = endpoint,
            .src_endpoint = endpoint,
        },
        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
        .clusterID = clusterID,
        .attributeID = attributeID,
        .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
    };
    esp_zb_zcl_attr_t *value_r = esp_zb_zcl_get_attribute(endpoint, clusterID, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, attributeID);

    if (value_r == NULL) {
    ESP_LOGE(TAG, "Failed to get attribute for cluster 0x%04x, attribute 0x%04x", clusterID, attributeID);
    return;
    }

    memcpy(value_r->data_p, value, value_length);

    esp_zb_zcl_report_attr_cmd_req(&cmd);
}
```

I get an error message in the logs: `Failed to get attribute for cluster 0xfff2, attribute 0x0000`
At the same time, if I use standard clusters, I do not receive error messages and can view the values in zigbee2mqtt

rcaceiro, 2024-08-19:
> @rcaceiro ,
> 
> The `esp_zb_cluster_add_attr()` might be suitable for your case. However, please note that the `ESP_ZB_ZCL_CLUSTER_ID_DEHUMID_CONTROL` cluster is not supported in version v1.4.1, meaning there are no dehumid control handlers for its commands. You **must** implement this yourself by registering the `esp_zb_raw_command_handler_register()`. You can refer to the following link for more details: [#94 (comment)](https://github.com/espressif/esp-zigbee-sdk/issues/94#issuecomment-1706280284).

Hi @xieqinan 
Thank you for your response I'll try and let you know.

rcaceiro, 2024-08-19:
@xieqinan 

Now I have this error:
```E (692) ESP_ZIGBEE_CLUSTER: esp_zb_cluster_list_check(74): Custom cluster id is not within specified range!```

when I do:
```esp_zb_cluster_list_add_custom_cluster(esp_zb_cluster_list, esp_zb_garage_door_opener_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);```

xieqinan, 2024-08-20:
@aerosh ,

Could you please remove the `ESP_ZB_ZCL_ATTR_MANUF_SPEC` access from the custom attribute and try it again?

xieqinan, 2024-08-20:
@rcaceiro ,

> E (692) ESP_ZIGBEE_CLUSTER: esp_zb_cluster_list_check(74): Custom cluster id is not within specified range!

The `esp_zb_cluster_list_add_custom_cluster()` function cannot add standard clusters. I believe you'll face many issues if the `ESP_ZB_ZCL_CLUSTER_ID_DEHUMID_CONTROL` cluster isn't supported in the SDK. Some of these issues may be difficult to resolve without access to the source code. I recommend closing this issue and opening a new one to request the addition of the `ESP_ZB_ZCL_CLUSTER_ID_DEHUMID_CONTROL` cluster.

rcaceiro, 2024-08-20:
@xieqinan ,
But it exists already, The ESP_ZB_ZCL_CLUSTER_ID_DEHUMID_CONTROL is from the esp_zigbee_lib.

xieqinan, 2024-08-20:
@rcaceiro,

The `ESP_ZB_ZCL_CLUSTER_ID_DEHUMID_CONTROL` is only a declaration; the dehumid_control cluster has not yet been implemented in the esp-zigbee-sdk.