espressif/esp-zigbee-sdk #407: Multiple Endpoints

Gazelle84, 2024-08-19:
### Question

Hi, I’m trying to create multiple endpoints for an esp33-h2 and just cannot get more than one to register in zigbee2mqqt. I’m using the zigbee HA light example which works fine with the provided code. I have used the same code and some snippets to try and add a switch, which compiles but doesn’t register the switch. 

if I comment out the light code, it builds and registers the switch, but with both there it only registers the light. 

Do I need to create a list of endpoints and register the list? Or do I register them individually? I have tried numbering them from 1 and it behaves the same.

I want to connect around 5 devices to the gpio pins and control them individually. They will be a simple on/off switch. I can’t find any example where multiple endpoints are registered.

xieqinan, 2024-08-20:
@Gazelle84 ,

The code below registers five endpoints to the device. You can use it as a reference for your project.
```
    esp_zb_on_off_switch_cfg_t switch_cfg = ESP_ZB_DEFAULT_ON_OFF_SWITCH_CONFIG();
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    for (int i = 0; i < 5; i++) {
        esp_zb_endpoint_config_t endpoint_config = {
            .endpoint = HA_ONOFF_SWITCH_ENDPOINT + i,
            .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
            .app_device_id = ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID,
            .app_device_version = 0,
        };
        zcl_basic_manufacturer_info_t info = {
            .manufacturer_name = ESP_MANUFACTURER_NAME,
            .model_identifier = ESP_MODEL_IDENTIFIER,
        };
        esp_zb_ep_list_add_ep(ep_list, esp_zb_on_off_switch_clusters_create(&switch_cfg), endpoint_config);
        esp_zcl_utility_add_ep_basic_manufacturer_info(ep_list, HA_ONOFF_SWITCH_ENDPOINT + i, &info);
    }
    esp_zb_device_register(ep_list);
```

Gazelle84, 2024-08-21:
Superb, thanks, i will give it a try and report back.

xGreenPandax, 2024-09-11:
Doesnt work for me. I only see the first endpoint in coordinator app

chshu, 2024-09-29:
@Gazelle84 @xGreenPandax Any update on this issue?

xGreenPandax, 2024-09-29:
@chshu nothing. Is it possible to provide complete example with multiple different endpoints?

Daniel-Craciun, 2024-10-02:
Hello everyone, I've managed to get multiple endpoints working with my home made coordinator. What solved the issue for me, was that In my end device I removed the code which updates the attribute reporting info ie. I removed this line from my end device `ESP_ERROR_CHECK(esp_zb_zcl_update_reporting_info(&reporting_info));` and everything started working like a charm. Hope this can help anyone :)

Gazelle84, 2024-10-03:
This is the code i ended up with:

file = esp_zb_light.c

/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 *
 * Zigbee HA_on_off_light Example
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
 */

// Include necessary headers that provide access to various libraries and functions
#include "esp_zb_light.h" // Zigbee light-related functions
#include "esp_check.h" // Error-checking macros
#include "esp_log.h" // Logging functions for debugging
#include "nvs_flash.h" // Flash memory functions (non-volatile storage)
#include "freertos/FreeRTOS.h" // Real-time operating system for task management
#include "freertos/task.h" // FreeRTOS task handling
#include "ha/esp_zigbee_ha_standard.h" // Zigbee Home Automation standard functions
#include "esp_err.h"
#include "string.h"
#include "driver/gpio.h"

// Check if the Zigbee End Device role is defined. If not, throw an error.
#if !defined ZB_ED_ROLE
#error Define ZB_ED_ROLE in idf.py menuconfig to compile light (End Device) source code.
#endif

// Define a constant string to be used as a tag for logging messages
static const char *TAG = "ESP_ZB_ON_OFF_LIGHT";

/********************* Define functions **************************/

//function to initialize the GPIO pins as outputs
static void init_gpio_pins(void) {
    gpio_config_t io_conf = {};
    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pin_bit_mask = (1ULL << SOLENOID_1_GPIO) | (1ULL << SOLENOID_2_GPIO) | (1ULL << SOLENOID_3_GPIO) | (1ULL << SOLENOID_4_GPIO);
    io_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    io_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    gpio_config(&io_conf);
}

/**
 * @brief This function initializes the light driver with a default state (off).
 * 
 * @return esp_err_t ESP_OK indicates successful execution.
 */
static esp_err_t deferred_driver_init(void)
{
    light_driver_init(LIGHT_DEFAULT_OFF); // Initialize the light driver with the light off
    return ESP_OK; // Return OK to indicate success
}

/**
 * @brief This function is a callback that starts the top-level commissioning process in Zigbee.
 * 
 * @param mode_mask Specifies the commissioning mode to start.
 */
static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    // Start commissioning and check if it succeeds. If not, log an error message.
[... 430 lines cut ...]
        .esp_zb_role = ESP_ZB_DEVICE_TYPE_ED,                       /* Set device role to End Device (ED) */ \
        .install_code_policy = INSTALLCODE_POLICY_ENABLE,           /* Set install code policy (disabled in this example) */ \
        .nwk_cfg.zed_cfg = {                                        \
            .ed_timeout = ED_AGING_TIMEOUT,                         /* Set ED aging timeout */ \
            .keep_alive = ED_KEEP_ALIVE,                            /* Set keep-alive interval */ \
        },                                                          \
    }

/*
 * Macro to define the default radio configuration.
 * This includes setting the radio mode to its native Zigbee mode.
 */
#define ESP_ZB_DEFAULT_RADIO_CONFIG()                           \
    {                                                           \
        .radio_mode = ZB_RADIO_MODE_NATIVE,                     /* Set radio mode to native Zigbee mode */ \
    }

/*
 * Macro to define the default host configuration.
 * This includes setting the host connection mode, which in this case is none.
 */
#define ESP_ZB_DEFAULT_HOST_CONFIG()                            \
    {                                                           \
        .host_connection_mode = ZB_HOST_CONNECTION_MODE_NONE,   /* Set host connection mode to none */ \
    }

xieqinan, 2024-10-17:
@xGreenPandax ,

Good job. If the `esp-zigbee-lib` and `esp-zboss-lib` versions are higher than v1.4.0, the ZCL data model cannot be reused. You'll need to create and register the model for each endpoint, just like the example you provided [here](https://github.com/espressif/esp-zigbee-sdk/issues/407#issuecomment-2391243470).

By the way, I think this issue has been addressed by yourself, could you please consider closing it?

Gazelle84, 2024-10-17:
Closing post as this appears to be working for others.

closed 2024-10-17 by Gazelle84

artc7, 2025-06-10:
> [@Gazelle84](https://github.com/Gazelle84) ,
>
> The code below registers five endpoints to the device. You can use it as a reference for your project.
>
> ```
>     esp_zb_on_off_switch_cfg_t switch_cfg = [...]

Sorry if the question is trivial, I'm just starting with Zigbee devices, but after registering the endpoints like this, shouldn't five on/off switches appear in the Exposes tab in Zigbee2MQTT?