espressif/esp-zigbee-sdk #408: How to add dynamic endpoint when coordinator detected a new zigbee device commissioned

mundevx, 2024-08-20:
### Question

In coordinator example, I need dynamically adding endpoints and clusters, when a new zigbee device is commissioned.

xieqinan, 2024-08-29:
@mundevx ,

The SDK does not currently support dynamic endpoints. Could you please provide more details about your application and why you need to add a new endpoint? There might be another way to help resolve your issue.

mundevx, 2024-08-30:
I want to builda  zigbee bridge that act as coordinator and gateway role. On commissioning new device, bridge must add dynamic end point.
If you can help with some already written code example for zigbee bridge then you can share with me. I don't want to use Matter zigbee bridge example.