espressif/esp-zigbee-sdk #409: Implementation of Dehumification and Barrier clusters or remove check from custum clusters

rcaceiro, 2024-08-20:
### Is your feature request related to a problem?

Yes I am implementing a garage door opener and a dehumidifier using zigbee, when I try to create a custom cluster it throws a runtime error of ESP_ZIGBEE_ATTRIBUTE: Custom cluster id is not within specified range!

### Describe the solution you'd like.

At my point of view or you have to implement those clusters or you must not check them, this will allows the specification evolve and this not means development of your side

### Additional context.

the code for the dehumification cluster that I am trying implement.
```
uint8_t zero = 0x00;
uint8_t other = 0x1E;
uint8_t another = 0x14;
esp_zb_attribute_list_t *esp_zb_dehumidifier_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_DEHUMID_CONTROL);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0000, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &zero);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0001, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY, &zero);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0010, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &other);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0013, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &another);
esp_zb_custom_cluster_add_custom_attr(esp_zb_dehumidifier_cluster, 0x0014, ESP_ZB_ZCL_ATTR_TYPE_U8, ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE, &other);
```

xieqinan, 2024-08-29:
@rcaceiro ,

We will add support for the `dehumidification_control` cluster in the next release. Thank you for your patience.

rcaceiro, 2024-09-09:
Hi @xieqinan thank you, do you have some prespective to implement the barrier control also?

rcaceiro, 2024-11-03:
I cannot use the dehumidifier on version 1.6.0, do you have some hint on this

xieqinan, 2024-12-06:
Hi @rcaceiro ,

We have release the esp-zigbee-sdk v1.6.1 which contain the `dehumidification_control` cluster support, please refer to https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_secur.h#L34.

By the way, the `barrier control` will not be supported temporarily, but it will take some time to implement the cluster.

jmsryan, 2025-03-30:
I would also like to see barrier control for garage door zigbee projects and would add my vote to that.