espressif/esp-zigbee-sdk #410: ESP Rainmaker Zigbee gateway

djorr5, 2024-08-21:
### Question

Hi, I am just playing around with the Rainmaker Zigbee gateway example. I noticed that in the `esp_gateway_control_light_on_off` function there is no `esp_zb_lock_acquire` when controlling the zigbee ED. Is this just because it has not been updated? Wondering if I need to apply this if I repurpose the gateway code in regards to Rainmaker. 

### Additional context.

[https://github.com/espressif/esp-rainmaker/blob/master/examples/zigbee_gateway/main/esp_gateway_control.c](url)

xieqinan, 2024-08-21:
@djorr5 ，

Yes, the `esp_gateway_control_light_on_off()` function runs on the Rainmaker task, so it's necessary to add a lock to ensure thread safety. The Zigbee API is also not thread-safe, so if you intend to repurpose it, you'll need to add a lock for that as well.

djorr5, 2024-08-21:
Thanks @xieqinan just wanted to confirm. You might want to get the Rainmaker side to update the code to avoid confusion.

closed 2024-08-21 by djorr5

xieqinan, 2024-08-21:
@djorr5,

>  You might want to get the Rainmaker side to update the code to avoid confusion.

No problem, I will do it later.