espressif/esp-zigbee-sdk #412: Adding Zigbee Plug to Zigbee Gateway

Shivi-6, 2024-08-23:
### Question

I was trying to add an anchor zigbee plug to a zigbee bridge on a ESP gateway kit. 
After flashing and monitoring, network steering starts but it is not able to detect the plug.
On searching a little more, I could find that the code is available only for zigbee on_off light. 
Can you please let me know how to add the zigbee plug to the gateway kit.

xieqinan, 2024-08-23:
@Shivi-6 ,

Could you please provide more details about the detection failure for the plug? If the Zigbee bridge is acting as the coordinator, it will wait for other Zigbee devices to join. A mismatched link key or misaligned joining method can trigger the failure. It would be helpful if you could refer to the [sniffer documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark) to provide a sniffer packet and the console log of the Zigbee bridge.

Shivi-6, 2024-08-29:
Zigbee bridge is acting as a co-ordinator but the rainmaker zigbee bridge example does not have any implementation for a zigbee plug. I could only find for on-off light. Also, the bridge uses steering but I think the devices are using install-code.

Is there a way to change the pairing method of the bridge?

xieqinan, 2024-08-29:
@Shivi-6 ,

> Is there a way to change the pairing method of the bridge?

Of course, you can refer to the example attached in this https://github.com/espressif/esp-zigbee-sdk/issues/398#issuecomment-2268591073 to use the install code way for pairing.