espressif/esp-zigbee-sdk #413: ESP Zigbee and Rainmaker

djorr5, 2024-08-27:
### Question

general question. I am playing around with the Rainmaker side of things and want to setup a Zigbee Gateway so it can be controlled through Rainmaker. However I am forced to use IDF 5.3 due to this issue:
[https://github.com/espressif/esp-matter/issues/1024](url) Ignore the Matter part, I was experiencing the same issue as OP described.
 
But I also remembered that you have stated that the Zigbee must be compiled with 5.1.4. Is there a way to compile the rainmaker code with 5.3 and the zigbee code with 5.1.4? Or will there be no issues with just using 5.3 for the whole project?

Nanowires, 2024-08-27:
@djorr5: Did you manage to have wifi and zigbee running in parallel?
I was only able to connect to a WiFi network, when zigbee wasn't started yet, and as soon as I start the zigbee stuff, the network connection is unusable, (80% of ping drops). Is this only on my side?

djorr5, 2024-08-28:
@Nanowires I am using a 2 SOC solution. 1 x S3 and 1 x H2. I don't believe you can use just the one SOC to do both Zigbee and WiFi at the same time.

chshu, 2024-09-29:
@djorr5 Yes, you can use IDF v5.3 with the Rainmaker Zigbee Gateway example. Let us know if you any issues.

closed 2024-12-19 by chshu