espressif/esp-zigbee-sdk #414: Untimely retry on repeated sending of a command associated with a custom cluster

Matthieu-Poulain, 2024-08-27:
### Question

Hello,

I am trying to build a remote system with a remote and a receiver. 
On this system, for safety reasons, the user has to press a button while he wants the receiver to drive a motor (for now it's just a simple LED). When he stops pressing the button, the motor has to stops within 300ms (timeout).

In order to achieve such a behaviour, i implemented a custom cluster (I'm not sure if this is even the best method though) from which i send every 100ms a custom command telling the receiver to keep the motor ON.
When its done i can send a 2nd command or just wait for the timeout to to stop it. 

**However, at that pace, it seems that sometimes a message may not be received or acknoledged. This ends in the command being sent again while the LED is supposed to be OFF and the LED starts for a short time to turn ON until next timeout.**

I searched for a way to fix this behavior, but neither of the ZBOSS defines i could find and changed solved my issue.
What is the reason behind this behavior? Is the timing (100ms) too short to send such short commands? Is this even the right way to build my system or is there any other API i missed while searching?

Thanks a lot for the help you could provide.

xieqinan, 2024-08-29:
@Matthieu-Poulain,

To reproduce your issue, I created an example based on your description, which is attached. It works as expected on two ESP32H2 chip, their distance is about 50 cm. 

We can discuss the issue based on this patch.
[0001-test-send-frame-in-100ms-interval.zip](https://github.com/user-attachments/files/16796918/0001-test-send-frame-in-100ms-interval.zip)

Matthieu-Poulain, 2024-08-30:
> @Matthieu-Poulain,
> 
> To reproduce your issue, I created an example based on your description, which is attached. It works as expected on two ESP32H2 chip, their distance is about 50 cm.
> 
> We can discuss the issue based on this patch based on esp-zigbee-sdk ([537b9aa](https://github.com/espressif/esp-zigbee-sdk/commit/537b9aa5e572177eb0d015ca0da1765ce6435fff)) . [0001-test-send-frame-in-100ms-interval.zip](https://github.com/user-attachments/files/16796918/0001-test-send-frame-in-100ms-interval.zip)

Hello,
I feel like an idiot... 
I put a debug output to print the received command counter inside the receive handler function... Removing it seems to allow the proper working of the receiver. I really must be more careful with printf & ESP_LOGx functions...

Thanks a lot for your help

closed 2024-09-29 by chshu