espressif/esp-zigbee-sdk #415: Zigbee and Wifi Coexistence

Nanowires, 2024-08-27:
### Question

Today I've tried to setup a WiFi-Zigbee bridge. For this a ESP32-C6 would serve as a Zigbee Coordinator, which would also be able to receive commands via WiFi e.g. for changing the light level of a bulb.
When I tried to connect to the WiFi, when the Zigbee stack was already initialized, the connection failed always. When first connecting to the WiFi and initialize the Zigbee stack afterwards, Zibee still seemed to work without problems, but the WiFi connection was nearly unusable, i.e. client requests weren't received and ping drop rate was at around 80%.

Reading the documentation of RF coexistence (https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-guides/coexist.html), there's nothing stated about WiFi and Zigbee coexistence, only Bluetooth and Zigbee (and WiFi + Bluetooth of course) is mentioned.

So my question: Is it even theoretically possible to use both stacks in coexistence (and usable in the future) or are both protocols incompatible for sharing a single RF module?

Nanowires, 2024-08-28:
Small correction:
WiFi is usable once connected, I've forgot to start the server...

But the connection issue is still present.

xieqinan, 2024-08-29:
@Nanowires ,

Now, is the current issue that the Wi-Fi always connect failed when the Zigbee stack has been initialized?

Nanowires, 2024-08-29:
@xieqinan,

Yes that is correct, sorry for the confusion.
After some more testing, I also detected, that the WiFi connection is lost after some time (something between 0.5 and 3 hours. I need to test a bit more)

OOHehir, 2024-09-14:
I tried something like this before & whatever (i.e Zigbee or WiFi) started first tended to 'grab' the radio & block the other. [This](https://github.com/espressif/esp-idf/blob/master/examples/zigbee/esp_zigbee_gateway/README.md) seem like the current solution.

xieqinan, 2024-10-17:
I think you only need to call `esp_coex_wifi_i154_enable();` before Zigbee has been initialized, and then the Wi-Fi connection can be made at any time.

djorr5, 2024-10-23:
@LoSk-p  is it stable?

LoSk-p, 2024-10-24:
@djorr5 I haven't specifically tested it under continuous use since my requirement is to have the Wi-Fi connection active only for a short duration. I simply turn it on, send the request, and then turn it off, while zigbee connection is active.

xieqinan, 2024-11-08:
Hi @Nanowires,

Do the above comments help you address this issue? If the issue has been resolved, please consider closing it.

Nanowires, 2024-11-12:
Hi @xieqinan,
Sorry for the late response.
I will try to test it today and check if this resolves my question.

Nanowires, 2024-11-12:
The solution seems to work, although it was a bit of a hassle to find out, that I had to explicitly add the "esp_coex" component to my project (didn't find this in any documentation, but maybe I didn't look deep enough).

I will let the device now run a couple of hours, to check if the connections (zigbee and wifi) are stable. Then I will close the issue

Nanowires, 2024-11-13:
With the call to `esp_coex_wifi_i154_enable();` the coexistence of WiFi and Zigbee seems to be stable :-)

closed 2024-11-13 by Nanowires